UnifiedEngineering/T-962-improvements #36: LCD issue

jieter, 2015-01-01:
I've identified the problem with the LCD being off:
[image]

Don't know what exactly happened... 

Package marking is LTKF which seems to be a [LT1617 micropower inverting DC/DC converter] (http://cds.linear.com/docs/en/datasheet/16171f.pdf) Will try to replace that part. Or find a compatible LCD module, which might be

xnk, 2015-01-01:
Oops! Did one of the pins break off, as in no easy way to re-solder the one you have? That's got to be the negative bias regulator for the LC assembly. Probably easiest to replace the regulator than to replace the display.

jieter, 2015-01-02:
I tried resoldering, but it was hurt badly. One pin broke of, I even tried resoldering that pin by grinding off a piece of the case. Did not help.
Will order a replacement part today.

xnk, 2015-01-02:
:( At least it's a trivial replacement procedure with one of those packages.

jieter, 2015-01-02:
yes, should be a breeze. Hopefully, nothing else is broken.

xnk, 2015-01-02:
You can always try to feed an external negative voltage from an isolated power supply through a 1k resistor or so to the output of the converter circuit and ramp up from 0 and see when you get decent contrast. I believe this negative voltage only goes out to the external contrast adjustment trimmer and then back in again. That way you can verify that nothing else is broken.

jieter, 2015-01-07:
Just tried with a 9V battery: provides just enough contrast to see it...

closed 2015-01-07 by jieter

jieter, 2015-01-09:
Alive again ;)

[image]

xnk, 2015-01-09:
Congratulations!!!