UnifiedEngineering/T-962-improvements #46: T962A ramp/speed test logs.

jieter, 2015-01-05:
I've implemented some very crude serial command code in https://github.com/jieter/T-962-improvements/tree/serial-command, enabling me to run both test profiles on the T962A.

files: https://gist.github.com/jieter/f5b6ac49e8ef38c64570

[image]
[image]

xnk, 2015-01-06:
Awesome, thanks a lot!! It would seem that your reflow fan isn’t running fast enough to properly heat all the air around the TCs (looking at the sudden readout increase when the fan starts to run during the final stages of both profiles). These were run with the min-fan-speed set to 7, have you tried increasing it? I would guess it would also improve the perceived slow heating performance during the ramp test as well (the oven isn't hitting the 245C target within the allotted time frame). I have been told that the smaller oven that I have are supposed to be heating slower that the larger A version, but I hit 245C just before it starts cooling down again.

The graphs shows that they are very similar in heating performance though so it would seem that one set of PID tuning values should go a long way!

I will look into this (and the serial code) a bit more tomorrow hopefully, it’s been a long day today.

/wj

jieter, 2015-01-06:
I'll run with an higher min-fan speed later, maybe tonight. I'll first have to be able to change the settings via serial...

closed 2015-02-13 by jieter