UnifiedEngineering/T-962-improvements #49: Compile error in vic.c

Stew1234, 2015-01-11:
Attempting to recompile to use DS1822 instead of DS18B20, 1 wire code should be compatible, however the family code of 0x28 is preventing the use of the DS1822 because it's family code is 0x22. Without 0x28 it assumes there is no temp IC. #define OW_FAMILY_TEMP (0x28)
simple solution change 0x28 to 0x22 and recompile.
Better solution:
#define OW_FAMILY_TEMP_DS18B20 (0x28)
#define OW_FAMILY_TEMP_DS1822 (0x22)
and an OR statement to allow use of either device.
First attempting to recompile with zero changes, I get three errors in VIC.C
I get the following error on Lines 45, 51, 59
Type expected ')' before ':' token
LPCexpresso is in it's default directory, T962 in C:\t962 but I'm guessing this is path issue ?

xnk, 2015-01-11:
That's odd, I compiled the project in lpcxpresso a few days back without issues. I can take a quick look when back at computer shortly. 

Regarding the DS1822, is it fully command compatible? If it will accept the 18B20-style commands then a simple or should work.

Stew1234, 2015-01-11:
I've tried the latest and an earlier version of the source and get the same error. Pulled the .zip in and attempted to compile with no changes. LPC Expesso Version 7.5.0 , would the makefile or .project be looking for a specific path ? 

It seems to be having a problem with this bit of code, asm("MRS %0,cpsr" : "=r" (retval));

The 18B20 and 1822 are  supposed to be code compatible, both are 9 to 12 bit user configurable. We just happened to have some DS1822 here.

xnk, 2015-01-11:
Aha, the issue is that for some reason the default configuration (Debug) does not build. The recommended configuration is Release, and that one does build. Not sure why Debug broke though.. Set Release as active configuration and you should be good to go!=

Stew1234, 2015-01-11:
That was it, changing to Release compiles and works, can confirm the DS1822 is changing cold-junction temperature on the bake mode screen.

Thanks for your work on this, even without the cold-junction working it was working better than new. We've got a 2008 T962A that has always either incinerated boards or left some paste that didn't melt.

It doesn't affect operation, but the "no version info" is on the lcd, from the source I'm guessing it's because it's compiled in LPCExpresso  ?

closed 2015-01-11 by Stew1234

xnk, 2015-01-12:
Great to hear it is working! Yep, the "no version info" is because of lpcxpresso. The git version stuff only works when building using the stand-alone cross compiler (the official releases are built by Travis-CI without any lpcxpresso involvement). 

I can most definitely add DS1822 to the official codebase, the only change you had to perform was the type byte (as you mentioned initially)? 

One last thing out of curiosity: what type and revision of lpc chip does the code detect when it boots in your oven? I believe 2008 is definitely among the older ones I've heard it run on!

Stew1234, 2015-01-12:
Changed only the family code from 28 to 22 and recompiled. Works fine, in bake mode it moves up a bit from 22C to 25C or so, after a lead free run I can change over to bake and the cold junction temp is around 35C. It correctly finds the LPC2144 cpu on bootup, but not the "rev". Just shows "rev -"

The hardware is very similar to the Variant A, the biggest difference is that C20, R2 and C8 were actually loaded on the pcb. Removed C20 and C8, R2 was conveniently a 4.7K, so it was loaded across C20 and R8.

xnk, 2015-01-12:
How handy that they already supplied the components! :) I wonder what that hardware was supposed to do in the first place, those extra components that is. The fact that it shows rev - indicates that the part is older than rev B for the 214x series according to the data sheet. Is it correct that yours might be rev A?=

Stew1234, 2015-01-12:
It's possible it's a rev A cpu, the revision history on the data sheet starts in 2005, the oven date is 08.01 and the serial number is D8006 or 08006. The data sheet doesn't interpret the package marking so no way to know it's date or rev, I guess. These had been for sale on ebay for a while before we bought ours, I suspect there are some 2007, maybe 2006 that were made.

xnk, 2015-01-21:
I have added support for both family IDs now and it's in release 0.4.1