UnifiedEngineering/T-962-improvements #50: 2 requests

Stew1234, 2015-01-16:
Couple of little items.
1. Can the bake temp be stored in eeprom ? resets to 30C after power off.
2. Is the rear fan running at full on cooldown ? It doesn't seem to be running the fan as hard as the old software on cooldown. Can the fan be ran faster after it gets down to 75C ? it takes a while to get it cooled down enough to make another cycle.

jieter, 2015-01-16:
about your second point: The code seems to max the fan out:
https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/io.c#L38-47

I never tried reflowing with the original firmware...

Stew1234, 2015-01-18:
It does until it gets down to about 50C, then it seems to slow down and the PID seems to be turning it on and off. We run a lot of back to back panels through it, and the quicker it can cool down we can get more product through it.

xnk, 2015-01-21:
My experience with it is that when running panels back to back you don't actually want the oven to get cooler than around 50C. I get a lot more consistent results when only force-cooling down to 50C. This is "configurable" in the code ( https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/reflow.c#L48 ). It is not yet available as a setup menu setting, but that is something I was thinking about (together with actually storing the bake temp in eeprom as mentioned above). Let's keep this issue active and track progress here.

Stew1234, 2015-01-28:
Can confirm 0.4.1 works fine with DS1822. We did notice one odd thing, the housing around the fuse on the power entry melted. Fuse was still intact, but the plastic around it got hot enough that fuse clips moved. Is it possible that the more frequent cycling of the heating elements draws much more current ?

xnk, 2015-01-28:
Hmm, if the power port got that hot it sounds like there's excessive resistance in one of the cable connections to it (or inside the IEC plug itself). This was a fairly old oven, right? What about corrosion on the fuse/fuse clips? My power cable assembly that was shipped with the oven had quite a bit of internal resistance creating a significant voltage drop. Replacing that improved heating power. When this was brought up in a discussion it was also mentioned that the internal cabling on the back of the power inlet is probably on the weak side, especially in the A version I would imagine. I will see if I can source an A version to test on as well at some point.

I don't see how the cycling can result in higher power levels. How much resistance do you read if you measure from the AC cord plug to the inside of the oven? Either neutral or the phase likely has excessive resistance. Another way to measure is to measure wall outlet ac voltage and internal voltage in the oven when heater is fully on (bake at high temp) to see the voltage drop. Sounds like it might be excessive and generate heat where you don't want it. 

Great that the temp sensor works though! 

---- Stew1234 wrote ----

Stew1234, 2015-01-28:
We did install a larger power cord from the wall socket to the over. I had not paid the power entry port much attention until it melted, but does look rather weak for a 16A fuse. Actually the wiring inside appears to be minimal for the job. No way to measure it after it melted, I've bypassed the fuse until a new one arrives. It does appear to reach a higher temp on the default lead free profile now. It is an older oven, the SCR failed a couple of years ago rather spectacularly, but we've not had any trouble with the power entry until now. It might be worth considering to put a power entry port with larger conductors as an upgrade for older ovens.

The software seems to be working very well, holding at 50C after cool down is letting us swap pcb's much more quickly, and they don't come out incinerated.

xnk, 2015-01-28:
Sounds great, except for the melted power inlet of course! I figured the 50C point was acceptable to handle the boards, anything higher and it gets uncomfortable..

I'm just curious as to which paste are you using with that lead-free profile? I've only run the three Chipquik pastes named in the three profiles, it would be interesting to hear what other pastes work too! 

Also have you measured how accurate the temperature regulation is in your oven, are the reported temperatures close to the actual one in the oven? The fact that you successfully run lead-free and reflow successfully indicates that it's hot enough at least, and it's not super hot either as the boards aren't incinerated, that's always a good thing! 

---- Stew1234 wrote ----

Stew1234, 2015-01-29:
We've run chipquik SMD4300SNL10, FCTAssembly WS888, and SMTsolderpaste.com PF606-PWA  all work well, but only the center of the tray is useful, we don't put anything within 2 inches left or right, and 4 inches from front or rear - otherwise they won't reflow. Not a software issue, just not enough heat. I haven't independently checked the temperature yet.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk