UnifiedEngineering/T-962-improvements #52: Reset after flashing firmware

jieter, 2015-02-04:
Is it possible to reset de uC after flashing the new software from the bootloader?

xnk, 2015-02-05:
As in jumping to the code from ISP mode? The NXP tool has a "jump to code" command that I used for initial RAM load/execute, that _should_ work for flashed firmware too.

jieter, 2015-02-05:
I did quite some flashing of new code yesterday, and it's quite annoying to have to turn of the oven to reset it after the upload is completed. Maybe I just have to figure out the DTR/RTS for reset/ISP...

xnk, 2015-02-13:
Aha, right! I have both the ISP and RESET pins hooked up. I haven't experimented with the DTR and RTS, but there appears to be a standard how to hook it up. Seems like most terminal programs isn't working properly when hooking dtr/rts up though (forces reset when opening the serial port). Not sure where that was mentioned though..

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk