UnifiedEngineering/T-962-improvements #56: FW v0.5.0 - unable to set calibration constants

toprpav, 2015-02-23:
I want to thank you for your work at first. You saved me a lot of time.

I just opened my oven for the first time and i flashed latest version of firmware. However, I am not able to set the constants - probably there is a problem with reading the button status/button releasing. (I did not examine the code - I guess only). I have no problem with firmware 0.4.1

I would be very pleased to see the current temperature when entering constants.

Thank you, Tom

jieter, 2015-02-24:
Hmm, I think ik broke it. But I can adjust the values, but they jump around. I'll look into it...

jieter, 2015-02-24:
#57 fixes this problem.

wysiwywg, 2015-02-25:
same here. I just set the offset in 0.4.1 and then reflash to 0.5.0. EEPROM data is kept on reflash

xnk, 2015-02-25:
Should be good now after @jieter fixed it, thanks!! A new release should show up shortly.

@wysiwywg : Sounds like you built one of the 4xMAX32850K interfaces, awesome! :)

closed 2015-02-25 by xnk

xnk, 2015-02-25:
Also see #57 issue comments for ideas how to calibrate (and why seeing the current temperature isn't really needed when adjusting the calibration)

toprpav, 2015-02-25:
When using another method it probably is not necessary.

It would be interesting to connect bluetooth module (10 USD) to LPC UART and write android app to management oven, automatic detection module by using the AT command and switching UART mode.
Now I am doing a different project. Otherwise I would help with that.

wysiwywg, 2015-02-25:
@xnk : 
yes it was actually quite easy to hand solder the MAX32850K on the pcb. just pulling one solderblob with the iron and done.
I soldered PLCC32 with chinese solder paste (something leaded lying around for a while) choosing the SYNTECH-LF profile (or highest temp profile, Sn/Pb doesnt work for me here)

results:                                                                                                                                                    
original firmware:
-partial burns of the pcb, yet some parts not soldered
v.0.4.0 with no cold-junction compensation:
-left and right corners in the back of the tray incomplete soldered
v.0.4.0 with DS18B20 : 
-left corners incomplete soldered
v.0.5.0 with MAX32850K interface:
-only the very left corner gets incomplete soldered.

The results are in line with the left temperature always being shown 10-15 degrees lower than the right temperature. Which is strange since the sensors are placed symmetrical (I'm using the small T-962 build ~2014-04.) Will test if it works better with 2 additional tecmocouples once they arrived.

PLCC is difficult to reflow anyways and the old solder paste doesn't make it easier so I'm expecting brilliant results with TQFP packages and new solder paste next month.

Sn42Bi58 Paste of course works like a charm but I heard you have to be strictly RoHs because the lead will lower the alloy melting point to 80-90C

@toprpav:
These modules (5 USD shipped from china) work really well for these purposes. with any serial console app you already have a great monitor taping an old android to the oven:-) )
 I usually take  http://appinventor.mit.edu/explore/ for quickly hacking together such apps.

xnk, 2015-02-27:
@wysiwywg That's great! The only part that is kind of impossible to solder when not reflowing is the exposed thermal pad, but hopefully you got reasonable thermal connectivity anyway.

It's strange that the Sn/Pb profile wouldn't work with leaded paste. I tested with my Amtech 4300 Sn/Pb solder paste that wasn't super old (age shouldn't affect the melting point you would think). That one reflowed just fine. Maybe your paste requires higher temperatures? May be difficult to find the intended reflow profile for some of the noname paste out there. Great that it seems to work at the higher temperature without burning the PCB at least!

wysiwywg, 2015-02-27:
@xnk right now i just had some solder paste under it and reflowed it after soldering, but to be safe I propably should've just drilled a hole through the pcb and then add a blob. Or thermal grease. On the other hand it seems  to dissipate 13mW max so I just went by a feelign telling me it should work:-)

I just posted reflow logs here: https://github.com/UnifiedEngineering/T-962-improvements/issues/60#issuecomment-76382560 => the allover temperature doesn't seem reach the neccesarry temperature (if i trust X1 that is) So a temperature offset shifting the whole profile a few degrees higher could be nice to have.

toprpav, 2015-02-27:
Hi guys. Today I first used the oven to solder MAX31855K thermocouple interface. I never solder such small things like this maxim ic. But times have changed and it is necessary now.
I measured pcb pins after reflow and found one bridge and more bad connections. I used very primitive hot air gun to reflow it again. Now interface working well and I am very happy!

I used external thermometer to measure pcb temperature. I think it is quite necessary to use third thermocouple to measure exact PCB temperature. Otherwise there is risk that the pcb does not reach the melting point. 

Thank you and good luck!

PR #59 (2015-02-25): Fix #56, increase raw values, not scaled ones.
Fixes #56
commit ba26cd15 (2015-02-25): Fix #56, increase and decrease raw values, not scaled ones.
commit e3022e58 (2015-02-25): Merge pull request #59 from jieter/fix-setup

Fix #56, increase raw values, not scaled ones.