UnifiedEngineering/T-962-improvements #57: Thermocouple calibration from scratch

otto44, 2015-02-24:
I screwed up the original blue pot setting and I wonder what is the best way of calibrating oven thermocouples from scratch. I calibrated my K thermocouple thermometer in boiling water corrected to barometric pressure to start with. Would it be OK to clip the thermometer thermocouple to the oven thermocouple to have accurate reference? Than setting offset and gain in V0.5.0 all to zero. Than run Bake at say 150 degC and adjust blue pots to the temperature readout of thermometer and then check it at 250 degC or vice versa? The temperature/voltage of K thermocouple should be reasonably linear, and main interest is in the region of 250 degC. Is this a correct way or would you suggest otherwise?

wulfmans, 2015-02-24:
use ice water  its constant  boiling is not as it depends on your altitude

xnk, 2015-02-25:
I would suggest the following:

As your gain is completely unknown at this point you most likely have to perform the calibration several times (as both temperatures will change accordingly). Zero both gains and offsets in the software as you said. I would then run bake at 150, wait for the temperature to stabilize and read the actual value using your external thermometer. It will most likely be nothing like 150C at that point. Adjust bake to 250C and wait for temperature to stabilize. Read external thermometer again. The _difference_ between these temperature readouts is what's interesting (there should be a 100C difference if your gain is correct). If you're only a few % off you can compensate in the setup menu. The blue pots control the gain, one per channel, if you need to adjust because you're completely off I would suggest a small adjustment and then re-measure the temperature difference until you achieve somewhere between 90 and 100C of measured difference (less than 100C because it's better to adjust the gain down than up in software because of the low resolution of the readout). The value you want to set in the corresponding gain adjust in the setup menu is [actual temp difference] / [ideal temp difference (100)], so for 90/100 you would set 0.90. This will make sure that the oven will now see the exact same difference as you just measured. That sorts out the gain.

If you have fitted the cold-junction sensor and you have the correct gain the offset adjustment is simply [external temperature readout] - [expected temperature]. For example: if your thermometer shows 245C but the oven thinks it reached 250C the offset value for that side of the oven will be +5). 

This should conclude the calibration!

If someone has a better idea regarding calibration please post them! Just keep in mind that ice water won't work for these thermocouples as the hardware ADC interface can't deal with the inside of the oven being colder than the controller board (cold-junction).

otto44, 2015-02-25:
This is a tricky operation, as the set temperature is averaging between L & R and it is constantly moving about and one affects the other. Finally I think I am almost there. I went back to V041, as the setup does not seem to work in V5. Also, the graph display goes back to Main Menu already at 89 deg C. I am sure I have seen mention of 50 deg C somewhere?

xnk, 2015-02-25:
Very tricky, that's for sure. The calibration stuff in setup menu should be fixed again in v0.5.1, now released. And yes the temperatures all affect each other which is why small changes will have to be done iteratively (making sure that the software gains are set to 1.0 before measuring every time. Did you get reasonably close to the desired temperatures?

Regarding the 89C, it very much depends on the reflow profile you're running. If running the normal lead-free profile with 245C peak temperature the oven will not be able to cool down fast enough to reach 50C before end of reflow, but it will continue to cool down to 50C even when it goes back to menu mode.

otto44, 2015-02-26:
I clipped meter thermocouples to the oven thermocouples with flattened miniature crocodile clips, and I managed at set 250C Left 248C, Right 250C and actual inside 250C. I tried reflowing with my old dried up Mechanic solder paste MCN-300 Sn63/Pb37- I had to soften it with a few drops of flux, set profile 4300 and tried few components on a small board, just in the middle of the tray, and SUCCESS! Very nice joints.
Thanks a lot for your advice and help. Going to try you V051. 
Wishing you good luck with your projects.

xnk, 2015-02-26:
Sounds great, and thanks for the feedback!

otto44, 2015-03-03:
This is perhaps slightly off topic, but I am not sure how else to get your comments to this mod:
I made 40mm dia. stainless steel recirculating fan and shoehorned it into T-962 oven above IR heaters and in-between thermocouples. I am very pleased with results, apart from licking my cuts from the innards of the oven which are sharp and cramped. My way of thinking is, if I recirculate 
the air inside, than despite the cooling fan opening at the back and perforated bottom the 
air pressure remains same and there should not be loss of hot air or addition of cold air from 
outside. Whatever number of thermocouples will not even out temperature across the whole oven. 

My configuration is T-962 oven running firmware V051 with cold junction compensation and 
Setup Cooling fan speed =0. I tried a smoke test (filled the oven with smoke) and watched if any 
smoke was exiting the oven and run some soldering tests. There was actually slight escape of 
hot air via cooling fan perforations, due to angle of those internal slots, redirecting the recirculate. I thought about a shutter activated during the heat cycle, which I tried by simply blocking the outside 
vents. But being a lazy so-and-so I realised, I can perhaps balance the air pressure, which 
is reasonably constant regardless temperature, I went back to the cooling fan control in 
Setup, set it to 8 and adjusted the recirculating fan speed with a series resistor to 8V, at 
which point it seemed balanced. This stopped the hot air escape and the temperature across 
the perforated drawer is now even, rise time is faster and actual temperature follows nicely 
the set  profiles. I used so far Sn63/Pb37 paste, which reflowed nicely from corner to corner of drawer perforation. I have not tried lead free or low temperature paste. 

I kept the invasion into the oven minimal, cut off a piece of thermal insulation and drilled 
3 small holes in top of the oven compartment. I wired the fan motor directly onto the 10.2V 
diodes via pot, and as the fan is silent, I leave it on all the time the oven is on.

jieter, 2015-03-03:
@otto44 interesting, can you post a picture?

otto44, 2015-03-04:
OK, here are some pictures:

[image]
Stainless steel fan, shroud, mount, spacers and motor.

[image]
Cutout in the heat shield of the T-962

[image]
Motor side of the recirculating fan

[image]
Fan Inside oven with the drawer removed. The circle above fan was fan scraping inside oven. Need to go back to drawing board and relax tolerances, as the adjustment was tricky to make it silent!

jieter, 2015-03-04:
Cool!

jieter, 2015-03-21:
care to share the drawings for it? I might want to try this too.

rharrison, 2016-10-09:
Otto can you give us some more details on this as I would like to try this addition on my T962A.

otto44, 2016-10-09:
OK, sorry I got sidetracked. I will do something about it shortly.

John-Nagle, 2016-12-28:
Good idea, as long as it's not too hot there for the fan motor.

radensb, 2018-01-21:
@otto44 ,
Good work on the recirculating air mod. I am interested in doing something like this. Can you provide more info?
1) Does your design pull air up into the propeller blades from the drawer or push it down towards the drawer (ie: which direction does the air flow?)
2) Reading your post, you mentioned that you seemed to get even heat distribution to all corners of the drawer for corner-to-corner reflow success. This is very interesting! Have you performed any other tests that show this result?

Even heat distribution still seems like an outstanding issue with these ovens, even with the new FW, that your solution apparently solved very well. I am surprised that there is not more discussion on this topic.

otto44, 2018-01-22:
​Hello, sorry, I have been ​away from the subject for a while. I had a
health issue, but now I am back... The air is drawn at the top and blown
sideways. Occasional use proved, that it has improved the overall
temperature distribution.
I am preparing kit for the mod. I had a technical issue with mounting fan
to the motor shaft, but I think I have solution now.
It is based on the original concept as shown on pics from 2015. I have
mounted motor above the chamber, and motor is spinning the fan via drilled
hole in the roof of the chamber. The mod. involves removing some thermal
insulation at the top of the chamber, drilling three holes in the roof and
mounting the motor, mounting plate and fan baffle on standoffs and
connecting the motor to DC. I am having the SS parts laser cut now.

radensb, 2018-01-22:
Glad to hear you are doing better.

> I am preparing kit for the mod.

A kit would be neat to see. I see a lot of value in implementing air circulation in these ovens. After trying to find similar parts (mainly that tiny fan blade!) I have started work on my own design that was inspired by yours using parts from a convection toaster oven I got from Goodwill for under $10. The fan motor runs off AC, so I was thinking of wiring it to the main power switch and variable speed controller so, like your setup, it is always running when power is on. Similar to your design (which I am glad you could confirm for me) the fan is mounted above the chamber and pulls heated air into a shroud and exhausts it back in the chamber to the right and left side. I have spend A LOT of time thinking about placement, airflow, etc. 

I have the T-962A, which I got because I know it would give me more room to work with when modding. The A unit has 4 heating elements with thermocouples in between the first and second and second and third elements. Unlike the T-962, the T-962A unit has the thermocouples in the center of the heat chamber, not one on each side which makes it difficult to place the fan in the center of the heat chamber. I am no thermodynamics expert, but I have settled on placing the fan between the third and fourth elements in the back. That way, I think it will pull air from the third and fourth elements and push it to the left and right side across the elements, which is close to how other convection ovens work. I will have to remove the thermal insulation to place the shroud and try to use the removed material to wrap the top of the shroud. I think its as close as I can get. I hope to start implementing this design in the next week or so.

otto44, 2018-01-22:
I have the small sibling, T-962, so I cannot comment on modifying the A -
version. There is not much room in-between and above heating elements,
that's why I made it quite small. And small DC motor is easier to control
with a pot rather than larger AC motor. Good luck!

mrguen, 2019-03-02:
Hello,

Since I plan to produce a couple of boards in panels and because I saw already on little boards soldering that the solder paste did not even melt in some places I decided to follow the mod for the T-962. I upgraded to 0.5.2 and added the DS18B20.

This morning I ran tests. I used an external TC connected to my multimeter. This apparatus is cheap and not very reliable but the results are so different from expected that obviously the T-962 cannot be used to solder middle to large parts. I could see that

·	Temperature is as much as 40°C lower on the left part as compared to the right part. It increases along the left-right axis, with an average value in the middle.
·	Temperature reported is indeed around 40°C higher than in real.
·	Many people thought about changing the TCs. This is not the problem. The temp gradient is real.

So I had to try to compensate and used the maximum offset possible (-25°c). The temp is still over rated. Changing Fan = 8 instead of 0 has approximately no influence. It seems absolutely necessary to find a way to distribute the heat. 

I found your interesting mod. @otto44 could you precise 1) if you made precise measurements (bottom, middle, back, left, center, right) of temperature when using the vent mounted on the top 2) If it works well, do you have a kit available to distribute.

Thanks.

mikeanton, 2019-03-03:
Where are you measuring the temperature when doing these tests?  Are you measuring the temperature of the PCB?  If so, having a large difference in temperature compared to the air temperature would be expected.

I find that raising the PCB off of the metal base helps a great deal in achieving faster heating and cooling.

Using a fan setting of 8 is probably too low.  I think I use something like 40 or so.  The fan value is out of 255, so I'm surprised that a setting of 8 even turns the fan.

mrguen, 2019-03-03:
The Fan 8 setting was the default after upgrading to 0.5.2. So I supposed this was the best choice. I tried 16 and found no big difference (in fact more the contrary, that is does increase the unbalance). I should try 40. 

I used the TC of a multimeter that I passed from under the tray and blocked it at the corner of the middle right / middle left / hole in the tray. The Tc was just at the surface of the tray. I could check visually its place.  Since these measures show a temperature much lower on the left than on the right, in coherence with the T962 measures shown on the screen I concluded the T962 TCs are not faulty. The temp gradient is huge while ramping up. It reduces a little after a couple of minutes at the same temp (it takes a long time). I suppose this is the heat that is spreading on the metal tray. 

I also used the same method but this time pushed the external TC higher up above the tray at about 1 cm and measured a temperture of about 10°c more than at the tray level. So it is very possible that the TC's temp is between 10 and 20°c above the temp at the tray level. That seems logical since the heating elements are on the top. 

The overall 40°c over estimate seems enormous so I was even wondering if the new version 0.5.2 (2018) is faulty?

mikeanton, 2019-03-03:
I'm not sure why you would care what the tray temperature is.  Put a PCB on the tray, or on standoffs (this is what I do), and measure the temperature on a pad on the PCB.  You will find that this is radically different than what the oven is programmed for, but this is to be expected.  Different board colors, and layout will behave differently.  That is why you must profile each PCB design, to get it to have the appropriate soldering profile as seen by the board.

That said, I have a T962A, which is a different beast, and doesn't seem to suffer from large differences in temperature.  I'm running the thermocouple add on board, rather than just doing the cold temperature mod.  With this done, the two thermocouples measuring the air temperature, only differ by around 5C or so, but I'm not sure if that is because I'm using a different oven, or because the thermocouple input board behaves differently.

mrguen, 2019-03-03:
Well, I don't consider using a T962A. The T962 surface is good for what I want to do. As fas as I know T962A has 4 heating elements. Maybe their physical placement explains a more homogenous temp.

Now the idea is to try to reduce the temp difference of T962 to around the same (+/- 5°c). 

Does the fan mounted on top gives this result? Is it possible to use reflective materials ?

mrguen, 2019-03-04:
Today I tested the temperature ramp after trying to block some holes in the tray with the hope that air would flow from the right part to the exhaust and not from the left part. Results are

1) Very repeatable temperature difference between left and right for all the test I conducted

2) No effect whatsoever of blocking the air flow in the holes when I blocked the 2 left columns as well as 3 columns and 4 columns out of 5.

3) No effect when I tried Fan = 8 and Fan = 40. Even a little worse with Fan =40.

Here is the average values of all tests

Fixed Temp| Reported Left Tc | Reported Right Tc

75 | 67 | 84
95 | 83 | 109
115 | 100 | 130
135 | 119 | 150
155 | 134 | 176
175 | 157 | 188