UnifiedEngineering/T-962-improvements #58: Firmware wish list

otto44, 2015-02-24:
Just playing with the oven at this stage, trying to calibrate it. I would like to ask Mr. Werner Johansson of Unified Engineering, if he could squeeze it in, if he think it's practical. 
1. to display along the graph max. temperature reached during the cycle
2. leaving the display with the graph on until a button is pressed, e.g. "S" as the display clears after the bleep and goes to the menu selection.
Many thank in advance!

xnk, 2015-02-25:
Interesting ideas! I've thought about the end of cycle behavior a bit myself, but as we now have superior graphing on the PC-side it's now much less of a problem. We could implement an option that when activated would keep the reflow UI on the screen after completion and exit when you hit S. At that point the maximum temperature could be shown as well somewhere

Again, if you want the data it's already available over the serial port and @jieter has even put together code to control the oven from the PC.

otto44, 2015-02-25:
Thanks, I think it would be nice, if you implement it, just for the quick check what went on, without tying up the PC.

xnk, 2015-02-25:
Let's keep this issue open for now. I'm very busy with some other work at the moment but this should be a simple addition.

otto44, 2015-02-26:
OK, whenever, no hurry. Your day job is more important!

NicholasTracy, 2018-02-13:
I would like to second this request, I have yet to do any permanent serial mods and really don't have any intent to tether it to a PC as it sits in my shop. This would be very useful for setting up temp profiles and gauging how close the temps came to their target, even if its not accurate with the LCD its still helpful for those of us not on board with using the serial protocols.

xnk, 2018-02-13:
Definitely a valid point!

NicholasTracy, 2018-02-13:
I'd be happy to take a look at it in my own fork if you would be so kind as to point me in the right direction of where to look! Can't imagine it would be too hard.
-Remove menu change at the end of timer
-Show some data at the end of timer (Max Temp, Total Time).
-Add wait for select button press to go to main menu screen (Display S icon and  maybe a label for "exit" or something similar).