UnifiedEngineering/T-962-improvements #60: X1 and X2 temperatures neglected for actual temp display + findings

wysiwywg, 2015-02-26:
I noticed X1 and X2 (from the MAX31850K interface) don't seem to be used for calculating the actual temperature shown in baking mode (shows 101.5C instead of 106C:

[image]

I can't say if X1/X2 are used when baking.
The X1/X2  temperature however seems to be taken into account when reflowing, here some tests I did. I ran every profile configuration several times with empty oven and minimum fan=12 and the graphs for each configuration are always similar:

L+R only used (L+R reversed in this picture) :
[image]

X1 next to L:
[image]

X1 between L+R but closer to the pcbs:
[image]

X1 next to R:
[image]

Does anyone have similar differences between L+R when the fan is on?
I just measured the distance from the tip of the sensors to the tray walls and they differ by 2cm with the right sensor being closer to the wall which would make sense as it takes longer to cool down (always has higher temperature when fan is on).
This could mean that the sensor placement is quite critical and I should readjust them. I will do some tests on replacing them now.

wysiwywg, 2015-02-26:
After repositioning R symmetrical to L it looks the same I would say:
[image]
That's good, so the difference before doesn't matter.

Then I looked at the csv files and the actual temperature shown there also is only calculated from L+R.
The temperature curve plot with X1 mounted 1cm above a pcb in the middle of the tray looks promising:
[image]

xnk, 2015-02-26:
Very true, X1 and X2 are for measurement/validation only in the current firmware. If you take a look at the MAXTEMPOVERRIDE define here https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/reflow.c#L36-44 then you might get closer to the result you're after. I haven't come up with a good way of configuring this without recompiling yet. There are quite a few scenarios to take care of, unplugged external sensors being one, external sensors left outside the oven being another. Any ideas?

It seems that a lot of these ovens have a cooler left side, especially when running the fan, and that temperature difference will clearly prevent proper reflowing. Could the airflow be redirected upwards perhaps?

wysiwywg, 2015-02-27:
I will try with MAXTEMPOVERRIDE, though I think I will need to set a temperature offset for the whole curve, as it will result in a even lower overall temperature at my current configuration? (if I understood the funtion right). When using the external sensors the TC offset in the setup menu doesn't have any effect. Propably good to change that?
I guess left outside of the oven can be detected very early in the soldering process due to no temperature response? Doesn't the firmware already detect missing sensors?
Toggling sensors
Probably add such test functions with a debug mode you enter starting the oven holding down"S"?

I did a few tests with the 63Sn/37Pb profile measuring X1 at  different corners of the oven (3cm from the tray walls 1cm above surface). Measuring points were like this

1________2
 ____3____
4________5

(1)
[image]
(2)
[image]
(3)
[image]
(4)
[image]
(5)
[image]

This correlates to my experience pcbs in the middle reflowing fine, to the right maybe and to the left not. With standard settings that is.  Is Bismuth paste the way to go for sensitive components/reliable reflowing then?
I got really good results with chinese Bi/Sn paste, but don't dare shipping out a hundred Bi/Sn soldered pcbs yet.

wysiwywg, 2015-02-27:
setting the fan to 0 definitely helps to move all curves a bit up here the curve at pos 5 with fan=0:
[image]
and pos3 with fan=0:
[image]

Then I took your idea of redirecting the airflow and installed an 32mmx85mm obstacle at the back of the tray:
[image]

pos 5 with fan=12 and obstacle1:
[image]
and pos3 with fan=12 and obstacle (tejp went off during reflowing, so sensor is a bit higher:
[image]
Now that looks promising:-) Good idea you had there!

Having X1 and X2 plotted in serial-control.py would be neat.

electronicamystica, 2015-03-01:
new test with measuring points placed like this:
A B C
D E F
G H I
first picture always with obstacle, second without.
A
[image]
[image]

B
[image]
[image]

C
[image]
[image]

D
[image]
[image]

E
[image]
[image]

F
[image]
[image]

G
[image]
[image]
H
[image]
[image]
I
[image]
[image]

I can't discover any deeper wisdom here apart from the simple obstacle not being a solution either and pcb color isn't taken into account either. I usually use black pcbs which should get warmer then green/white ones.
Wenn a preheated oven definitely helps it seems as i did the last set of tests without obstacle shortly after another.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk