UnifiedEngineering/T-962-improvements #71: TC gain and offset values don't seem to do anything

rogerclarkmelbourne, 2015-10-17:
Hi,
I'm using the latest version 0.5.1 and I've wired in the cold junction compensation.

And, the TC gain and offset inputs on the setup screen don't seem to change the displayed values, either the "Oven temperature" reading on the display or the output on the debug.

If I use the 'about' command I get these values

Initializing improved reflow oven...
Part number: LPC2134
EEPROM contents:
0x0000: ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 64
0x0010: 00 00 00 00 00 64 00 00 00 00 00 00 00 00 00 00
0x0020: 00 00 00 c8 00 00 00 00 00 00 00 00 00 00 00 00
0x0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0x0040: 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0x0050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0x0060: ff ff 4a 57 09 0a 03 64 64 64 64 09 00 1e ff ff
0x0070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0x0080: 56 57 00 16 00 23 00 32 00 41 00 50 00 5f 00 6e
0x0090: 00 7d 00 8c 00 9b 00 9b 00 9b 00 a5 00 a5 00 a5
0x00a0: 00 a5 00 a5 00 a5 00 aa 00 b9 00 c8 00 d7 00 e6
0x00b0: 00 f5 00 f6 00 f5 00 e6 00 d7 00 c8 00 b9 00 aa
0x00c0: 00 9b 00 8c 00 7d 00 6e 00 5f 00 50 00 41 00 00
0x00d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0x00e0: 00 00 00 00 31 6a 00 00 3b 5d 00 00 00 11 00 00
0x00f0: 07 6f 00 00 40 00 00 00 16 00 00 00 07 00 58 59
Sensor values:

         Left: 54.2degC
        Right: 22.6degC
Cold junction: 22.0degC

So it looks like my sensors are not calibrated.

So I went into the setup screen and tried changing the Left and Right TC offsets, however it didn't make any difference to the values that were output to the debug terminal.

I know both thermocouples are working, because if I heat the oven for a 30 secs and check get 'values', the TC values change.

         Left: 88.7degC
        Right: 75.8degC
Cold junction: 22.0deg

I suspect that somehow I've stuffed the original factor calibration values, as one of my attempts to flash the firmware failed, but I'd like to be able to recalibrate and I presume its best to NOT mess around with the trimpots, and to just adjust the TC calibration values in the firmware ?

rogerclarkmelbourne, 2015-10-17:
Update.
I rebooted after selecting -25 for the Left TC and now the left TC seems to be reading 25 deg lower.

It seems odd that I need to turn off and on again to action the TC cal settings

BTW. I'm also not sure why I'd need to set one TC offset to its max neg value - perhaps bad initial analog calibration or a fault on the board or the TC :-(

thopman, 2016-01-18:
Hi Roger,

i have the exact same issue with the thermocouples. 
Left = 50degC
Right = 22.5degC
cold junction = 22.5degC 
(fluke with thermocouple measures 21.8degC) 

trimpots have no effect what so ever. If i swap the thermocouples on the terminals the left one stays at 50degC and the left at 22.5degC. Somehow i think the offset with the left thermocouple is hard coded in the firmware. Don't see any differing values on the pcb when comparing both channels. 

i actually haven't tried the oven yet since i just got it in and was still playing with the mods, so it's nice to see i can compensate in the firmware. 

greetings,

Thomas

xnk, 2016-01-18:
Just so you know: The onboard opamp isn’t very precise when approaching zero Volts measurements, so I’m not really surprised that the readings are a bit off at room temperature. Calibration and offset adjustments should _not_ be done at room temperature but with temperatures closer to your reflow temps (as that’s the temperature that really matters). It’s been a while since I had TCs connected to the internal opamps, but it sounds odd that the calibration does not “take” until after a reboot, I must have messed something up. 

I don’t have the code right in front of me right now (out on a business trip), but will hopefully be able to look into that in the coming week. I know I have been away from the project for a while, I simply haven’t had time to work on it. Hopefully this will change soon.

/wj

rogerclarkmelbourne, 2016-01-18:
I had to put things on hold while I bought some new thermocouples, because even using external Maxim chip thermocouple interfaces I could not get the probes to read consistent results.

So i suspect that thermocouples appear to be faulty as well as possibly the opamp circuit being faulty.

I now have a selection of different thermocouples to try, but have not got around to testing them.

Its hard to know if the problem is that the internal heat distribution is terrible or whether the thermocouple is defective. Possibly both.
As I tried manually heating the oven using a hot air paint stripper, which blew air all around inside the oven, and still did not get consistent results.

But at the moment I dont have time to work out why the oven is working so badly.

thopman, 2016-01-18:
Well at least i'm able to confirm that the offset and gain vaiues get updated without needing a restart. i played around with it a bit tonight and it's working quite ok. my fluke measures a bit of overshoot but nothing i won't be able to compensate for in the coming days. Also i thought the system fan mod was optional and after sitting next to the thing for a couple hours i can confirm it is not optional. lol Gonna order a silent fan tomorrow and do the pwm mod. Have some boards i want to try with it as well with some big inductors as well as smaller parts so should be interesting.

rogerclarkmelbourne, 2016-01-18:
@thopman 

I didn't have much luck soldering anything with the oven so far.

I think I posted this information on Hackaday, but I'll post it here as well...

I bought some of those Maxim 31855 thermocouple to SPI interface IC's (actually I bought modules from eBay), but the firmware doesnt support SPI (I think the SPI pins on the mircorprocessor are used for something else).
So the workaround is to use an I2C to SPI bridge IC.

I didnt have any of those IC's so I wrote an Arduino sketch to run on a little Arduino Pro mini board, which acts as an I2C slave and interfaces between my 2 MAX31855's and the oven.

But I had some issues with the Arduino I2C slave code as the library has a bug, and you have to modify the I2C library so that slave mode actually works.

But I then had the issue that the Arduino bootloader does not finish before the firmware in the oven had checked for I2C and not found any.

So I had to modify the oven firmware, and add a delay so that it doesn't search for I2C devices until about 1 second after startup.

I could of course probably have run a bare metal Arduino without the bootloader but it was easier just to add a delay into the oven firmware.

One note if you want to add a delay. There is a function in the firmware that appears to be a delay, but it doesnt work because it doesnt feed the watch dog timer, and hence it times out and reboots the code. So I had to add a few delay function which kept feeding the WDT (this isnt ideal as it could crash in the delay and would not be caught by the WDT)

Anyway, after all of these changes and putting in the MAX31855 interfaces etc, I still got wildly different readings in the two thermocouples.

So I ordered 5 new thermocouples from eBay

However I have not had time to get back to fixing this (nightmare)

I have a suspicion that it may never be possible to get heat distribution inside the oven without fitting a fan inside to swirl the air around, as I don't think that simply pulsing the fan on the back of the oven really causes enough air movement.

xnk, 2018-02-05:
An Arduino Pro mini as I2C-SPI bridge chip emulator sounds like a very interesting hack! Feel free to add info to the wiki if still relevant.

closed 2018-02-05 by xnk