UnifiedEngineering/T-962-improvements #72: Manual bake - incorrect display after returning from main menu

rogerclarkmelbourne, 2015-10-17:
If you enter Manual Bake

Set a temperature (e.g. 100 deg)
Press F3 to Select INF (I presume this means infinite)
(Temperature starts to rise)
Press S to exit to main menu (Do not press F4 to stop)
(Bake is stopped)
(Main menu is displayed)
Press F3 to enter Manual bake
(Note screen is displayed as if the bake is running in INF mode, however in reality its in STOP mode)

jieter, 2015-10-20:
Hmm, thanks for reporting. I hope to find some time in the coming weeks to address some issues with the bake functionality. I didn't like some aspects while using it too.

rogerclarkmelbourne, 2015-10-20:
No worries

I actually need to make another customization for my own use, so I've forked your code.

If I get chance I may take a look at fixing this problem myself, it doesn't look too hard to fix ;-)

BTW.

I needed to connect some MAX31855's but I didnt have the SPI to I2C bridge, so I've written some Arduino code to do the same job, and I'm now using an Arduino Pro mini (3.3V) instead of the bridge IC

The only drawback is that the bootloader in the Arduino takes about a second to run, so I will need to delay the I2C scanning function by about a second 
(Hence I'll need to customize the firmware for my own use)

I'll post the Arduino code when I its all working.

Stew1234, 2015-11-04:
If the manual bake is updated, will it be possible to add the last temperature that was set is stored in eeprom and reloaded by default ? We use the bake feature a lot, it's allowed us to set glue and such, but we always use the same temperature for bake.

rogerclarkmelbourne, 2015-11-04:
I have added the MAX31855s to my oven, and had to add a new delay function during startup to wait for the Arduino mini that I'm using as an I2C slave to get out if its bootloader.

But unfortunately I've not got more issues with the termocouples in the oven. I suspect one of them is defective, because even with using a MAX31855 one reads a lot higher than the other.

I did consider if this was just hotspots caused by the heating tube, but I did some more tests using a hot air paint stripper gun, to blow air into the front of the oven (drawer) at over 100 deg, and I still get stupid readings, i.e very inconsistent or incorrect.

So I'm now waiting for more thermocouples to arrive from eBay, before I can continue fixing it.

Hence I'm not doing any more SW updates until I can get the hardware working, otherwise I don't have a baseline to test from.

rogerclarkmelbourne, 2016-02-21:
@ Stew1234 

Do you still want this feature ?

I think the best way I can do it, is to add it as a setting in the NV storage, i.e put it into the settings.

So if you enter "bake" with no value it uses the one from the settings.

I could probably also get it to store the value from bake xxx into the NV storage

BTW. I've just made a load of changes to my fork of the FW, because I found that the TC offset and gains were only being applied to the ADC TC sensors not to the external MAX31855 based TC's, but I tried 5 different TC's (from eBay) and none of them read the same values unless I can apply gain and offset correction.

Also, I found that the "settings" values were not immediately being applied to the gain and offset (arrays), in sensor.c, so I've fixed that issue as well ;-)

rogerclarkmelbourne, 2016-02-21:
BTW. I have now fixed this issue in my fork of the repo.

timer needs to be set to zero when the S (abort) key is pressed

i.e
```
			// Abort bake
			if (keyspressed & KEY_S) {
				printf("\nEnd bake mode by keypress\n");

				mode = MAIN_HOME;
				Reflow_SetBakeTimer(0);
				Reflow_SetMode(REFLOW_STANDBY);
				retval = 0; // Force immediate refresh
				timer = 0;// Clear the timer, so that its zero when we re-enter this screen.
			}
```

closed 2016-02-21 by rogerclarkmelbourne