UnifiedEngineering/T-962-improvements #73: T962 Fan control and DS18B20 cold-junction wiring ?

hallard, 2015-10-20:
Hi There,
I just received this morning my T962 and of course will try to mod and upgrade the firmware but I have 2 questions that may be some of you may clarify :
- For DS18B20 cold-junction, looking at the picture, it's connected in parasitic mode (no VCC), right ?
- since it's a T962 with Fan controlled by opto-coupler/triac (no SSR) and output pin, is the firmware is able to manage FAN during heat or do I really need to use the other fix with GPIO AD0 and control fan with transistor ?

Thank you very much for any explanation.

jieter, 2015-10-20:
Yes, parasitic mode:
[image]

The oven fan is controlled by default. If you want to make the device silent if not warm, you can apply the [mod described in this wiki page](https://github.com/UnifiedEngineering/T-962-improvements/wiki/System-fan-control)
[image]

hallard, 2015-10-20:
@jieter 
Thanks for your answer.
Just upgraded FW  ;-)
I just realized that FAN control is not for the big fan on rear during reflow but the little noisy one to refresh PCB. Sounds good using DS18B20 also to detect and activate this one ;-) 

Just for information, you don't need to unground n_ISP to upgrade firmware (at least on my T962), this mean that simplify FW upgrade with a FTDI: 
Just wire n_ISP to ground and connect DTR of any FTDI 3V3 USB to nRESET and start upgrade, that should be straightforward. Of course after flash, unwire n_ISP

Next step, basic FET transistor mode for FAN, got some BS270, simplest way, cut the GND wire (the dot dot dot one, not the dot line dot) of the fan. Then connect drain and source and then gate direct onto PCB (well it should have been like this)
[image]

Unfortunately I cut the wire too short, needed to solder 3rd on the FET gate, may be too long day, anyway, final wiring as follow:
[image]    

[image]

Everything sounds good, may be a test fan connected to this AD0 output on firmware would be great so see if everything is well connected ;-)

Really thank for this marvelous firmware, amazing and let so much space on target that would allow much more mods !!!!

xnk, 2018-02-05:
Just a note on the system fan testing issue: The firmware will turn that fan on briefly during boot which makes modification verification simple

closed 2018-02-05 by xnk

gongfan99, 2020-06-20:
According to the DS18B20 datasheet, a FET is needed in the parasite power mode. Why is there not a FET in your design?
[image]

xnk, 2020-06-20:
Hi Sam,

There’s no need for an external FET, that functionality is handled by this line of code using the internal GPIO pass element:

https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/onewire.c#L294 <https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/onewire.c#L294>

/wj

gongfan99, 2020-06-22:
Thanks for the clarification, Werner! My T-962 will arrive tomorrow. I cannot wait to try the mod out.

southern-boy, 2020-06-22:
> image

Why wouldn't any thermocouple work here? Does it have to be a one wire 
device?

Thank you.

Jim Pruitt

mikeanton, 2020-06-22:
@southern-boy, because a thermocouple requires that the cold junction temperature be measured, so it can be compensated for, and that is what the one wire sensor is providing.  This is what the oven is missing from the factory, which causes the oven temperature to drift, as the cold junction temperature changes.  A thermocouple measures the difference between the thermocouple junctions.  One of the junctions is in the oven, and the second one is the cold junction that is formed where the thermocouple is attached to the PCB (every set of dissimilar metals is considered a thermocouple junction).  If the temperature of the cold junction is known, and never changes, then technically you wouldn't need to measure it.  Alas, this is not the case, and the electronics chamber where the cold junction is located, heats up as the oven is used.

Now, if instead you use the 4 channel thermocouple interface board, it has the cold junction compensation built in, so in that case the DS18B20 is not required.  This is the route I took when modifying my oven, and it does indeed work very well.

xnk, 2020-06-22:
I can only second that idea, the built-in ADC front-end is really bad. The 1-wire thermocouple interface is a much better design, but by all means give the built-ins a go first!