UnifiedEngineering/T-962-improvements #74: Cold Junction DS18B20 not detected

chipemulator, 2015-11-12:
Hello 
I received today my T962A. I installed firmware 0.51 using FTDI MM232R without any problems.
After that, i installed DS18B20  following instructions, but it´s DS18B20 is not detected by firmware. I have always the message "no cold-junction TS".
As i work normally with onewire devices, i removed DS18B20 and i connected it to my PC using "onewire viewer" software and the device is working showing 21.6 ºC (room temperature).
I soldered it again to T962A PCB and i started to measure the voltages on the pins. on Pin 1 and 3 i have 0V and on pin 2 i have 3.3V. This means the 4.7K pullup resistor is ok and gnd connection is ok.
There is good continuity betwen Pin2 (Dq) and LPC2134 pin31 (po.7). On this pin i have 3.3V (correct voltage driven by 4k7 puul up resistor).

On oven  boot the firmware detects LPC2134(/01/) revision F. 
Oven PCb is labeled as T962-2 New2 
The production date of oven is 15.05

Anyone can help me to identify cause of problem?

thanks in advance

chipemulator, 2015-11-12:
Hello
I found cause of problem. I was using DS18S20 instead of DS18B20.
I will buy DS18B20 it will work for sure.

Thanks anyway

dknuth0101, 2016-07-01:
Hello, I'm having the same problem, and after seeing your comment I checked and I also have the incorrect part number! thank you for posting!

anyone know if the DS18S20 could work?

dknuth0101, 2016-07-26:
@chipemulator Hey, I got it working with DS18S20. Until the pull request goes through, you can find it [here](https://github.com/bitfab-technologies/T-962-improvements). Look in ~/src/onewire.c, line 97 defines a family ID for the DS18S20, and lines 352 & 358 are where it's used. Basically, your device was working, but it was sending back a family ID which the program wasn't handling. They're almost the same sensor in every other regard, though. hope this helps. :)

xnk, 2018-02-05:
PR (finally) merged

closed 2018-02-05 by xnk