UnifiedEngineering/T-962-improvements #76: Maintaining a list of T962/T962A problems, improvements and suggestions how to get good results

beyon, 2015-12-12:
Hello, I recently ordered a T962A and expecting to receive it next week. I was prepared to take it apart and apply the most critical fixes like replacing masking tape and changing the firmware and so on. However the more I looked around the web the more potential problems I found that even if not critical it seems like they would be good to address to avoid future problems or improve the oven.

I think it would be a good idea to add to the wiki a list of all known problems (with info about which version it applies) as well as proven improvements along with ideas of improvements. The idea would be to collect as much info as possible in one place and reduce the need to trawl the web (HaD thread, various sites, youtube) to know what you get into when deciding to buy one and what you can do to make it better if you have one.

Since I don't have my T962A I don't want to start a wiki page just yet but I'll list my findings and it would be good to hear from those that have had the ovens for a while which issues are worth worrying about and any tips to get good results.

## Firmware and build problems:
* Original firmware is not very good
* Masking tape should be removed/replaced with kapton tape before use
* Hot glue everywhere, not sure how much of a problem it is outside the oven chamber
* Ground wire might not be properly connected to chassis
* Drawer
  * Not tight fit allowing hot air to escape? [link](http://www.briandorey.com/post/T-962-Infrared-Reflow-Oven-Vent-Hack.aspx)
  * Hot glue inside front panel that will start melting, common or just on some units? [youtube](https://www.youtube.com/watch?v=e31WZFBAMuY)
  * Window glass not made for heat? Actual problem or just one person's theoretical conclusion? [link](https://www.youtube.com/watch?v=rznKTOkcvcc)
  * Plastic parts used in drawer slides, can they handle the heat?

## Potential improvements
* Improving isolation/reducing thermal mass with heat reflective tape/sheets and isolation in selected places
* Adding bottom heating element to get more even heat and maybe quicker temperature rise. Might be hard due to limited space and moving parts (drawer)
* Convection fan for more even heat. Like otto44 seems to have done: #57 Maybe use existing convection oven replacement parts.
* Wire quartz tubes individually and potentially control them individually. This comment on youtube claims having multiple tubes in series may lead to uneven stress [link](https://www.youtube.com/watch?v=CE6wjQPuxFE)

## Suggested upgrade paths/steps
1. Minimum
  * Replace masking tape
  * Fix ground connection
2. Good start
  * 1
  * Change firmware
  * Install thermocouple cold junction sensor
  * Fan control
3. Better?
  * Switch to digital thermocouple readings with add-on board
4. ???

I really appreciate all the work you've done and I hope we can work together on collecting useful T962/T962A reflow oven information.

beyon, 2015-12-12:
## How to get good reflow results
* Preheat oven?
* Avoid edges of PCB tray
* Lift PCB up to allow air to circulate under it by laying it on junk PCBs?

colinoflynn, 2015-12-12:
Hi Beyon,

 

Fixing the masking tape, firmware, and grounding wire are required fixes (with the firmware this also means adding the proper temperature reference part).

 

The other fixes you mention aren’t required, but can be done to improve performance/usability. I’ve used this oven for the past 1-2 years with only the basic fixes and reflowed a number of boards, including BGA (see http://colinoflynn.com/2015/10/low-cost-smd-soldering-setup/). I haven’t seen issues w.r.t. temperature issues such as glass, plastic parts melting, etc. Basically the oven performs far better than I expected for a $350 oven. It’s not build to the standards a $5k-$15k unit you could reliably do small productions runs in and/or can be used to reliably meet specific soldering standards.

 

I did build a fume hood around it, which I also use when cleaning the boards with iso. That was a worthwhile investment, and the link you have about sealing the bottom side would also be a good solution. I don’t think it’s needed to seal the bottom-front as they have done unless you are building the vent – your oven is blowing hot air out the bottom vent, I don’t see why it matters it’s happening 6 inches ahead of that too? You shouldn’t be using it on a heat-sensitive surface or you’re going to have a bad time.

 

I also wear safety glasses when looking into the front glass, which is generally a good idea anywhere you have stressed glass (yet alone somewhat unknown quality glass which may or may not have been treated properly).

 

I added a white LED light on the outside shining in through the glass which helps view what is going on inside the oven.

 

The other “fix” is I found I needed a piece of masking tape to keep the drawer closed. Once the cool-down cycle starts it might slide open a bit from air pressure. Might also be my table isn’t level.

 

Regards,

 

  -Colin

xnk, 2018-02-05:
Issue cleanup. As always feel free to add info to the Wiki

closed 2018-02-05 by xnk