UnifiedEngineering/T-962-improvements #77: Reboot when Lamp turns on

michielbrink, 2016-01-09:
First of all, Thanks for this awesome firmware.
When I flash the v0.5.1 pre release, everything works. 

When I compile the V0.5.1 branch and flash the firmware (no errors when compiling) it looks like everything works. But when the t-962 wants to turn the Lamp on, the micro-controller resets itself and boot again.

Is it a problem with my compiler? Or is there a bug in the source code?
If I compile the firmware the .hex looks different then the pre release hex.
This is my make log: http://pastebin.com/A5r4CGJw

zittix, 2016-01-11:
I had the same problem. You have to use the 4.9 branch of gcc for it to work - https://launchpad.net/gcc-arm-embedded/+download

I was using the 5.2 branch but this create the bug you're referring too.

xnk, 2018-02-05:
The firmware is now built with the gcc-arm-embedded 7.2.1 from 20170904, this issue is not reproducible at least in automated builds.

closed 2018-02-05 by xnk