UnifiedEngineering/T-962-improvements #78: Failed to autobaud. (Operation failed)

rs98, 2016-01-15:


RasmusB, 2016-01-15:
Have you tried with the settings suggested in the Wiki? There is a suggested baud rate there, so you should not need to autobaud anything. 
https://github.com/UnifiedEngineering/T-962-improvements/wiki/Flashing-the-LPC21xx-controller

rs98, 2016-01-15:
Of course I followed suggestions in the wiki (including suggested baud rate 57600):
https://github.com/UnifiedEngineering/T-962-improvements/wiki/Flashing-the-LPC21xx-controller
Even so Flash Magic is trying (and failed) to autobaud and LPC2000 says "Cannot communicate with test board". 
I also made a new serial cable and also tried optional DTR and RTS control of RST and ISP pins.
Is there any requirements for waveform / timing of n_RESET and n_ISP pulses?

RasmusB, 2016-01-15:
No special requirements, once you have entered the bootloader you shouldn't need to touch RESET or ISP before flashing. I just did it with simple push buttons without problems.

rs98, 2016-01-15:
What was your COM port settings? 
My are (defaults) 57600 bps, 8 data bits, parity: none, 1 stop bit. Flow control: none. I also tried Flow control: Xon/Xoff.

sapperdaddy, 2016-01-22:
I had the same issue. Turned out I had the Rx and Tx backwards.

piyushverma, 2016-10-29:
I am using linux ./lpc21isp build/T-962-controller.hex /dev/ttyUSB0 57600 11059 to flash but facing similar problem. 

any Suggestion ?

I have not used LPC21... So I thought to buy sample board and programmer to test first then flash in this.

I believe any USB -> RS232 dongle should do it like pL2303 please suggestion how to fix it

rharrison, 2016-11-17:
I spent 4 hours trying to flash the oven with flash magic. The solution in the end was to set the baud rate to 14400 and everything worked ok after that.

pandaritrl, 2016-12-05:
It worked at 2400 baud rate for me.

Mavrikant, 2017-03-17:
I had the same problem. I used 9600 baudrate. And also removed reset cable after entering bootloader mode because when I hit "Read Device Signature" option, device resets itself and gets out of mode.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk