UnifiedEngineering/T-962-improvements #79: v0.5.1 firmware operation WITHOUT the temp hardware mods?

brianjaod, 2016-01-18:
Howdy, great work everyone!  I've just uploaded the latest firmware to my T-962 (small), but don't have time yet to implement the hardware mods. Can the oven, with this latest firmware, still do a decent job at reflowing leaded boards?  I just need to pump out a couple of quick prototypes and then I'll circle back to doing the mods later. The serial output stream is showing a ColdJ value of zero (probably because I haven't done the mods yet), but is that a problem? In other words, what hardware mods, if any, is this firmware dependent on for "decent" reflow functionality?  Or would I be better off, for the moment, reflashing the stock firmware?  Thanks!

xnk, 2016-01-18:
Hey Brian,

Leaded boards should work reasonably well (and I believe that the custom firmware will do a better job than the stock one no matter what). ColdJ 0 is OK as long as the actual temperatures ,measured are reasonable. If you go to bake mode, does it show Left and Right temperatures in the 25-30C range when oven has been allowed to cool down? I make an assumption that the cold junction is 25C when no method to measure it is present.

/wj

brianjaod, 2016-01-18:
/wj, thanks for the quick response!

On the lcd screen on bake mode (haven't run any reflow yet) it shows a set temp of 30, with an "actual" of about 34.5, but the left temp is reading about 43.5 +/- 0.5 degree, and the right temp is not moving at all, and just shows a rock solid 25 C.  That's probably not good, huh?  Here's the serial output for the first few seconds after power up.  Thoughts?  Thanks again!

```
T-962-controller open source firmware (v0.5.1)

See https://github.com/UnifiedEngineering/T-962-improvement for more details.

Initializing improved reflow oven...
Reset reason(s): [POR][EXTR]
Running on an LPC2134(/01) rev F
Waiting for keys to be released... Done!
Buzzer_Init
ADC_Init called
OneWire_Init called
Scanning 1-wire bus... No devices found!
SPI_TC_Init called
SC18IS602B_Init - No chip found
SystemFan_Init
# Time,  Temp0, Temp1, Temp2, Temp3,  Set,Actual, Heat, Fan,  ColdJ, Mode
   0.0,   25.1,  25.0,   0.0,   0.0,   50,  25.0,    0,   0,    0.0, STANDBY
   0.0,   48.9,  25.0,   0.0,   0.0,   50,  37.0,    0,   0,    0.0, STANDBY
   0.0,   47.6,  25.0,   0.0,   0.0,   50,  36.3,    0,   0,    0.0, STANDBY
   0.0,   46.6,  25.0,   0.0,   0.0,   50,  35.8,    0,   0,    0.0, STANDBY
   0.0,   45.9,  25.0,   0.0,   0.0,   50,  35.5,    0,   0,    0.0, STANDBY
   0.0,   45.4,  25.0,   0.0,   0.0,   50,  35.2,    0,   0,    0.0, STANDBY
   0.0,   45.1,  25.0,   0.0,   0.0,   50,  35.0,    0,   0,    0.0, STANDBY
   0.0,   44.9,  25.0,   0.0,   0.0,   50,  35.0,    0,   0,    0.0, STANDBY
   0.0,   44.8,  25.0,   0.0,   0.0,   50,  34.9,    0,   0,    0.0, STANDBY
   0.0,   44.6,  25.0,   0.0,   0.0,   50,  34.8,    0,   0,    0.0, STANDBY
   0.0,   44.4,  25.0,   0.0,   0.0,   50,  34.7,    0,   0,    0.0, STANDBY
   0.0,   44.3,  25.0,   0.0,   0.0,   50,  34.7,    0,   0,    0.0, STANDBY
   0.0,   44.4,  25.0,   0.0,   0.0,   50,  34.7,    0,   0,    0.0, STANDBY
   0.0,   44.4,  25.0,   0.0,   0.0,   50,  34.7,    0,   0,    0.0, STANDBY
   0.0,   44.2,  25.0,   0.0,   0.0,   50,  34.6,    0,   0,    0.0, STANDBY
   0.0,   44.2,  25.0,   0.0,   0.0,   50,  34.6,    0,   0,    0.0, STANDBY
# Time,  Temp0, Temp1, Temp2, Temp3,  Set,Actual, Heat, Fan,  ColdJ, Mode
   0.0,   44.5,  25.0,   0.0,   0.0,   30,  34.8,    0, 255,    0.0, BAKE
   0.2,   44.2,  25.0,   0.0,   0.0,   30,  34.6,    0,  60,    0.0, BAKE
   0.5,   44.2,  25.0,   0.0,   0.0,   30,  34.6,    0,  99,    0.0, BAKE
   0.8,   44.3,  25.0,   0.0,   0.0,   30,  34.7,    0, 116,    0.0, BAKE
   1.0,   44.1,  25.0,   0.0,   0.0,   30,  34.5,    0,  67,    0.0, BAKE
   1.2,   44.2,  25.0,   0.0,   0.0,   30,  34.6,    0, 124,    0.0, BAKE
   1.5,   44.2,  25.0,   0.0,   0.0,   30,  34.6,    0, 100,    0.0, BAKE
   1.8,   44.1,  25.0,   0.0,   0.0,   30,  34.6,    0,  83,    0.0, BAKE
   2.0,   44.1,  25.0,   0.0,   0.0,   30,  34.5,    0,  90,    0.0, BAKE
   2.2,   44.1,  25.0,   0.0,   0.0,   30,  34.6,    0, 107,    0.0, BAKE
   2.5,   44.3,  25.0,   0.0,   0.0,   30,  34.7,    0, 124,    0.0, BAKE
   2.8,   44.0,  25.0,   0.0,   0.0,   30,  34.5,    0,  59,    0.0, BAKE
   3.0,   44.0,  25.0,   0.0,   0.0,   30,  34.5,    0,  98,    0.0, BAKE
   3.2,   44.2,  25.0,   0.0,   0.0,   30,  34.6,    0, 132,    0.0, BAKE
   3.5,   44.2,  25.0,   0.0,   0.0,   30,  34.6,    0,  92,    0.0, BAKE
   3.8,   43.9,  25.0,   0.0,   0.0,   30,  34.5,    0,  66,    0.0, BAKE
   4.0,   44.1,  25.0,   0.0,   0.0,   30,  34.5,    0, 114,    0.0, BAKE
   4.2,   44.2,  25.0,   0.0,   0.0,   30,  34.6,    0, 124,    0.0, BAKE
```

xnk, 2016-01-18:
It looks reasonable actually. The stock fw just averages the two readings and you get a “pretty ok” value. I wouldn’t count on room temp readings to be anywhere near ok though, because of offset voltage issues with the opamps. The temperature that really matters is the reflow peak temperature that you want to hit reasonably close. If you first run a dummy cycle and then measure the temperature inside when you run a second cycle (the SnPb one most likely) you should hopefully be reasonably close to the actual peak temperature. If you don’t have a DMM with a TC the only way you can test it is by running some test board through and see if it reflows and not burning the board in the process. You should see both TCs start increasing when the oven heats up.

The stock firmware will not give you a more accurate temperature (and there’s no calibration stored anywhere that is used by the stock firmware).

/wj

brianjaod, 2016-01-18:
Yeah, I won't be switching back to the stock fw, no worries there. I'll run the dummy cycle and see what happens.  I have a TC/DMM to measure with.  But why is the one at 44 and the other at 25 but not moving?  Are there settings that I need to tweak to fix that or offset it or something?  Does the 25 not moving mean that TC is faulty/broken?  Thanks!

xnk, 2016-01-18:
Most likely you have a slight negative offset on the one forced to 25C. As the ADC can’t measure negative voltages it will never read a value below the ColdJ temp which will be hard coded to 25C without the temp sensor mod.

Heat the oven and see if both temps start increasing, that’s the only way to know whether the sensor is broken or not. If the temperatures readout does not increase when heating then something’s clearly broken, but a solid 25C readout at room temp isn’t a clear indication.

I wouldn’t mess with the TC offsets or gains at this stage, you need to do measurements at higher temperatures for the adjustments to make sense. I know I wrote something about the intended use of the TC cal stuff somewhere in the issues on github but can’t find it right now.

/wj

brianjaod, 2016-01-18:
/wj I just ran the first dummy cycle (ever) and my DMM showing a peak of about 109 C, even though the serial output was showing a set/actual peak of about 210.  Is this because it's the first cycle and it'll get better on the next one?

Btw, you were right, the other temp was going up during heating too, but both sensors are still about 15 C off.

brianjaod, 2016-01-18:
Yeah, much better.  Just ran two more dummy cycles, and measured (with DMM) a peak of about 212 C, even though the serial output was saying otherwise.  Thanks again!

xnk, 2016-01-18:
Great that you seem to get a reasonable temperature! The oven takes some time to heat up all the internals, and the heating elements just barely keeps up. If your DMM TC was touching the metal on the drawer it will read a lot lower before the drawer heats up.

A quick tip: When reflowing, always try to have your boards to be reflowed “floating” above the drawer, use old (empty) boards as spacers in the corners to lift the board to avoid having the drawer cool the boards.

/wj

KLEYNOD, 2017-10-18:
Hi

One  way  that can verify your TC is that using soldering pen which you  can adjust temperature  (I use pace soldering machine )  to 150C then put it on TC to see if it show right temperature.  Verify it first with external DMM ( I use  pico   TC-08
https://www.picotech.com/data-logger/tc-08/thermocouple-data-logger )
Another things that could be verified is the op amp 
There are many version of this op amp which has different offset voltage. Make sure that you have one with low as offset . Myself changed them to better one so called zero drift . Like opa333 .
They have almost 0 V offset ,very very low .
In this way you make sure that opamp doesn't put any dc out when input is zero.
If still show L and R different then I think there is bug in FW.
Otherwise how is possible to get different temperature on L and R in initial condition . I mean not even running preheat.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk