UnifiedEngineering/T-962-improvements #83: Unable to build under LPCXpresso

duncanellison, 2016-03-23:
Hi, Duncan from Cape Town here.

Received my oven yesterday at 19:00 and by 22:00 had completed all the mods.  Thanks to everyone here who contributed - an awesome difference in usability.  

Now I'm trying to drill into the code a little bit and find I can't compile under LPCXpresso [V8.0.0 build 526] due to these annoying syntax errors in vic.c :

**	asm("MRS %0,cpsr" : "=r" (state));**
../src/vic.c:43:20: error: expected ')' before ':' token

**  asm("MSR cpsr_c,%0" : : "r" (mask));**
../src/vic.c:57:22: error: expected ')' before ':' token

I get by in C but I'm not a real expert.  From what I can see this is some discrepancy between GCC and the syntax used by the compiler embedded into LCPXpresso and has me stumped.

Can anyone assist ?

Duncan

colinoflynn, 2016-03-23:
This had been broken for a while,I have a pull request that should fix it. 
You need to switch to gnu99 mode in the compiler.

Either patch the project file as indicated in my pull request, our change 
the mode via the GUI. I don't remember where this is in the gui, but look 
for c99 under build settings and change to gnu99.

duncanellison, 2016-03-23:
Thank a million.  That fixed it.
I would have been a while on that one.

closed 2016-03-23 by duncanellison