UnifiedEngineering/T-962-improvements #89: Initial Test Result

duncanellison, 2016-03-27:
Not really sure where to post this as it's not really an 'issue' as such , but I've done some experiments with my oven that are as scientific as I can make them with my set up and thought I would share them.  Please feel free to move / edit / comment / delete as necessary.

I have a T-962 (not 'A') with the original analogue TC1 and Tc2 flashed with Werner's latest 'master' code as at 27/3/16.

I have not attempted to calibrate the TC's at all and all slopes and offsets are at 0.
Min Fan Speed is set to 12.

Using the serial protocol on the T-962 and an accurate temperature logger with the TC fixed to a small PCB in the centre of the oven (mounted on pcb standoffs and with a small amount of thermal paste between a track on the board and the TC junction), this is what I got -

[image]

Yellow is the setpoint tracking for SAC305.
Orange and Grey are the two TC's
Blue is the temp as reported by the oven (the average of TC1 and 2)
Green is the *actual* temperature at the board.

Interesting points are -

1) Between 55 and 115C and again at 163C to 217, the two thermocouples track almost perfectly, but outside that, they vary wildly with a really odd kink centered around 139C and in the cool down they are miles apart.

2)  The actual board is slower to heat up than the setpoint by quite some margin, but does eventually reach the peak temp at about the right time.

3) With these settings, the board gets around 13C hotter than the intended peak.

4) The cool down rate of the board is way slower than the air temperature (which is to be expected), but means that you can't accurately rely on the oven reading to give you the cooling profile unless you have a TC on the actual board.

I have one of the 4 channel digital boards arriving soon and will try to repeat the exercise to see how it compares.  I have a feeling that there's quite a bit of non linearity in the response of the analog TC circuit.

I'm thinking that the only real way to go is to have the two additional TC's permanently fixed to sacrificial PCB's either side of the one I'm baking.

Comments welcome.

GitLang, 2016-04-04:
Let's face it, these ovens aren't very good :). At least we attempt to make them better. My feeling is that the digital TC interface will help a lot - at least you will know the temperature that the MCU sees is the same (+/- 0.5C or so) as the temperature of the ovens TC's for real. Now we have to make the air flow over the board the same temperature as the TC's read. Your idea of two TC's on a board next to the board to be soldered is a good idea. In my opinion, it would be good to have some freedom of what constitutes temperature feedback. Maybe we could even weight each TC according to the thermal map over the floor of the oven. Once you have the digital board and four TC's fitted, you no longer need the TC Gain and Offset settings, so we could lose them and free up some scrreen area for TC weighting maybe?. Another thing that may help, is to show the TC average of the 4 TC's, then show 4 independant diffs from that average, to give an idea of temperature variation. We also need to clearly see how well we are following the profile, and the display of this LCD screen isn't good enough IMO. At the very least, we need a big font clear view of oven temperature. With a better scree that would turn yellow then red as it went away from the profile.

I've had some radical thoughts and looked on ebay for convection oven fans... :)

duncanellison, 2016-04-08:
Making more progress now.  Got my 4 x TC board up and running and made some significant upgrades to my windows monitoring code so that I can graph each TC reading / average and setpoint on the same chart together with deltas from the average.

I have the two additional TC's on the board fixed to sacrificial boards either side of the main area.

What I'm seeing was quite shocking at first.  The two TC's on the boards seem to get waaaay hotter than the two inbuilt TC's.  The two inbuilt ones were not even at reflow point as the board level ones were climbing over 265C.

If I'm reading the code right, the PID tries to follow only the first two TC's unless you use the define which gives any TC with a delta > 5C priority.  I'm going to have to look more deeply at how to make the code average better.

BTW are we all just making an assumption that the inbuilt thermocouples are Type K ?

I've also spotted that that MAX chips don't attempt any linearization and neither does the code.  I'm not sure how important that is in the 50 - 250C range without running some numbers.

mikeanton, 2016-04-08:
Seeing a large difference between a board probe, and an air probe is pretty normal.  On commercial reflow ovens, it is the air temperature that is controlled.  The operator has to profile the oven for each PCB that is run, by putting a probe on the PCB, and setting up the oven to get the correct profile as seen by the PCB.  This oven will require the same calibration process to be done before a particular board type is run.  As you found, the board temperature can be much higher than expected, and this is do to IR absorption.  This affects black components as well, and you will often find that the solder reflows on those parts before others.

duncanellison, 2016-04-08:
mikeanton, thanks for that, it makes sense.

What surprises me is that the PID controller seems to be taking no notice of the rising temperatures at the board and continues to drive the lamp even after the temp at the board is way above the setpoint, it seems to be using only the air temperature probes.

Here's another screenshot from my controller program to illustrate :[image]

Green is the setpoint
Purple / Green are the air sensors
Blue /Orange are the board level sensors

I'm going to have to dig into this code a bit more to see if I can't get more control input from the board sensors.  I guess not everyone has the 4 TC setup, so in that case it would make sense to only use TC1 & 2.

Also interesting to note that on the way up the fan cycling constantly at 10 makes for very smooth temperature gradients, but when the fan starts to kick in the cooling phase, the differential cooling effect on the probes is very noticeable.  I'm guessing that means that the airflow is very uneven in the chamber.

mikeanton, 2016-04-08:
Yes, it is probably using only the air temperature probes as it should.  It is a pain to always have to fasten a thermocouple to a PCB when running the oven so no commercial ovens do it that way.  Hence the requirement to profile the oven for a given PCB run.

timofonic, 2016-04-08:
Would an IR camera improve the process? How to put it inside and avoid damage?

mikeanton, 2016-04-08:
An IR camera could probably look through a window that passes IR, and be outside the oven.  I think saw a video on youtube where someone was doing this, but I can't find it at the moment.

That would probably only allow you to look at one board, so wouldn't control the average temperature of the oven as well as multiple thermocouples might.

GitLang, 2016-04-08:
Duncan, as I'm sure you've seen in sensor.c,  it's clearly using only TC1 and TC2 -
avgtemp = (tctemp[0] + tctemp[1]) / 2.0f;
Easy to change but, as you say, it needs some careful though about the algorithm you use. And as Mike says, whatever you do there is a need to do a dummy run to calibrate the oven for a specific setup. I also recall something in the "manual" saying that it was advisable to put the PCB on top of another piece of PCB material about 100mm x 100mm. That could be an airflow thing, as the big fan at the back sucks cold air in and effectively blows it over the board, through the slots in the tray and out the bottom of the oven. It would be good if we control mixing of that cold air and the hot air better

Can I ask to obvious chicken and egg question - did you use the oven to solder the  MAX31855 board? :). Did you use the one with the slot cut out from "dirty PCB"?, if so, how did you manage componnets on both sides of the board?

duncanellison, 2016-04-09:
OK, so I did a quick hack and used all 4 TC's to create the avgtemp.  No weighting, just a plain average.  Here's the result
[image]
Rather nicer !

Max temperature reached at the board level was around 245C and the TAL was about 75 seconds, more or less perfect for SAC305 which was the profile I used.

Fully agree with Mike, but since this (for me at least) is a prototype oven, it's no real hardship to have the extra TC's permanently affixed to boards in the drawer.

So, for now, I'm probably going to add a command line option to allow TC1/TC2 or TC1/TC2/TC3/TC4 or TC3/TC4 to drive the PID so I can easily switch these and leave it at that.

I'll run some real test boards later and post back the results.

@Gitlang re the chicken and it's egg ;-)  I did order PWTL's very nice board from DirtyBoards.  Actually I used my toaster oven with  a separate controller to reflow and it did a very nice job (!)  Only one solder bridge on the SPI chip because I dragged the paste during placement.

I reflowed the top components, then hand assembled the bottom with a hot air pencil.

FWIW, I went the extra mile and ordered the stencil and that made the job much easier.

GitLang, 2016-04-22:
Hi Duncan, how is it going?. I tried to contact you a couple of times by email at the address you gave, but I've not heard back. After calibrating the gain and offset of the thermocouples it has been going much better. I used a dual K type thermocouple thermometer, and taped the TC's standing from the floor of the tray, such that when it was slid in, the two pairs of thermocoiuples were just about touching, so I could now accurately measure the temperature of each thermocouple. I ramped the temperature up slowly and recorded several points of actual TC temperature and TC reading on machine. I put it all in excel, did a scatter graph and fitted a best straight line to each thermocouple. I could now correlate the graph offset with TC offset, and the graph slope with TC gain. That will keep me going until I get a MAX31855 board sorted. Just finished building it, now to test and fit it. I used the I2C/SPI conversion chip in a TSSOP package and did the solder paste for it using one of the dedicated chip stencils from Proto Advantage, and the rest with a syringe and needle (for the last time I hope!). BTW, I had components both sides and had no trouble doing a run for each side. I used standoffs each side of the board being soldered, and but a blank PCB under it when doing the second side.

I could really have done with the new version of your controller/graph program. Any chance of a download please?.

Oh, I worked out a reasonable profile for that SD528T chinese Sn42/Bi58 paste that is on ebay for about 12.60usd for a 100gm syringe. It's rediculously cheap and the results were OK, but I'm nervous of its quality and joint strength. Very attractive to have a 170C profile peak though.

duncanellison, 2016-04-22:
Hi GitLang,

Actually this oven is taking up waaay too much of my time, but I'm on a mission now.

I've modified my oven code to add an extra item on the menu - tempmode.  This allows you to use the 2 air TCs or the two board TCs or all 4 or to ignore any that are more than 5C different to drive the PID.

Unfortunately, I did not check out a copy of MASTER, so it's not under VC, but I'd be happy to share with anyone who would like to see what I did.  Copy here : 

[src.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/232429/src.zip)

Results : using the two air sensors resulted in an undercooked board.  Using the two board sensors, was almost perfect.  Nice reflow across the board, no big overshoot.

So bottom line is that I think the two board sensors only are probably the way to go for me.  FYI, they are taped to two small scrap PCB's either side of the main board area and I use them to stand off the board I'm soldering from the deck.  they route out the back of the chamber via the fan area to the board, so they are not really intrusive at all.

I'm now struggling to understand why the left air (original) sensor is so far off the rest.  I'm seeing up to 40C difference (lower).  I know it's a tendency for these ovens to be cooler on the left side, but that much ?

I thought that I had a bad channel on the board, but on swapping the channels, the problem stays on the left hand side.  I tried a replacement thermocouple and that didn't help, maybe even made the issue worse. 

I need to look at the code again, but I think that the gain and offset code doesn't operate on the values when they come from the MAX31855, presumably in the assumption that these should be accurate, but I'm going to look into that  

I have done some extensive mods / bug fixes to the controller program including the ability to save and retrieve the profiles and to automatically allocate run numbers to each cycle.  I've attached a .zip, unpack it before install and let me know what you think.  If you already have my old version installed, it should just update it.

[Reflow Controller.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/232415/Reflow.Controller.zip)

Just to warn you as well, this version only updates the UI every 0.5 seconds as I thought it was overkill to update at the full PID rate.

Thanks for the tip on the Chinese paste.  I'm actually in South Africa, so Ebay doesn't work so well here, but I bet it's available through Aliexpress.

GitLang, 2016-04-27:
Late reply, sorry, health not good. OK, finally fitted the MAX31855 based TC board today and have it working fine. First test in bake mode saw the temperature going down fast with lamp full on, so I stopped it and put the TC wires correct way round :). I watched it for quite some time at different bake temperatures. First observation, as I think we all know, is that the fan PID needs some tuning, it'd definitely oscillating. I especially watched  that "L going lower than R, especially when the fan is on" thing and I didn't have any idea until I forced myself to think Rear and Front, not Left and Right. It seems to me that when the fan comes on the rear (L) TC gets a bigger blast of room temperature air than the front (R) sensor. It makes perfect sense. When the oven is cold, arrange for the fan to come on and put your hand just above the tray with your finger tips near the TC's. It's surprising what a blast that fan puts out. So, it comes in at the back and makes a beeline for the TC's. It has to slightly raise the oven pressure in order to push hot air out of the vents at the bottom. It has to go round and through the tray to achive that. It may well be mixing but I believe it is mixing *after* at least the rear thermocouple and to a lesser degree the front TC too.

I want to try something. I want to make a diverter to divert air round the rear TC and maybe a the fron TC too. I am think a small piece of metal about 20mm (3/4") high and 50mm (2") to 75mm (3") long. I will bend it in the middle across the small dimension until the included angle is about 60 degrees so it is like a Vee looking down on it. I would then somehow fix this so it shields the rear TC and allows air to be diverted each side of it. With that we could at least prove if fan air directly on the TC is a source of problems. In the longer run, it needs a more thourough study of air flow in the oven, so it can be properly baffled for maximum mixing, and thus so we end up controlling the air temperature. Part of the that exercise may involve an internal fan too. If you look at how fan assisted convection ovens work, with the sensitive parts of a fan outside a firewall and the metal impeller inside the oven. We have a chance of doing similar, as the there is a metal roof to the oven that separates the top section where the circuitry is, from the oven - perfect.for adding a fan. I'm just thinking out loud really, but I believe that air diverter is well worth a first try.
Late reply, sorry, health not good. OK, finally fitted the MAX31855 based TC board today and have it working fine. First test in bake mode saw the temperature going down fast with lamp full on, so I stopped it and put the TC wires correct way round :). I watched it for quite some time at different bake temperatures. First observation, as I think we all know, is that the fan PID needs some tuning, it'd definitely oscillating. I especially watched  that "L going lower than R, especially when the fan is on" thing and I didn't have any idea until I forced myself to think Rear and Front, not Left and Right. It seems to me that when the fan comes on the rear (L) TC gets a bigger blast of room temperature air than the front (R) sensor. It makes perfect sense. When the oven is cold, arrange for the fan to come on and put your hand just above the tray with your finger tips near the TC's. It's surprising what a blast that fan puts out. So, it comes in at the back and makes a beeline for the TC's. It has to slightly raise the oven pressure in order to push hot air out of the vents at the bottom. It has to go round and through the tray to achive that. It may well be mixing but I believe it is mixing *after* at least the rear thermocouple and to a lesser degree the front TC too.

I want to try something. I want to make a diverter to divert air round the rear TC and maybe a the fron TC too. I am think a small piece of metal about 20mm (3/4") high and 50mm (2") to 75mm (3") long. I will bend it in the middle across the small dimension until the included angle is about 60 degrees so it is like a Vee looking down on it. I would then somehow fix this so it shields the rear TC and allows air to be diverted each side of it. With that we could at least prove if fan air directly on the TC is a source of problems. In the longer run, it needs a more thourough study of air flow in the oven, so it can be properly baffled for maximum mixing, and thus so we end up controlling the air temperature. Part of the that exercise may involve an internal fan too. If you look at how fan assisted convection ovens work, with the sensitive parts of a fan outside a firewall and the metal impeller inside the oven. We have a chance of doing similar, as the there is a metal roof to the oven that separates the top section where the circuitry is, from the oven - perfect.for adding a fan. I'm just thinking out loud really, but I believe that air diverter is well worth a first try.

GitLang, 2016-04-27:
Sorry about that, there must be a limit on message size, but instead of telling you that you have reachged it, it just copies your message from the start. Very annoying, it's thrown out the second half of my message without warning :(

duncanellison, 2016-04-27:
Sorry about your health.  Quick question - are you using a T-962A by any chance as you mention of front and rear are confusing me a bit. On my T-962 (the small one), the TC's are more or less equidistant between the front and back and I just assumed the variants were like that also.

Given that I've got access to all 4 TCs now and hacked the firmware to follow either the two air or the two board ones, after a few dozen test runs, I've kinda given up on ever getting the air TCs close and have settled on using just the board level ones.

Using this set up, I've managed to successfully reflow a couple of boards with 100% success with no evidence of board burns using lead free solder paste. 

There is a note somewhere else on this Git of someone trying to add a small fan and I believe he found that quite successful if very fiddly to do.

Were you able to make use of my controller program and did you have any observations on it ?

Here's an further updated version that allows you to store liquidus temperatures for the different profiles and automatically calculates TAL.

Duncan
[Reflow Controller.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/239447/Reflow.Controller.zip)

GitLang, 2016-04-28:
Sorry, yes, I'm using the T-962A which has the thermocouples down the middle, front and back. I have had your reflow controller talking to release 0.5.1 rev F OK, but I *STILL* have not managed to build a bin/hex which will run on the oven. I can build a binary OK with zero errors or warnings on two different compilers, and flash it and verify it ok, but always it crashes, usually at the start screen. Regardless of that I need to get a new firmware running so I can talk to your controller ok. Could you possible upload or email your source tree so I have a known working LPCXpresso project with correct settings. I may also be able to compare it's settings to see what I am doing wrong.

duncanellison, 2016-04-28:
Ah - I guess my L and R peppered in the GUI I wrote doesn't make much sense if you have a -A version then !  That's the downside of writing code for something you haven't seen.

Below is a full zip of everything in my LCPXpresso workspace, I wasn't sure how to export this properly, but hopefully you will be able to easily import it.  I just built, uploaded and tested it OK.

[T-962-Controller.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/241097/T-962-Controller.zip)

You sure it's nothing to do with your programmer settings (wrong frequency / part etc)

GitLang, 2016-04-28:
Many thanks Duncan. Well, I flashed your Release hex file as received, and it basically works but has quite a few problems 
1. On First menu screen, Oven Temperature shows as 0C
2. Switch to Bake screen, target temperature shows 25C. It can be increased ok, and when decreasing will not go below 25C.
3. Before starting the bake timer, the cold junction temperature is showing as jumping between 12C and 4...... full maximum 32bit number
4. Before starting the bake the TC's are showing about 18C, roughly 10C low.
5. Start the bake timer and the Lamp comes on solid and stays on. TC's increase but show low. Cold J continues to oscillate between 12C and NAN
6. All this time, the small electronic cooling fan is a) ramping up to about 1/4 speed over a 4 second period, or full speed over a longer period.b) It stays at 1/4 speed for about 4 seconds or full speed about 15 seconds. c)It then stops and rests for 10 or 12 seconds before starting all over again.

The lamp staying on means I cannot sensibly test a profile.

GitLang, 2016-04-28:
> You sure it's nothing to do with your programmer settings (wrong frequency / part etc)

Mised that, sorry. No, the programmer is reading back the flash memory and verifying it against the original file.

GitLang, 2016-04-29:
OK, the only one working of the 4 combinations (your compiler, my compiler, Debug build, Release build) is your Debug build. Both my Release and Debug builds just reset and crash back to the signon screen when trying just about anything. You Release build is doing the above strangeness. Debug builds often do things likeadding spare stack round auto variables with identifiable patterns in (deadbeef etc, we've all seem 'em). So if you have variables stamping outside the stack space, or over other auto variables, just trying a debug build can disguise the problem but tell you wahere to look for problems. What I don't undersatnd is that your Release build runs OK for you. Erm... doesn't it? I noticed you have Debug as the default build, is that what you are testing, the hex from the Debug directory or have you tried the hex from the Release directory too?

duncanellison, 2016-04-29:
Hi  Gitlang - Man, you are having a lot of trouble with this !  I just copied this directly from my FlashMagic screen, so I guess I AM using the release version  

.....\Desktop\T-962 Reflow Oven\LCPXpresso\T-962-improvements-master\T-962-Controller\Release\T-962-Controller.hex 

I'm pretty sure I also tried the Debug with no adverse effects, but that was a while back and I've been compiling with release as the target recently, so I'm guessing the .hex in debug is much older and therefore the reason it's working for you.  There's probably a clue in there somewhere.

What are you using to flash the code ?  I'm using FlashMagic 9.65.4115 and it seems fairly bombproof.

[image]

GitLang, 2016-04-29:
Yes, looking at your archive, the last debug build was a couple of weeks before the last release build. Weird that the debug works for me but the release doesn't. I'm using 9.6.4101 flash magic, and I'm sure that is ok because it verifies against the file. I've even run flash magic on a different occaision and done just a verify and it is ok. I've just updated it to  9.73.4146 but I doubt it will change anything. Yes to -gnu99 too. From what you say, I need your source from 8th April when you last compiled a debug build. If you're anything like me with backups then you don't have it. In which case a bit of comparitive reverse engineering may be called for :(

GitLang, 2016-04-30:
OK, getting somewhere. Did a debug build of your last source after importing it a bit differently. STill crashes back to info easily, but I have logged some serial info on your monitor. I had not realised the serial stream started so early  or I would have tried it before. Here is what I get ...

```
T-962-controller open source firmware (0.5.1 (C))
See https://github.com/UnifiedEngineering/T-962-improvement for more details.
Initializing improved reflow oven...
Reset reason(s): [EXTR]
Running on an LPC2134(/01) rev F
Waiting for keys to be released... Done!
Buzzer_Init 
ADC_Init called
OneWire_Init called
Scanning 1-wire bus...
 Found 6b0115c2224dff28 [DS18B20 Temperature sensor]
SPI_TC_Init called
SC18IS602B_Init - Done (addr 0x28)
Probing for MAX31855 devices...
SS0: [SPI Thermocouple interface]
SS1: [SPI Thermocouple interface]
SS2: [SPI Thermocouple interface]
SS3: [SPI Thermocouple interface]
 bake setpoint values: 0, 1e, 30
SystemFan_InitInvalid temperature mode ! 
# Time,  Temp0, Temp1, Temp2, Temp3,  Set,Actual, Heat, Fan,  ColdJ, ModeInvalid temperature mode ! 
Invalid temperature mode ! 
Invalid temperature mode ! 
Invalid temperature mode ! 
Invalid temperature mode ! 
Invalid temperature mode ! 
Invalid temperature mode ! 
 bake setpoint values: 0, 1e, 30
Invalid temperature mode ! 
Invalid temperature mode ! 
```
"SystemFan_InitInvalid temperature mode ! " is just a lack of a "\n", SystemFan_Init is OK, the "Invalid temperature mode ! " message is coming from Sensor_DoConversion(), in the if (tcpresent[0] && tcpresent[1]){}  conditional. It fails with that message in the default case in the switch (mode) after getting mode from 

`int mode = NV_GetConfig(TEMP_MODE);`

So I think NV_GetConfig() is returning nonsense and causing the failure. That part of the source is where you choose what combination of sensors you will use for temperature feedback, but I can't find anywhere the NV is initialised or where you choose temperature mode and store it in NV. Remember, the data I have in the NV will be left over from a previous version and will not make sense to your structure.

duncanellison, 2016-04-30:
Hi there !  Great, seems like you are moving along, but I need to give you a health warning.  Remember that my code dump isn't an official branch, I just took a copy of Master as it stands now (0.5.1) and hacked it a little for my purposes (I added the (C) to the version number). It works for me but isn't an 'official' branch.

Having said that, I've successfully reflowed boards using this version.  The main change is the 'Temp Mode' which is why you are getting errors, you probably have random garbage from the EEPROM in that slot.  Do a 'help' in the command line and you'll see that there is now an additional setting 'temp mode' and help will explain what mode is which.  I recommend mode 3 with your two additional sensors taped to the board or some scrap boards next to it.

You need to change modes with 'setting 6 3' (I think !) and reboot, let me know if that works.  If not, you are coming across some h/w mode at your end that I didn't cater for in my hack.

I'd recommend making sure you have the toolchain fixed by making some innocuous change and attempting to reflash, then grabbing the 'official' master branch again and seeing if you can compile that with the same process.

Duncan

GitLang, 2016-04-30:
Understood. I'm using your firmware as it talks fully to your remote controller with 4 channels, and I've completely lost track of what repository builds work for me or not, and I also don't know if any of the repository hex builds have support for the four MAX31855's and the system fan mod. I know it's in the latest repository source but I can't build a working binary. GRRR!. Seriously, I'm epileptic and have really poor memory because of it and if I stress myself over this I'll kick off a cluster of seizures again.

I slipped a   NV_SetConfig(TEMP_MODE, 0);  after   SPI_TC_Init();   in main.c and ran it once to set it to using the first 2 TC's. I don't know what I will do with the second two yet, I 've had good control and results with the two TC's, but just not calibrated to the real world, that is the main reason I wanted the MAX31855 interfaces. If anything I will add two TC's in the oven so I have L, R, F, B and average them for control. my main problem isn't the oven it's the damned software!. It's throwing me a bit not having a true debug link for tracing, breakpoints, etc. BTW, the system fan control has only ever worked once, erratically, as described with one of your debug builds. That proved the AD0 drive is there, and I've checked the hardware out, I'm guessing it's linked to whatever the problem is that I have.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk