UnifiedEngineering/T-962-improvements #90: multiple defintions of avgtemp

majbthrd, 2016-04-02:
Both sensor.c and reflow.c independently define a global variable called "avgtemp".

The link used in the Makefile could/should have thrown a warning/error, but the map file indicates that it is silently collapsing both symbols into the same memory location.

If the intent was for each source file to share the same variable, one could be declared as "extern".  However, if the intent was for each source file to have its own variable, they could both be declared as "static".

GitLang, 2016-04-04:
Ouch. This is when the C++ crowd start shouing Namespace!

majbthrd, 2016-04-04:
You would be hard pressed to find someone more opposed to C++ in embedded devices than I am.  This is just a simple coding oversight with a choice of straightforward fixes.

GitLang, 2016-04-04:
Fully agreed. Took me many years to be convinced even dynamic memory allocation was safe in the early days.

I don'y know what GCC compilation does, but with LPCXpresso the makefile is created before use and removed by a clean or at the start of a build. Sure, it's an easy error to make and easy to fix, but the fact that the compiler (several of them it seems) is not reporting this as an error/warning is amazing.

majbthrd, 2016-04-04:
Yeah, it popped out as an error only when I compiled the code under Rowley Crossworks (also gcc).  Informal online searching suggested that it *might* be a "feature" of invoking the linker via gcc (as opposed to using gcc's ld) as the compiler unifies all the symbols before linking?

GitLang, 2016-04-05:
I think you may be right. Every other closely linked compiler/linker I have tried has thrown up a linker error after the compiler innocently passed it the information.

Anyway, I think we can safely rename it in reflow.c, as it assigns it thus 

reflow.c -
   avgtemp = Sensor_GetTemp(TC_AVERAGE);  // function is in sensor.c

and every other reference in reflow.c lookslike a reference to that value it got from Sensor_GetTemp(), so they are two separate variables, assigned the same value at one point only. I'm going to rename avgtemp in reflow.c to average_temp

mikeanton, 2016-04-29:
The compiler won't trap it as an error, but the linker should.  The compiler knows nothing about what is in other source files.  The linker combines them into the executable, so it would be aware of two variables defined with the same name.  Unless one of them is declared as static, it should complain.  Perhaps the warning level is not set high enough, though really this shouldn't be a warning anyhow.

GitLang, 2016-04-29:
As far as the compiler is concerned it's valid C, isn't it, however daft it may be?. The linker correctly knows it's a link error, but from the compilers point of view, those two files could have been destined for different targets.

mikeanton, 2016-04-29:
Yes, it is valid C.  The compiler doesn't even know the other file exists, so it doesn't see a problem.  It is only when they are combined that it shows up.

John-Nagle, 2016-12-28:
It's not a link error. Unfortunately, it's valid C. Here are the C rules:

- If you want a variable to be local to a source file, put "static" in front of the declaration. Like

     static float avgtemp;

This is the *normal* case. **Use "static" everywhere you don't want global.**

- If you really want to share a global variable across source files, put the declaration in an include file and include that file in each file where you want to access the variable. 

- If you initialize a global variable in its declaration

   static float avgtemp = 0.0;

and do that in two different places, you will get a link time error, because the initializations clash.

John-Nagle, 2016-12-28:
Looking at the code, this looks like it doesn't cause trouble. The only assignment to "avgtmp" in "reflow.c" is 

    avgtemp = Sensor_GetTemp(TC_AVERAGE);

In sensor.c, Sensor_GetTemp is
```
float Sensor_GetTemp(TempSensor_t sensor) {
	if (sensor == TC_COLD_JUNCTION) {
		return coldjunction;
	} else if(sensor == TC_AVERAGE) {
		return avgtemp;
	} else if(sensor < TC_NUM_ITEMS) {
		return temperature[sensor - TC_LEFT];
	} else {
		return 0.0f;
	}
}
```
Calling that for TC_AVERAGE returns "avgtemp".  So the assignment in "reflow.c" just assigns "avgtemp" to itself. That wasn't the intent, but I think it works out OK.

xnk, 2018-02-05:
Fixed in PR #122

closed 2018-02-05 by xnk