UnifiedEngineering/T-962-improvements #93: Main Fan doesn't run

BigFootoo, 2016-07-08:
Hello
I noticed that after doing all the mods to my oven, the main cooling fan does not run even though the fan light is on. Let me clarify a little, the fan will run when full on is applied but not at slow speeds. I removed the back cover and discovered that my T962 has a PAPST 4606X fan. I was surprised at this since these fans are very nice and made in Germany.  New ones cost around $50. I noticed that the fan rotor has a fair amount of mass and when the fan light is anything but full on, it just sits there and twitches. Has anybody else discovered this and replaced their fan? I suspect that a different fan with less rotor mass would work ok.
Ben

GitLang, 2016-07-26:
That's the standard fan and works for eveyone else. So I'd look elsewhere. It sounds like maybe it's not getting enough drive so maybe the triac is faulty, or you've knocked a wire off, or pur a wire back in the wrong place. With all power removed and fan unplugged, is it free to spin by finger?

BigFootoo, 2016-07-26:
Hello Gitlang

I have replaced the fan with one that has lower spin mass and it seems to work fine. It has approximately the same air flow. I’m thinking that the original fan just needed to run for a while to loosen up the bearings. The unit I have has been sitting new in the box for at least three years. Thank you very much for responding and for making the firmware available. I got it loaded just fine. The only problem that I am having is that the left side of the oven is about 10 degrees cooler than the right side. I used separate thermocouples attached an external meter to measure this. It’s probably inherent in the design and not fixable. Thanks for adding thermocouple off-set and gain to the software. It helps. 

Best Regards

Ben Fisher K9BF

koendv, 2016-08-06:
Do you still have the original fan? Pabst fans have a date code om the label: week.year ; eg. 48.03 is week 48 year 2003.

I ask this because I am wondering whether Puhui is using "recycled" fans. I was looking at the small 40mmx40mm system fan of my T962. The system fan looked worn, with what might be a date code for Nov. 23, 2008. [image]

BigFootoo, 2016-08-07:
Hummm

There is a number on the stick on label, 2893, but the label looks new and the fan does too. Well at least newly painted, hihi. 

Ben

BigFootoo, 2016-08-08:
Koen

I took the fan apart this afternoon and found rust and worn bearings. So I though it a way.

Ben

KtroniX, 2016-10-19:
If someone would be interrested I did small modification to program code and lowered tresholds for electronics cooling fan of -10*C. This allows it to start and cool down heter triac radiator. Base for changes was software 0.5.1

koendv, 2016-11-08:
Wouldn't it be safer on the T962A to 

- when baking, switch the small fan on
- when not baking, switch the small fan on if the temperature is too high 

 The fan making noise all the time is too much. But I can live with the making noise during the baking cycle, if that means the triac works reliably.

koen

KtroniX, 2016-11-08:
on T962A there is solid state relay which controlls heating IR lamps. Sorry I forgot to add information that this is applicable to smaller one.

koendv, 2016-11-08:
I'm sorry. I meant the T962 (the small one).

KtroniX, 2016-11-09:
Yes, It is reasonable way of working. This change which I made was the simplest one. I didn't have time to analyse flags handling in source program etc.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk