UnifiedEngineering/T-962-improvements #95: DS18B20 cold junction sensor not detected

Octoate, 2016-08-20:
Hi,
first of all: thanks for your awesome work here. Without your work there would not be such a great and cheap reflow oven available :-)!
I have currently followed your instructions and set up my T-962 oven, but I have the problem, that the DS18B20 is not recognized by the firmware (I am using the latest HEX file from the releases here on GitHub - 0.5.1), which means, that it shows "No cold-junction TS!" in the "Bake / Manual Mode" menu.
Measuring Pin 2 of the DS18B20 with a multimeter, I read 0,8V if the DS18B20 is connected and 3.3V if I disconnect it. So it seems that the 4,7k resistor is working. To be sure, I have also replaced the DS18B20 with another one, but with the same result.

Do you have any ideas, what could be wrong?

You can see my connection of the DS18B20 in the image below.
[image]

GitLang, 2016-08-20:
Are you certain it is a DS18B20 from a reputable source?. I can't see any writing on the TO-92 device at all and the Maxim parts are usually quite clearly marked. You have tripled checked pins 1 & 3 are connected to the ground plane with a meter, right?. Lifting the red wire and checking the voltage was the right thig to do. At least you know it is drawing current. It should draw <1uA in standby and 1mA when active, so your 0.5mA sounds about right for the device in periodic use. 
So... 
Is the firmware OK? 
Is the 4K7 resistor OK?. 
Does your red wire short anything out around that resistor?
Have you dropped any solder blobs?
Have you scratched any tracks?
What does bake mode say with the red wire disconnected?
Have you tried different firmware?

I'm not sure what it will prove, but if you put the T962 in ISP mode, then the DS18B20 should definitely be in standby and drawing < 1uA so the 4K7 should only be dropping a few mV

Octoate, 2016-08-20:
1&3 are connected to GND, yes. I got the two DS18B20 sensors from two different distributors - I am not 100% sure, but I guess at least one of them should be ok. On the photo the text was not readable because of the light (see photos below). 
I also double checked for scratched tracks or solder blobs and haven't found anything. At the moment I only tested the oven with the 0.5.1 firmware, so the next thing is to flash an older version of the firmware and test it again.
[image]
[image]

GitLang, 2016-08-20:
Yes, the device looks genuine ok. Couple of things that are stretching it a bit, but they have both caught me out over the decades. 

The middle leg of the DS - it looks bent very sharply and very close to the encapsulation.  I have long ago caused a device to go open circuit like that. The leg has an incredibly fine wire welded to it and to the Si chip and any strain will break it. I was taught to always hold the leg close to the package with fine nose pliers and do the bend by holding those pliers and bending the leg with more pliers, so the device gets no strain.

Another thing I've done in the past is roughly tacked on a wire for mods or testing, to a PCB component, like your red wire to the resistor. Not much flux around, so the solder blobs. Once in a blue moon there will a tiny scratch in the solder resist and unknown to you the joint sporadically touches the ground plane, leading to one of those horrible faults which appears then disappears. I'd unsolder the wire, clean up both ends with a solder sucker or wick, and resolder the wire freshly with flux, it should make a good clean joint. You need 3 hands :)

These are oth long shots, but I'm having trouble thinking of what else it could be. Ideally you need to 'scope whatever is happening on pin 2.

If Duncan is reading this, he will remember better than me, but there was something about the last "master" release not having the last few mods incorporated. If that is still the case then you need to build your own hex file from the last source.

Octoate, 2016-08-24:
Well, I will test to flash another firmware version first, before I start to desolder and resolder the components again. Hopefully I have some time during the weekend so I can test version 0.4.1. The "nearly broken" middle leg is just a bit of plastic of the sensor housing. Looks more fragile on the picture than it really is, but I will also double-check this as well.
I also set up a virtual machine to compile the source code, but if I understand you correctly, I should use the source code of an already released version instead of the source code of the master?

Octoate, 2016-08-26:
Ok, I have flashed it with firmware v0.4.1 and also tested v0.5.0 and v0.4.0 - unfortunately with the same result. The DS18B20 was not detected. I also resoldered the modification and used completly new parts - also without success. I double checked the connections to the CPU and also to the DS18B20 and everything seems to be ok.
When I hooked up my scope, I was not able to see a clear communication on Pin 2 of the DS18B20 - however, I am not sure what to expect when using the parasite power mode of the sensor.
Maybe it would be an option to use the DS18B20 in non parasite power mode?

dspdon, 2016-09-04:
I have a similar head-scratcher with  0.5.1 --- I've failed to get a DS18B20 working, and am  just seeing "No Cold-Junction TS!" on the manual/bake screen after trying a few different devices and a few re-solderings.  HOWEVER ... your issue may be simpler if my eyes don't fool me.  I think you've got a 4.7ohm (470) not a 4.7kohm (472) SMD resistor in your diagram.  If true, the overly-strong drive would easily cause the issue you're seeing.

GitLang, 2016-09-04:
You had me going for a minute there until I realised it's a 4 digit code 

4701 = 4K7

Shame, that would have been a nice easy fix.

dspdon, 2016-09-07:
What oven model and production date do you have?  Mine's T-962A, production date 16.03, with LPC2134FBD64/01.  Could PCB, parts or timing have changed?  Anyone got a recent oven PCB working with TC?  For debug, I see  boot-up messages  fine over ISP + FTDI, confirming no TC sensor detected.  I am monitoring the 1-wire line via logic analyzer.  Can't yet explain lack of connectivity to the TC.

Octoate, 2016-09-07:
I have a T-962, produced 16.02. When looking at the debug output during boot-up I see that the 1-wire sensor wasn't detected. The resistor is defintly a 4k7 (I double checked it before soldering it to the board :-)).

[image]

dspdon, 2016-09-09:
Ok: sample size of 2 "newer units" that fail to work with 1-wire.  I tried the usual parasitic-mode setup first.  Next I tried what "matthiasm" did:
           http://www.matthiasm.com/img/20160113_153215_hi.jpg

He connected his DS18B20 to the 3 pins where the 4k7 pullup resistor is added, so as to NOT use parasitic mode.  I did the same, and found my T-962A froze during bootup - no beep, no LCD display.  I removed the DS18B20 from the 3 pins, and T-962A resumed normally.  So there's two tests indicating newer T-962A's are different.  (Or ... both of our DS18B20's are peculiar.  Doubtful.  I tried 3 different units from 2 different sources now.)

dspdon, 2016-09-09:
BTW, my unit is a T-962A, 120v / 60Hz, yours is T-962 220v / 50Hz.  Both 2016, both fail.

I will try to "beep out" that GPIO line being bit-banged for 1-wire, to ensure it's still going to the same "middle" pad.  And confirm the 3.3V pad is still adjacent.  I'm not seeing logical signals on the analyzer that make me feel confident things are right.

I wonder if I'm overlooking something really lame.  Just checked that the power cord is plugged into a live AC outlet. ;-)

Octoate, 2016-09-09:
I tested the non parasitic mode of the DS18B20 yesterday, too - with the same result: The system doesn't boot up. Hmm... strange. Is it possible to get the schematics of the newer version of the oven?

wulfmans, 2016-09-09:
you will need to create them. I created the last one but you have the
different board.

GitLang, 2016-09-09:
So.. remind me, what is the result of running the latest firmware from here, 0.51 is it?. Does the firmware work as design but with the exception of the cold junction compensation, or does it work with bugs, or does it not work at all. If it basically works, but fails when you add the DS18B20, then it does sound like they are using the gpio pin as an input for something important.  So yes, trace it out, make sure the solder pad you are using does go to that pin. If it doesn't then maybe the original pin is usable, you never know.

My recommendation would be dump the DS18B20 and build the *proper* thermocouple interface using the Maxim.  You have probably read this -

https://theembeddedworkshop.wordpress.com/my-projects-2/arm-projects/improving-the-t-962-reflow-oven/

You can get 10 unpopulated PCB,s from dirtypcbs.com, and very good quality they are. You can assemble in your oven as it stands if you are careful and set temperature about 20C lower than normal. Scariest part is the TSSOP chip but begineers here have suceeded with it. Then you have four channels of very accurate thermocouple interface, digitally connected, no trimmers to fuss with, and the cold junction compensation all done for you. That board has an optional DS18B20 if you want to play with it, but leave it out. The only thing you will miss is possible future temperature control of the electronics enclosure, but the CJC on the Maxim chip could give the required information anyway. The only interface it needs is the existing IIC so I doubt they would have changed that. From what I have read, that PCB fixes the temperature problems for almost everybody and is very successful.

dspdon, 2016-09-09:
Yes, 0.51 works just fine - no issues ... except cold-jct comp fails due to lack of h/w connectivity.

Agreed on direction.  The ease of DS18B20 integration was motivational, but it now looks like it might be out of reach.  I've already got an I2C/SPI bridge breakout on order: a bit big, but it may fit.  Awaiting arrival.

On the subject of MAX31855, I recently developed fairly comprehensive support for MAX31856 (newer than the '855, where nonlinearity comp is builtin for example), but that support is for Arduino frameworks (tested with AVR, Photon and RedBear Duo).  I used that with Reflowster (a prior retrofit I did - T962A oven should be MUCH better). I am NOT intimate with the NXP LPC2134, however, and I don't see Arduino integration - so my interrupts, timers, other code would need revisiting.  ** Is there Arduino framework support for LPC21xx? **

Assuming no Arduino support, I installed LPCXpresso toolchain a day ago and am "booting up" on that, in an effort to keep using the T962A microcontroller PCB and UE's core firmware.  I installed the latest LPCXpresso however and all sorts of bad on initial compile is occurring.  More learning.  May need earlier release of tool chain.

GitLang, 2016-09-09:
> MAX31856 

Why bother, you are just giving yourself work for no gain. We don't need even the precision of the 3185x, let alone curve correction, it would be a total waste of effort. Just get the PCB, assemble it, wire it up, and forget about it. You won't control the inside of that oven to within better than a decade of what a 3185x would give you at 12bits over a 10C to 250C range.

dspdon, 2016-09-10:
I may order that MAX31855 PCB, and in the meantime, trace out where that 1-wire GPIO0.7 pin goes.

dspdon, 2016-09-11:
SUCCESS!  It was the DS18B20's.  Well, one batch of them...

I finally tested the PCB pads on my "16.03" oven, and they went where they were supposed to go: both VCC (marked C20 on my PCB) and P0.7 GPIO (marked R2).  My PCB differs in its silkscreen, supporting the theory of a possible PCB trace change. I realize other PCB photos I've seen online also show differences among themselves as well.  But concrete proof was the connectivity and voltage tests - they confirmed no change in routing of these pads.

So then I tried a few more DS18B20's with renewed confidence.  I mistakenly tested two from the same vendor when I first did this - both turned out to be from an Amazon purchase.  With a bit more care, all of my Digikey-sourced components are working with the oven.  The original batch I bought via Amazon all fail.  Uniformly.  They are all marked "Dallas / DS18B20 / ..." however, and are easily confused with the Digikey parts.  I'll read/publish the remaining markings once I get a magnifying lens on it.  Checking my Amazon order, they were from "Ace Seller / GXParts" (who??)

Both parasitic and non-parasitic connections work with the Digikey devices.  FWIW, it seems much easier to connect the DS18B20 in non-parasitic mode, since GND / P0.7 / VCC are all adjacent.  No ground wire to run across the board.  And fewer caveats regarding mode-handling of the DS18B20.  Just need a little more care in having both the sensor and the SMD resistor both on the same pads.

So apologies for the false alarm, all is well with these newer ovens.  Just BEWARE of where you get your DS18B20's.

GitLang, 2016-09-11:
That's great news!. Unusual to get a bad batch of devices these days. Now you can take the DS18B20 off and fit the proper TC board :)

dspdon, 2016-09-11:
Well, next up is that fan connection.  I  plan to pursue MAX31855 once parts arrive -- I rarely pay for speedy shipping so it could be a while!

Octoate, 2016-09-16:
Ok, good news everyone. I finally got it working, too. It was the same issue which @dspdon had. After I read that he got his oven working with a new DS18B20 I ordered some replacement sensors from my usual distributor and replaced the ones, which I got from eBay, with it. Now the cold junction sensor is detected and shows the correct temperature.
Thanks a lot for your help!!!

closed 2016-09-22 by Octoate

KLEYNOD, 2017-10-14:
Hi 
I think it is better to connect them directly to 3.3v bus, there is 3.3v regulator on board, if you run them with 5v system then according to datasheet pullup are around 4.7k ,
CPU (LPC2134) has 3.3v then pullup should be arrounfd1k, I see many people run has 4.7k with 3.3v which leave not so much current for sensor

PCB4u, 2018-03-04:
Hi,
i had the same Problem with a 18B20 from Amazon.
I tried also with 1K, not working, then i tried to connect directly to 3.3V, also not working.

Then i bought a new DS18B20 at Reichelt.
Connected directly to 3.3V also not working, then again in Theft Mode, now it works.

Not working one from Amazon: DALLAS 18B20 1718C4 +158AC
Working one from Reichelt: DALLAS 18B20 1629C4 +518AC

rom1nux, 2018-10-31:
Hello,
Exact same problem here. DS18B20 buy from AMAZON (AZDelivery) not working on T-962-controller v0.5.2 in parasitic and non-parasitic. (But work in both mode with lax timing on other application).
Official DS18B20+ from MAXIM INTEGRATED buy from FARNELL (Fref : 2515553) work perfectly in parasitic and non-parasitic on  T-962-controller v0.5.2. Time to calibrate thermocouple now.

c-klinger, 2022-10-06:
Hello, just wanted to confirm the issue with cheap DS18B20s. This issue probably saved a lot of debugging time. Serveral genuine looking DS18B20 from Amazon and ebay were not working for me. New DS18B20 from Reichelt worked with 4.7k.

structuredDesigner, 2022-10-09:
Question: For the 4.7k SMD resistor needed, what's the size that best fits those pads? Would 0805 be fine?

c-klinger, 2022-10-09:
1206 was a good fit for me, but 0805 might work as well.