UnifiedEngineering/T-962-improvements #96: LED Blinking after Unified firmware 0.5.1

conjure86, 2016-09-23:
Is this normal? The LED on the board near the ISP header blinks slowly after the flash. 

Firmware works fine, just want to make sure this is normal..

conjure86, 2016-09-23:
Also I added the NPN Transistor to the ground on the side fan, with a 4k7 resistor to AD0. The fan briefly turns on at bootup, but never again. Should it ever be on? If so, when?

christaylor99, 2016-09-26:
Source code says the system fan is run 2 seconds as a power on self-test and does not turn on until cabinet temp gets to 40C or above, reaching full speed at temps 50C or above.  If no TC sensor  found the system fan runs full on as a precaution. See /src/systemfan.c

xnk, 2016-09-27:
Yes, the LED behavior is normal, it will flash slowly during normal operation. This was added early on to verify timing (as that was a major issue with the original firmware). Nothing to worry about. 
/wj

conjure86, 2016-09-30:
Great, thanks for the reply. I suspected it was, I just wanted to be sure.

I'm waiting on a new multimeter with thermometer so I can better calibrate this oven, but I can't for the life of me figure out what Gain and Offset actually do in the menu. Offset seems easy enough, you give it a + or - (in degree values?) in the menu to offset the temperature.

But what does Gain do? Provide more power to the element(s)?

xnk, 2017-01-10:
The gain simply will adjust the slope of the conversion curve, I've explained the idea here: https://github.com/UnifiedEngineering/T-962-improvements/issues/57#issuecomment-75911035 (but I gave up on the built-in ADC a long time ago)

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk