UnifiedEngineering/T-962-improvements #97: Inconsistent Results - Gain/Offset/Etc

conjure86, 2016-10-02:
This oven is a nightmare to work with. I haven't had consistent results whatsoever since I purchased it a little over a week ago

Modifications I've done:

**- Unified Engineering v0.5.1 firmware (Min. system fan = 12-15)
- Replaced masking tape with kapton tape
- System fan PWM mod
- Cold junction compensation**

I'm waiting on my new DMM with temperature monitoring so I can "properly" gauge what's going on in this oven. 

Right now I'm using SAC305 lead-free ChipQuik solder on roughly 4.5" x 3.5" PCBs. Getting this stuff to completely wet without having to run it through a second time has been a total nightmare. And running it through a second time completely discolors the solder that wet the first time... I've set a custom profile with the correct ramp up times, peak temp/times, etc. and it still doesn't work. 

I'd rather not just blindly switch to leaded solder. Yes, it'll "fix" the wetting issue and I'm sure I'll get more consistent results, but the obvious problem right now is temperature and consistency

Which brings me to my next question: When my DMM comes and it's time to properly calibrate this thing, what do the gain and offset settings do? TC Gain = Gain on the thermocouple. What does that actually do in terms of getting temperature accuracy? Likewise, what does the thermcouple offset actually do? I can set positive and negative values. 

I don't mean to sound short and/or impatient, I'm just kind of at my wits end with this thing, and documentation between the official firmware and this one is kind of sparse. 

Should I go ahead and do the 4x MAX31850K TC interface mod? Should I be using risers of some sort? I'd rather not use other garbage PCBs as standoffs for the PCB I'm reflowing. 

I know this is "only" a $200 oven, but I could get better results right now sticking a PCB in a toaster oven and watching it until the solder wets, so there's not much value in this oven for me right now.

Help!

wulfmans, 2016-10-02:
hmmm  I just did 25 boards 2 weeks ago that were perfect. /one thing you
really need to do is profile the heating curve
mine was reading way low. once we got the profile right it was perfect.

conjure86, 2016-10-02:
Thanks for the reply. What did you do to profile the heating curve? Looking for any tips or tricks to get this right.

Appreciate the help :)

mikeanton, 2016-10-02:
To profile the oven, you put a reference probe from your meter onto a pad of the PCB.  Then run the oven through a cycle.  Then iteratively tweak the profile temperatures, and times used to get the correct profile shape as measured on the PCB.  It usually takes me a couple of hours to do this.  Note that the profile used will often need to be different for different PCB layouts, so if you are wanting the best accuracy, then you profile a board before you do a run.  The patterning on the PCB, and the solder mask color can have a huge effect on the IR absorption of the PCB, which in turn affects the profile that needs to be used.

Likely the gain and offset of the sensor have to do with getting it to measure the air temperature correctly.  It won't change the profile much, unless it is out by a long way, and profiling the oven will take this into account anyhow.

conjure86, 2016-10-02:
Cool thanks, I appreciate the reply.

Right now I'm mostly doing 0603 stuff with a few QFPs, red or green soldermask depending on the two types of boards I'm running, finish is HASL Lead Free (though that seems to heat up a bit before the SAC305 paste does).

wulfmans, 2016-10-02:
I just used some old dud pcbs i had loaded them with resistors and caps
and reflowed them.
I made a custom profile for my high temp paste ( 10 years old  LOL  )
worked well.
only problem i had was operator error while applying the paste with the
stencil.
too little on a few pads.

conjure86, 2016-10-02:
Does running from cold, versus a preheated oven make a difference on these profiles?

When I do get this correctly calibrated with my DMM, should I be able to run this from a cold start and reflow properly in one shot?

wulfmans, 2016-10-02:
i never tried from cold. I heated up the oven first.

mikeanton, 2016-10-02:
The oven I usually use, not a T-962, starts with the oven warmed to 80C.  If you do the profile starting from cold, it should work ok, as long as you always start from cold.  Whatever you do for a starting temperature you need to do consistently, or you may get different results.  You do need to watch the temperature rise from cold, so perhaps starting with the oven not too warm would be advised.

Like I said, you will find that different boards behave differently.  It is difficult to create a profile that will work correctly in all situations.  When the PCB is different, a profile that worked on another board, may under or over cook the new board.  This is the down side to regulating air temperature vs board temperature.  The oven must be profile to get the desired board temperature.  That said, all commercial ovens are done this way, as it doesn't require a temperature probe to be on the board during production.  There are some hobbyist built ovens that do regulate the board temperature directly, so they don't have this problem, but instead have the requirement to put a thermocouple on the PCB being reflowed, which can be awkward.

mikeanton, 2016-10-02:
One other note, I found when using a T-962, that it helped to put the PCB on small blocks at the edges (but only the edges), to get it off of the bottom screen, which sinks a lot of the heat away from the board.  Again, whatever method used to support the board, should be done when you profile the board.  Little changes like that, can really affect the temperature seen on the PCB.

conjure86, 2016-10-03:
Thanks yeah I may try some sort of standoff before calibration.

http://www.compuphase.com/electronics/space-holders.jpg -- Like this.

I'll have to find something wide and short enough for this oven.

I'll report back with any calibration issues :)

mikeanton, 2016-10-03:
Sure, something like that would work.  I just use little blocks of metal on a couple of sides, nothing fancy.

GitLang, 2016-10-03:
I just use an M3 x 6 screw through each mounting hole with a nut on the other side. Stops the thermal contact, and it's enough space for double sided component boards usually.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk

Shaische, 2020-06-12:
Hi all, I'm a newbie at this so I apologize for the stupid questions. I just got the oven, uploaded latest firmware, and did cold junction mod (firmware recognizes it). However, when I run my custom profile I made, around its peak, the oven reads 234 C and my multimeter thermocouple reads 200 C. Almost 70 C difference. My PCB is slightly elevated off the tray also and the thermocouple is on it, in the center.

I am completely clueless how to calibrate the TC. How do you adjust the gain/offset? Based on what? Can someone provide a scenario please or explain how they would adjust it based on the two readings I provided above? I would really appreciate it!

mikeanton, 2020-06-13:
Shaische,

You would only adjust the gain and offset values in the firmware, if the oven temperature doesn't match your thermocouple reference.  This would be done by putting your reference sensor very near the oven sensor, measuring the AIR temperature.  Then you can calibrate the oven to match, though in truth this doesn't really matter that much, as you will see below.

That much temperature difference is a bit excessive, but not unusual.  You need to profile the oven for each type of board you are going to run.  This isn't just a one time setup that is done to the oven.  The oven controls the air temperature, not the temperature as seen by the PCB, which is highly affected by the IR of the oven elements, and it is the PCB temperature that we care about.  So, the process is usually to log the data from a thermocouple that is placed onto a copper pad on the PCB that is well connected to the PCB.  As in one that is fairly large, coupled to ground planes, or the mid plane layers.  Run the profile, and check if the logged temperatures match the desired profile for the solder paste you are using.  Continue to modify the oven profile, until you get the desired readings as seen by the PCB.  This usually takes me a number of tries to get right.  Often I'll start with a profile for a board that is similar.  Usually I keep the values for a give board in a spreadsheet, for future oven programming.

When using leaded paste, I shoot for starting the soak at 150C, and allowing the temperature to gradually climb to 175C during the next 90-120 seconds.  Then the peak reflow temperature to reach is 215C.  Measure the time from 183C (solder melting temp) on the rising edge to 183C on the falling edge.  Ideally you want this to be no more than 60 seconds, but this is probable not possible in a T962, as it cools far too slowly.  I'm usually happy if I can keep this to less than 90 seconds.

It probably isn't worthwhile trying lead free paste on a T962, as it is just too underpowered, and too slow to cool down, and higher temperatures just make this worse.  When I ran one of these ovens I never used lead free for that reason, as it had a hard enough time with leaded paste.  Having a proper convection fan in the oven would really help.  In the end I just opted to get a T-937M instead, which is 3300W, runs on 240V, and has a convection fan.  It works much better, and can achieve the proper reflow times, but the firmware is terrible, and doesn't regulate temperature worth a darn.  Someday, when I have time, I'll rewrite the firmware so this can be fixed.

Shaische, 2020-06-13:
@mikeanton Thanks for the long explanation, but this still doesn't help me understand how to adjust gain/offset. And. I fixed my comment - it should've said 300 C, not 200 C. My multimeter reads 300 C from the TC placed on the PCB when the oven reads 234 C. This makes no sense as it's 70 degrees higher than what the oven reads and is such a high difference.

So again, how to properly adjust gain/offset? what's the recipe/instructions?

mikeanton, 2020-06-13:
Well, if it were me, I'd leave the oven off, and measure the room temperature with the oven and the reference.  Set the offset to get this to be correct.  Set the oven to some higher temp, say 150C, adjust the gain to get this reading.  Probably you will have to do this a few times, or calculate the offset and gain required, based on two temperature readings.  The math is pretty simple.  But, do all of this measuring the air temperature as I suggested.  The temp on the PCB is meaningless for calibrating the oven.

Often with a T962, with no convection fan, the PCB will be hotter than the air temperature.  The PCB absorbs the IR from the lamps much better than the air does, and what you are seeing is the result.

Repeat your previous experiment measuring the air temperature instead, and then you will know how far out you really are.

Shaische, 2020-06-13:
Thank you @mikeanton ! That helps paint a clearer picture. So from my understanding at ambient temperature you adjust the offset and at 150C you adjust gain only. I suppose that makes sense. 

I read a lot of reviews that the oven doesn't get hot enough, so I figured that would mean my actual temp would be under the reading, yet I'm like 70 C over. Quite surprised. Perhaps the best mod would be to rip out the thermocouples and route them a different way so they touch the PCB somehow without knocking over components or getting pinched by the drawer?

mikeanton, 2020-06-13:
Another way to look at the calibration, is to adjust the gain first.  Pick two temperatures, and measure the difference between them.  If they are supposed to be 100C apart, adjust the gain until they are.  Don't worry about the absolute values at this point.  Then is should just be a matter of setting the final offset once the gain is set.  Since you are not using a zero point as one of your readings, the gain setting will affect the offset setting.  So setting the gain first will probably be the easiest.

To calibrate the oven you need to measure the air temperature in the same location as the thermocouple on the oven.

How hot the PCB gets depends on the PCB itself, and it is different for every board.  This is especially the case for an oven without a convection fan.  Make sure you are measuring the temperature of a pad, and not the solder mask.

Your min fan setting is probably too high.  I seem to remember running something like 40 or so.  Since that fan pulls in cold air, that cools off the air temp, but many not cool off the PCB temp.  Having this fan does help even temperatures out, but the penalty is you are heating up more air, and this makes the performance of the oven even worse.

The oven is likely not really designed for a lead free process even though they might sell it as capable of that.  I generally find that the Chinese suppliers over estimate the capabilities of their products.  The T-937M that I'm using now certainly doesn't come anywhere close to the temperature stability claims that they've made, and it appears that they are not even using PID control of the elements.  It appears more like bang bang, where they just turn the elements off once the temperature is reached.  It then overshoots by at least 10C, triggering the cooling fan, which then causes an undershoot by 10C.  Surprisingly, even though the air temperature is oscillating like that, the PCB temperature only moves a little bit, but it does tend to make the oven a bit unpredictable.  That said, I've probably run 500 boards though this oven so far, and it has performed fairly well, but it is a pain to profile, though still much easier than a T962A, since it has PC software that drives it, and allows profiles to be saved.

To adjust the soldering profile, you measure on the PCB, and adjust the profile to match the desired profile.  This is the way all reflow ovens work.  While putting the thermocouple on the board would probably work, that's not ideal for the reasons you outline.  Good luck with getting that to work reliably, without long wires, and since you want the thermocouple located on a pad that will be covered with solder paste, it gets even more difficult.  You would also have to mount the thermocouple to every board, or use a spare board.  There's a reason that ovens are built as they are.  The T962 is not unique in this regard.  So, just get used to profiling the oven.

The T962 oven is under powered, and suffers from the lack of a convection fan.  Don't plan on being able to use the full tray area, and focus on just the part in the middle.  I could never reliably reflow a 12" long board, without an inch from each end not reflowing (on a T962A, with the board placed diagonally).  Though, this was a 6 layer board, with heavy copper.  It didn't help that there was a BGA part within that last inch.

Do you have a T962, or a T962A?  If it is the former, I really wouldn't count on running lead free paste, since it is much lower power.

Shaische, 2020-06-13:
Well, as I've outlined my experience, my PCB gets way hotter than required. It reaches 300 C. I am using this reflow solder, which has a low melting point of 220 C: https://www.digikey.com/products/en?keywords=TS391SNL50-ND

I have the T962. The T962A, which comes with a vent hose attachment on the back looks like a fake. It still pushes air in I bet rather than vent it out. But that's out of scope of this discussion.

I will try running my fan at 40 instead of 76. You make a good point it may not cool the PCB as much as the air since the tray blocks airflow considering the PCB sits inside. I will be reflowing 4 layer and 2 layer 1.6mm boards primarily with pours.

So now you're saying I should adjust gain at 0 and 100 C? Not offset at 0 and gain at 100 C ? Is gain more like larger adjustment and offset more like finer tuning?

mikeanton, 2020-06-13:
Yes, I would expect that the PCB might be hotter than the air temperature.  That is not uncommon.  Whether or not 70C is reasonable, I can't say.  It is due to IR absorption of the PCB.  Different solder mask colors, and PCB copper pours change the behaviour (black is likely the worst), which is why you will find that every different PCB will need to be profiled.

It's not just that it will cool the air more than the PCB, since it isn't really the air that's heating the PCB.  It's the IR from the elements, that likely heats more than the air.  So, even though inlet air will cool the air, it likely cools the PCB far less (within reason mind you, as too much airflow definitely reduces the ramp rate of the PCB).

That's SAC305 paste, which is what I run as well, though I'm using the Loctite GC10 paste.  The GC10 paste is really nice, but the jar size is a bit large for a hobbiest.  I find I can use most of a jar in a year if I'm busy, so for me it is worth it, and is actually a bit cheaper than what I was paying for leaded paste, once refrigerated shipping was factored in.

The T962A oven I have does not come with a vent hose.  It looks exactly like a T962, but is higher power, and is a bit larger.  The T937M that I also have has a vent hose, and all the fumes from the oven come out here.  It is a much better design, with an inlet fan, and an outlet fan (push pull arrangement), so it move a lot more air, with a convection fan.  The oven is sealed except for the inlet and outlet holes, so it doesn't leak smelly air everywhere.  You will find that you need to find a way to run the oven in a well ventilated area if you are doing much soldering.  I have a makeshift fume hood that I used for this purpose, and it was essential.  I couldn't believe how much smoke comes out of the bottom of the T962A when I started using it.  The fluxes from lead free pastes are nasty too which doesn't help.  This idea of lead free being more environmentally friendly, really only applies to the disposal of the product, and not to the manufacture of it, which is probably far worse than rosin based pastes.  I couldn't believe the warning labels on the GC 10 paste.  It sounded way worse than the Kester leaded paste I've always used.

No, I'm not saying to adjust gain at 0, and 100C.  I'm saying that the gain affects the offset, since you can't calibrate easily at 0C (or maybe even 0 Kelvin, depending on how this is handled in the firmware).  I am saying to pick a spread between two temperatures, say room temperature, and 150C.  As in measure the temp at room temperature on the oven and your meter, and then do the same at 150C according to the oven.  Set the gain of the oven to get the actual change required, i.e. if the room is 20C, and the oven is at 150C, that's a difference of 130C.  If your meter doesn't indicate that there was a 130C change, then figure out the gain factor, and put that in.  If you run the same test again, it should be close.  Then at whatever temperature you like, adjust the offset of the oven until it matches the meter.  Now you should be calibrated, but you might want to check this.

Note that I haven't tried this, as my oven was really good, but then I didn't just do the cold junction mod.  I used one of the MAX31850K 4 channel thermocouple boards that reads the thermocouples directly, and has cold junction compensation, and an A/D built in.  That works really well, as it bypasses all of the Chinese electronics attempting to do that job badly.  You can get the PCBs from OSHPark for just a few dollars, and I sampled the parts from Maxim, so it was pretty inexpensive to do that way.

Shaische, 2020-06-13:
Referring to this T962A: https://www.aliexpress.com/item/32324022514.html?spm=a2g0o.productlist.0.0.778e366dLdU92R&algo_pvid=b4981eb3-8ecd-43a2-ba3d-74af3dd7d443&algo_expid=b4981eb3-8ecd-43a2-ba3d-74af3dd7d443-42&btsid=0be3743615920289384772786e158c&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

You can see that they have a vent attachment in the back. Though I suspect the fan is pointed the wrong way (same direction as T962), therefore not pulling any fumes out. I say this because I see the sticker of the fan in the images and my model T962, the sticker is also visible like that. So same direction fan. If this is true, the people who designed this reflow are really dumb and don't bother testing lol.

Understood about the gain I think. I started playing with it at 100C. At first had the following readings with just offsets set and gain of 1:

L: 115    R: 86    M(multimeter): 120

Now adjusted:
L: 102    R: 98.5   M(multimeter): 109

So, seems like they're getting closer, just need to fine tune it more. Now I just need to build a vent attachment for all the holes under the reflow oven. I was thinking of going with 2 inch vent rather than 4 inch to save desk space. Thoughts? I don't plan to have the 2 inch be very long.. maybe 3 ft most as it will sit right by the window, so perhaps 2 inch hose won't constrict it too much. Otherwise a 4 inch would require a bigger service loop.

mikeanton, 2020-06-13:
Yes, that T962A is different than mine, which is probably fairly old.  It would make a lot of sense for that fan to blow out, rather than suck air in as you say it might.  If it blew out, it would suck in through the vent holes in the bottom, which would be better, but as long as the holes are there, it's likely the fumes will leak out anyhow.  My oven might do that as well through the cooling fan, which is not vented.

One thing not to do is suck on your vent pipe.  This needs to vent without a pressure drop, or it will pull cool air into the oven anyhow.  I figured that out with my T937M, as it has a vent pipe, and I have an exhaust fan, but connecting the two pipes together didn't work well.  Instead, I leave a 2" gap between the pipe ends, and don't run the exhaust fan at full speed.  If I start to see the fans in the oven spin when the fans are off, I know I'm sucking too much air through the oven.  I did try it with the pipe connected, and the oven couldn't heat up well at all anymore.

It looks like the calibration wasn't as far out as you thought initially (unless the gain is way off, which will show up at higher temperatures).  If you don't get the gain set first by looking at the differnece between two temperatures, you will have to iterate back and forth more times.  Adjusting either the gain or the offset can get one temperature set correctly, but that doesn't really help much, because as soon as you change the temperature, it will be incorrect again, and you will have to adjust both gain and offset again to fix it.

I wouldn't plan on restricting the airflow much.  Using a 2 inch pipe is likely much too small.  You need to keep the airflow as high as possible, of the oven won't have any chance to cool down fast enough.  The grain size of the solder is related to cooling, so you don't want to cool too slow either.  You don't want to keep the semiconductor parts at the elevated reflow temp for any longer than you have to (within reason).  Slow cooldown also makes for very long cycle times.

I've seen some people set the oven on a metal tray to act as the pickup on the bottom side.  That seemed like a reasonable solution.  Don't underestimate how hot the outside skin of the oven gets when it is running for a long period of time.

One other thing to watch out for.  I found out the hard way that the fuse holder on the back is not rated for the current the oven pulls.  Now this was on the T962A, so you might be ok, but mine melted down at one point, and the oven quit working.  I ended up replacing the switch with a circuit breaker switch, and took out the fuse/power entry module, and replaced it with just a power entry module.  There don't appear to be any fuse/power entry module combinations that were rated for high enough current for the fuse rating.  The standalone PE modules come in much higher ratings.  Also, check the grounding on all the metalwork if you haven't already.

Shaische, 2020-06-13:
Got it, will use 4 inch. I'm tempted to just see if this fan can be flipped to suck air out. I poked it with a screw driver and it has metal fins and the body seems to be metal too. If the air is cooler than the PCB, then perhaps it's ok to use the fan on the T962 to suck air out? I'm tempted to try but then again, I may end up melting it and it's labeled as 110V...

mikeanton, 2020-06-13:
The air isn't always cooler than the PCB (eventually they should reach steady state and be more or less the same given enough time), so you should probably expect that soldering temperature air will be hitting the fan.  My oven does run that way though, but I'd expect that the fans will have shorter life, since they are not really rated for those sorts of temperatures.  Now, I did notice that the fans on my oven are actually high quality ones, and not some Chinese variety (maybe take a look at yours), so that may make a big difference to service life.

GitLang, 2020-06-13:
> I'm tempted to just see if this fan can be flipped to suck air out

Don't even think about it. It sucks air in to cool the oven. The only way you can control the temperature of something above ambient is to have both heating and cooling, and proportionally apply one or the other depending on if the controlled object is above or below the desired temperature. Without the fan sucking in ambient air, the only source of cooling would be natural loss to the environment which is one or two orders of magnitude less than forced cooling. So leave the fan sucking air in, and don't change the fan - the one fitted is a good quality industrial fan that can handle the temperature. If the fan were reversed it would suck through the side vents in a very restricted manner and probably suck the cool air stright out over the top of the PCB.

If you want to improve that area, look back in the archive and find the directional baffle somebody here made. It basically divides the air flow into 6mm square cells, the idea being to remove the turbulance and stop the tendency of the air to preferencially cool the left (i think) side of the oven.

As others have said the gain and offset pots shouldn't really need messing with. You're better to make any adjustements in software, especially in the PID control which the comments in the source admit it is very rudimentary.

I didn't see where you got, the 70C difference from, the figures you gave showed around 38 I believe. That is on the high side but it is not unusual for this oven to show 30C difference between the left and right thermocouples.

Whatever you set the temperatures to in your profile, the temperatures on the PCB will be different and will vary across the board, Eg, an 0603 resistor may get hotter than a wacking great SM inductor, as a lot of the heating is actually down to radiation rather than conduction from air at a specific temperature. Reflowing a board correctly is more of an art than a science and in inductry you will find people tweaking a profile carefully until it is the best compromise for that specific PCB, rather than going by set temperatures.

Shaische, 2020-06-13:
How do you make adjustment for the PID control in the software (I think you mean firmware?)? I'm no programmer, so any help appreciated.

mikeanton, 2020-06-13:
@GitLang I believe you mean it would suck the air in through the bottom vent slots.  The side slots just ventilate the upper chamber where the electronics are, and are not connected to the oven chamber below.

Pulling air in through the bottom slots may actually be beneficial, since that would be cool air coming right into the PCB.  Since hot air rises, sucking from the high side of the oven helps keep it moving in that direction, rather than trying to push the hot air down.

What I've noticed about these ovens, is that when the cooling fan turns on, the actual PCB temperature increases for a bit before it starts dropping.  This is probably due to the heat being removed from the metalwork of the oven itself.  I find that I have to set the oven temperature a bit low during the reflow spike to account for this.

Shaische, 2020-06-14:
Can anyone recommend anything off the shelf that can be retrofitted under the oven to then connect to a duct? I was thinking of using a 1 inch oven tray under the oven and have it protrude out the back and then connect a 4" duct to it going up the back but can't find anything that is 1x7x12ish. 

I was able to luckily find a very convenient rack that fits inside the oven like a glove, which allows you to elevate your PCB so it doesn't touch on the tray. No tacky screws required :) https://www.amazon.com/gp/product/B078Q4CPR5/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

If you can suggest something for the exhaust, please do! I'm trying to avoid having to cut stuff as much as possible.

mikeanton, 2020-06-14:
On my new oven, I use tensioned wires from front to back to support the PCB.  I did this by putting in aluminum standoffs with long screws into the slots on the front and back.  Then I made up some wires using buss wire, and crimped ferrules, to make loops in the wire ends.  A spring on each wire keeps the tension.  It works well, and adds pretty much no thermal mass to heat.  You can a picture in my forum post here: https://groups.google.com/d/msg/desktop-pick-and-place/nfSHd17XZZ8/BmIBYPNWCwAJ

GitLang, 2020-06-14:
> software (I think you mean firmware?)

No, I meant software. Firmware was originally used as a term for software in non erasable mask programmed ROM's that could not be changed, hence "firm". Hardware was already in use :)

GitLang, 2020-06-14:
> 
> 
> @GitLang I believe you mean it would suck the air in through the bottom vent slots. 

Yes, that is what I meant, sorry. As you say, the side vents cool the top electronics compartment.

I could see "bottom suction", if you'll excuse the term, being of benefit iff the air inlet was above the PCB's and well baffled/laminar. As it stands, it seems to work ok for me, the two big problems being left to right temperature disparity which an air guide is supposed to fix, and bad PID control leading to overshoots etc.

MaikBoe, 2020-06-14:
Hello, I´m still a newbie in this forum as well for the improvements of this oven.
 This is the first step for me with SMD and reflow, so I´m not familiar with everything.
 I got the oven 1 month ago and made the basic changes (Kapton Tape; grounding; Power Switch wiring; UE Firmware; cold junction mod; system fan PWM mod; high temperature Silicone for the glass)
I ordered the MAX31850 4 channel board 3 times from OSHPark (minimum quantity)

I am interested in the part of the rear fan from the 2020 modell.
In the picture it shows the label of the fan, but the manufacturer has this type only one way.
So i´m not sure if it really sucks hot air out. Some posts are about changing the direction of the fan, but I´m still unsure if it benefits.
Pro:
 - better cooling in the end of the process
 - the material under the oven must not be heat resistent

Con:
 - min fan speed from the UE Firmware changes ensures that the heat is distributed more evenly, but
   if the blowing direction of the fan is reversed, it sucks cool air from the bottom in, instead of 
   distributing the air evenly
 - the wires of the fan are on the inside where the heat is, especially when it is running.

I changed the direction, but the first Con point is the most important issue for me to switch it back.
In my opinion the best solution is sucking cold air through the back with min fan speed till the peak of the profile. From the peak of the profile sucking hot air out through the back. Perhaps 2 fans in the back. One with diffrent direction on the outside of the rear frame and switching with a relay when the peak of the profile is past.

I´m looking for a k-type datalogger to check and adjust my own profiles based on the pcb´s i take.
In the posts above it was mentioned.
Anyone got some hints?
The four I found are:
https://www.amazon.de/dp/B0874292P4/?coliid=I15PAFEN2SR1PS&colid=1NGWFT317OIO2&psc=1&ref_=lv_ov_lig_dp_it

https://www.amazon.de/dp/B0872S1DG4/?coliid=I57O1GI927UH1&colid=1NGWFT317OIO2&psc=1&ref_=lv_ov_lig_dp_it

https://www.amazon.de/dp/B07TVRZRVC/?coliid=IX7Z4UGZ4Q8I3&colid=1NGWFT317OIO2&psc=1&ref_=lv_ov_lig_dp_it

https://www.amazon.de/dp/B07TTRF8VR/?coliid=I3DQQV2QW8G53B&colid=1NGWFT317OIO2&psc=1&ref_=lv_ov_lig_dp_it

southern-boy, 2020-06-14:
We can't see that web page since it says you have to be a member and 
apparently is limited membership.  Do you have a different link to those 
pictures?

Jim Pruitt

mikeanton, 2020-06-14:
@southern-boy Sorry about that, I forgot that that group requires membership.  Here it is:

[image]

GitLang, 2020-06-15:
@MaikBoe - seriously, do NOT reverse the direction of the rear fan. Beside the oven not functioning as well, it will destroy the fan. Think about it - at the end of a lead free cycle the oven air is maybe 250 degC. With fan reversed, it will now go full speed and suck that 250 degC air through a fan rated for about 70 degC max.

The design as it is will work fine. What it needs to prevent side to side temperature diffewrence is an air diffuser next to the fan. That is the word I was struggling for yesterday. The example a member here made was a hollow square tube with brass walls, about 100mm x 100mm section and about 100mm long. Inside the tube were many vertical and horizontal sheets of brass , each about 100mm x 100mm which divided the tube into many small square tubes in the direction of air flow. Best is to make it out of thin brass or copper sheet soldered together. Then you can put it inside the oven so the fan blows through it. The airflow is then almost linear rather that a swirling cyclone.

mikeanton, 2020-06-15:
@GitLang, I don't think it is as big a problem as you are making out to be.  The T-937M I have has two fans, one blowing cold air in like in the T-962, and the second one sucking the hot air out, blowing it into and exhaust pipe.  I was surprised that the fan would last under these conditions too, but I've run hundreds of reflow cycles at this point, and so far it is fine.  Now perhaps the additional cold air being forced in helps.  The air is ducted as well, with cold air blowing directly into the PCB chamber, and hot air is pulled from below the PCB tray, and up into the exhaust tube)  This oven cools down far faster than any I have used before (though it still has the PCB temperature increasing at the start of cooldown).  They do use a good quality Pabst fan in this oven that is rated for 85C.

The T-937M has the top mounted convection fan as well, and yes, it does help to even out the temperature.  For some reason though, they turn it off during cooling, which I can't understand.  Someday I'm going to wire it to a switch so I can just leave it on to see if that makes any difference during cooling.

Shaische, 2020-06-15:
What's the fan model # that's used on the T962 to force air in from the back? Anyone know? It's not easy to take out the fan to check so I'd prefer not to mess with it, but I am thinking of trying to reverse it to suck air out or replace it with one that can withstand the temp. The one I have seems to have metal fins though.

mikeanton, 2020-06-15:
@Shaische, for some reason, AC powered fans seem to commonly have metal fins.  I personally haven't ever seen one that didn't, though they do exist.  It probably doesn't matter what the fan part number actually is, at it won't have a temperature range that extends past 85C, and might be more like 70C.  The oven is most likely to use a metal fan, as that location actually gets quite hot, just due to proximity to the oven chamber, even without having oven air moving through the fan.

The bottom line, is that fans are not rated for the temperatures that would be seen when sucking hot air out of an oven, but it is done in some ovens.  It is possible that over the time that it takes to remove all of the air, that the fan temperature doesn't really increase all that much, but I haven't measured this to be sure.  That might prove to be a useful experiment.

southern-boy, 2020-07-04:
I ordered this same tray
https://www.amazon.com/gp/product/B078Q4CPR5/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1
for my new T962 and it will not fit.  It is slightly too high so it will 
not let the T962 tray slide back in if your oven tray is placed on the 
existing rack/slide of the T962.  Did you trim yours down a little so it 
would fit and slide in and out?

Jim

Shaische, 2020-07-04:
> I ordered this same tray https://www.amazon.com/gp/product/B078Q4CPR5/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1 for my new T962 and it will not fit.  It is slightly too high so it will not [...]

You need to push the legs from the rack into the slots of the T962 tray or clip them. I pushed them in.