UnifiedEngineering/T-962-improvements #98: Documentation for controlling the oven using a PC

rharrison, 2016-10-07:
Did jieter ever document or provide some software to control the T962A oven?

jieter, 2016-10-07:
No documentation apart from the things the software says over the tty. Should be pretty easy to understand once you connect to it.

rharrison, 2016-10-09:
I'm hoping to be able to upload custom profiles saved to the PC as well as log and graph running profiles. I'm working on a new addin board which plugs into the isp header and will provide USB connectivity. 4xtemp and fan control for the noisy little on as well as provide a control and power for an internal circulation fan should someone want to try this.

KtroniX, 2016-10-19:
Hey 
if you connect to this oven with Terminal Baud Rate:115200 8bit parity none , sytop bits 1 
and just asking it: ? or help

Oven controller answers you:

T-962-controller serial interface.

 bake <setpoint>       Enter Bake mode with setpoint
 help                  Display help text
 list profiles         List available reflow profiles
 list settings         List machine settings
 quiet                 No logging in standby mode
 reflow                Start reflow with selected profile
 setting <id> <value>  Set setting id to value
 select profile <id>   Select reflow profile by id
 stop                  Exit reflow or bake mode

rharrison, 2016-10-20:
Thank you that's perfect

bake Enter Bake mode with setpoint
setting Set setting id to value
select profile Select reflow profile by id

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk