UnifiedEngineering/T-962-improvements #99: Designing a new hardware module to enable the functionality provided by the software changes.

rharrison, 2016-10-27:
Firstly a big thank you to UnifiedEngineering for the firmware rewrite.

I have spent quite a bit of time reading all about the changes people have made to improve the oven and I decided to try to make the hardware changes easier by designing this board.

http://gerblook.org/pcb/BrakV6Z7HyfLLxrfhcNuEd#front

http://i.imgur.com/zBS3u0X.png

http://i.imgur.com/5sX9DTj.png

This is an advance version of the UnifiedEngineering TC board

- This will allow easy initial and future reprogramming of the controller.

- Ability to monitor profile data every second during re-flow

- Ability to easily add new re flow profiles

- Motor control for an internal oven air circulation fan

- Data to control the mini fan connected to the side of the unit using the cold junction temps on the sensors.

- Supports up to 4 thermocouples with cold junction compensation. Moving of the two exisiting ones and the option for two more for monitoring PCB surface temperatures during re-flow.

christaylor99, 2016-11-10:
This is a good idea to lift the cut and splice changes into a single interface board. I like that this includes the USB interface to the ISP connector and System Fan (12v) connection. There have been several comments when using Flash Magic to upload firmware to the different variants and MPU chip differences. I assume the MCP2221 will be programmed to allow for the variation in signal interface behavior to the ISP connector.  Please make a schematic available and possibly an OSH Park link for the board when your testing is complete.  Do you plan to add the MOSFET (2N7000) to your board for the PWM System Fan control?

rharrison, 2016-11-10:
I have added the transistor fan control to the board and a dc brush motor controller for an internal circulation fan. I'm expecting a delivery of 40 boards in the next few days and will report back how I get on with testing. I'm working on some client software to allow you to graph the flow in realtime and to run your own custom profiles.

I wasn't aware of the programming issues.

The nice thing about the MCP2221 chip is that I have wired up two of the GPIO pins to the reset and n_isp pins so that the oven can be put in boot loader mode by sending command to MCP2221 chip and not having to bugger about with manually making the lines low.

I might be looking for volunteers to test this controller out on alternative architectures too to see if there are any issues to iron out.

Once done I plan to sell the boards as blanks pretty much at cost plus a small bit for my time and post or fully assembled and tested for the cost of the components and 20 pounds towards my time. I will also document and video the installation and use of the board and I'm happy to provide the gerber files to people who want to print their own board.

Should be putting an update out in about a week after testing has been completed. This has taken me about a month of work to research, test, design and bring together the PCB more to do with my learning of Kicad and basic electronics but hopefully this will make life easier for people going forward.

Robert

BewilderedAgain, 2016-12-18:
HI Robert,

How is progress. I have an oven arriving soon -I's certainly like to pop one of your boards in there.

Nick

rharrison, 2017-01-03:
Hi Nick,

I'm nearly there on revision 5 of the board and I think that's it. I will let you know how I get on over the next two weeks and then hopefully we will be there.

PS I will be selling the un-populated PCB board for 20 pounds and the populated and tested on for 99 pounds. 

This is due to the time it takes to make and test the board and source the components and the fact that I doubt I'll ever sell more than 20 of these.

Purchase info can be found [Here](https://www.theicarusproject.co.uk/index.php?route=product/product&path=59&product_id=52)

Installation video can be found here. [Here](http://www.youtube.com/watch?v=svrDu1Apf7g)

Robert

BewilderedAgain, 2017-01-16:
Glad to know you are making progress. Robert. Well done.  

I have upgraded my oven with the cold junction fix and improved firmware.  I then calibrated the probes and did some testing.  I very quickly learned that the temperature sensed by the thermocouples is lower  by around 25C from the board temp. The tells me that 10 bit ADC resolution is not top of the list of problems:)  The location of the thermocouples is the biggie.  Hopefully your board, having four t/c inputs will solve this if I locate two of them them on the tray and mounted onto scrap FR4  that can be place near the target board.  For the moment I iterated a profile using a meter with thermocouple as the standard.    I note the 'stirrer' fan seems to be responsible for the left/right difference.  I wondered if a re-design of the fab baffle might help somewhat.

Also, on earthing, I bonded all the metal surfaces to safety earth as they were all insulated from one another by paint.

Oh,  is there a BOM - did I miss it?

mikeanton, 2017-01-17:
BewilderedAgain,

Yes the oven needs to be profiled, which is to say calibrated to the specific PCB design.  Most commercial ovens that I have seen measure the air temperature, but what we really care about is the temperature on the PCB.  So, the oven must be profiled to get the board temperature to match the desired reflow profile.  The oven settings will often be different with different PCB designs.  This is just part of the setup process that is required to get a given board design to reflow properly.  The entire industry runs like this, so it is not unusual that this oven works in the same way.  Usually, this takes a few iterations to get correct as you have found.  Note that for this to work properly, the temperature within the oven does need to be fairly even, so if that is not the case, it would be worthwhile finding a way to fix.

I have seen some hobby controllers, that actually put the thermocouple on a PCB, and regulate the oven to that temperature.  While I don't like that since it is fairly fiddly when doing a production run, it is an option for someone that just wants to run a few boards.

rharrison, 2017-01-17:
I'm working on chnageing the firmware to do auto profiling. Using the data from the pcb attached thermocouples to produce a custom profile for the board. You set the max temp (reflow temp) and the period you want to hold that temp for normally a few seconds. Run the oven and the profile will be created and saved for you. That's the theory need to put it into action.

mikeanton, 2017-01-17:
You might want to add the soak temp, and time as well (the stage before reflow, that activates the flux).  Auto profiling would be really nice though.

rharrison, 2017-01-17:
I'll discuss this more as I get around to making the changes. I'm sure it's doable. Will also need to take into account the lag between turning the lamps off and the PCBs continuing to rise in temperature.

I've had really good success with this oven so far and it's about the 10th of the price of a commercial oven. Though I do think the modifications are necessary so it can be profiled properly.

rharrison, 2017-01-17:
I'm working on changeing the firmware to do auto profiling. Using the data from the pcb attached thermocouples to produce a custom profile for the board. You set the max temp (reflow temp) and the period you want to hold that temp for normally a few seconds. Run the oven and the profile will be created and saved for you. That's the theory need to put it into action.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk