UnifiedEngineering/T-962-improvements #100: Where is the Max31850K TC support in the v0.5.1 source?

christaylor99, 2016-11-01:
I am in the process of building the 4-port UnifiedEngineering thermocouple interface and started looking through the source code for the branch support mentioned in the writeup before installing the new interface board in the T962 oven.
Better thermocouple interface
Werner Johansson  Nov 5, 2015_
https://github.com/UnifiedEngineering/T-962-improvements/wiki/Better-thermocouple-interface
"The MAX31850K utilizes a 1-wire interface as well which makes it a lot easier to interface to the existing board. Master branch code is already capable of detecting and using this interface if present."
Thinking it was in the OneWire code I find only mention of the DS18b20 and DS1822 sensors in the v0.5.1 code.  Is there another prebuilt hex file that contains the Max31850K?

christaylor99, 2016-11-05:
I looked closer at a smudged copy of the Maxim Max31850 datasheet and saw the family code that looked like "38" and was looking in the onewire code in the src code and saw TC listed as "3b".  Going back "to the well" so-to-speak, I looked at the Maxim site at the datasheet where the family code of 3b is the identity for the MAX31850. So it appears that the 31850 is supported in the v0.5.1 hex as "TC".  Perhaps the next edit should report the chip in the same manner the DS18B20 and DS1822 sensors are currently identified.

closed 2016-11-05 by christaylor99

christaylor99, 2016-11-05:
Suggest the "TC" model identified in the onewire.c src be edited in the next release to identify the family code for the MAX31850 (3b) as the current onewire sensors (ds18b20 and ds1822) are identified at startup.

darkroomai, 2024-02-19:
How do the addresses (0-3) of the 31850 map to "Left", "Right", "X1" and "X2" that gets displayed when the 31850s are detected?  One picture I found has the built-in ones hooked to 2 & 3, and another has them hooked to 0 & 1.