UnifiedEngineering/T-962-improvements #103: Anyone have a copy of the original firmware as a hex file

rharrison, 2017-01-03:
Hi I'm doing some videos about the t962a and could do with putting the original firmware on the controller. I don't suppose anyone has a copy ?

rharrison, 2017-01-05:
I have ordered 4 controller boards from PHUIT so I will take a copy of the hex file when they arrive for prosperity. IIRC the chip is not locked.

xnk, 2017-01-10:
> Hi. I'm interested too by the original firmware. Since I did the hardware upgrades and flash with prebuild 0.5.1 firmware, the hoven (T-962C) does not heat enough to melt SAC305.

That is really odd. Was the T-962C confirmed to have the same IO setup (I’ve never seen that oven in action). I should be able to hook you up with the original firmware, as long as it’s identical with the one in the T-962 oven.

Have you tried turning the fan off completely during reflow to see if it heats enough then? (as the fan is off in stock fw) Just wondering how much margin they actually have in the T-962C. I know the margin is really small on the T-962 and slightly better in the T-962A.

/wj

vmarguerie, 2017-01-11:
Even if i pretty sure the stock fw won't be the solution, it should be interesting i test it again, now i've done some adjustments. Can you send it on the forum ?
With the fan off, there is no visible difference ! 

Regards 
Vincent

romansavrulin, 2017-12-14:
@rharrison Are you managed to dump the firmware? Which tool did you use?

MikhailNatalenko, 2017-12-15:
To dump hex, hust save file in Flash Magic 

_File > Save As_

romansavrulin, 2017-12-16:
@MikhailNatalenko Can you please attach firmware file here if you have it?

MarcoAntonini, 2018-01-08:
I uploaded successfully today on my T962A with LPC2144.
Everything works properly with this firmware!  if necessary I dumped the original firmware.

katmani, 2018-07-19:
[T-962A_original_ID0x0002FF12_BLv2_13.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/2209734/T-962A_original_ID0x0002FF12_BLv2_13.zip)

katmani, 2018-07-19:
Thank you! from Russia! New firmware launched
[image]
[image]
[image]
[image]

rharrison, 2018-07-19:
Nice one

[img_6059]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41431664%2F42944929-ab56495c-8b6f-11e8-9f00-9ca5a2d7ffbe.jpg&data=02%7C01%7Crharrison%40hgf.com%7Cd99cc891aa9248ce023a08d5ed7a9542%7Cbd4a58fa2d194bfa921bf6f222043055%7C0%7C0%7C636676033110488001&sdata=9ObuuEBmEI5B1CkC7t5ZM5f43u%2Bqx9ciWUC3iIK%2Fxy0%3D&reserved=0>
[img_6045]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41431664%2F42944926-aaef772c-8b6f-11e8-9de1-0e855333c75b.jpg&data=02%7C01%7Crharrison%40hgf.com%7Cd99cc891aa9248ce023a08d5ed7a9542%7Cbd4a58fa2d194bfa921bf6f222043055%7C0%7C0%7C636676033110488001&sdata=Zq17wSBPQdfb9K%2BdzfV32EfnJw%2FHWWkkF4M8CUTQI5w%3D&reserved=0>
[img_6046]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41431664%2F42944927-ab12a756-8b6f-11e8-99f0-91a00eb8b54f.jpg&data=02%7C01%7Crharrison%40hgf.com%7Cd99cc891aa9248ce023a08d5ed7a9542%7Cbd4a58fa2d194bfa921bf6f222043055%7C0%7C0%7C636676033110488001&sdata=DYvuFKJkeoiEfvAYgnRP5SNVWw6Wrhp3uXUKnyl3FC0%3D&reserved=0>
[img_6048]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41431664%2F42944928-ab36bf56-8b6f-11e8-823c-8a13501f1eea.jpg&data=02%7C01%7Crharrison%40hgf.com%7Cd99cc891aa9248ce023a08d5ed7a9542%7Cbd4a58fa2d194bfa921bf6f222043055%7C0%7C0%7C636676033110488001&sdata=KBWXeB44BbEjgp1DsEbr%2BR%2FVfvCWFgfgrMrE3yY4UwY%3D&reserved=0>

DomoticX, 2020-01-15:
Downloaded the firmware from T-962 unit:

Puhui T-962 - Firmware original NXP LPC2134

[Puhui T-962 - Firmware original NXP LPC2134.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/4065483/Puhui.T-962.-.Firmware.original.NXP.LPC2134.zip)

Have fun!

jwestmoreland, 2020-04-06:
I'll add the file I downloaded from my unit:

[LPC2134-20200405-021352.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/4435599/LPC2134-20200405-021352.zip)

YMMV,
John

DomoticX, 2020-04-06:
> I'll add the file I downloaded from my unit:
> 
> [LPC2134-20200405-021352.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/4435599/LPC2134-20200405-021352.zip)
> 
> YMMV,
> John

From wich unit "model" is this?

jwestmoreland, 2020-04-06:
Hello @DomoticX,

T-962A - circa 2013 - but it does have the same processor as what's discussed on the site here.

Regards,
John