UnifiedEngineering/T-962-improvements #104: Knowing you're in bootloader mode? Trouble flashing

skeezix, 2017-01-06:
I've got the T962A, with the single-board LPC2134 driver inside. (Second pcb is the LCD.)

Using a FDI (Future Design) USB-ICP-LPC2K (USB->breakout) ISP tool (which is listed as supporting the LPC2134 :)

Tried doing the manual bootloader mode, but didn't seem to help; realized the "F1" trick, so hold that down while powering on..
- if a normal power on, there is the 4 item menu
- when holding F1 and powering on, the screen is not blank, but has a graphic

When using Flash Magic, I get two errors.. either the can't autobaud, or if it autobauds, it gets a read/transfer error.

If I set 19200 it works very nicely in Flash Magic's Terminal; using variants (9600 say) it autobauds okay, but still read error. 

Question - how do you know you're in bootloader mode and working?

- should there be a graphic or blank on lcd?
- the Terminal at 19200 always echoes correctly - if I hit 'a' I get an 'a' back, etc; tried ? and return; tried whole keyboard, it echoes okay; at other bauds, it echoes partially okay, so I figure poor; I don't get anything back other than echo .. I had thought when you do ? you'd get a pile of stuff back
- given the Terminal is echoing, I know that GND and the RX/TX are hooked up (and not inverted)
- when booting with F1 down, the oven lamp doesn't seem to go on, so I think its in bootloader

I've only hooked up RT, TX and GND to the ISP. I've also tried with hooking up RESET and nRST(to the ISPs PO.14 or whatever that was); the variations sometimes cause blank or garbled screen as it doesn't init the LCD and is held in reset or the like ..

... not sure if ISP is working okay or not? seems to be, but at this point, I'm questioning all parts of the surface area :)
.. assuming ISP is okay .. what wires do you actually need to do the flash?

I'm more into STM and such, so just need this to flash once and call it a day :) Kids crying, must run .. sigh. Thanks for your thoughts!

skeezix, 2017-01-09:
okay .. in the Terminal, it seems to echo  near anything, when in fact no echo is going on.

Putting an o-scope on the Tx and Rx ..
- With ISP Tx to T962 Rx, and Rx to Tx of course
... I can see the Terminal sending values out, correctly
... when I tell Flash Magic to start, it seems to be sending out a "." or something first character
... I'm not getting any trigger on the line back from T962 to ISP
Flash Magic seems to say autobaud is okay for some bauds and not others, so somehow it knows; I know in the Terminal I did get some junk (garbage) back for some various combinations of things, but never actual readable values.

I'm not convinced holding F1 down actually goes to bootloader mode .. please advise? (I see a graphic when using F1 on power up); if I manually reset by grounding the GND pin on T962, there is no graphic. 

If I try to manually trigger bootloader mode, I can get some garbage back on terminal, but seems wrong baud or something .. but ISP surely not getting anything.

The crystal is correct as per everyone else (11.0..whatever).

Sure seems like either watchdog is on so I can't get to bootloader period, or a baud mismatch always, or something.

(using Windows 10 and Flash Magic.)

Will try other software combinations when I can .. I don't have a USB->serial beyond the ISP, and need yet to try it under linux.)

skeezix, 2017-01-09:
Ah, realized the FDI ISP is really just an FTDI anyway, so it works fine for linux and as a USB->serial. Fired up virtualbox with a debian image and yep, I can talk out the serial.

Still no response from T962, using minicom on linux, using an assortment of baudrates. It just sits trying to 'synchronize....................', in lpc2isp tool..

The t962 menu works okay.

1) Wonder if theres a way to verify the firmware version from the menu
2) Not convinced I'm getting into bootloader mode .. is there a time constraint with how fast to toggle the lines? and the F1 menu .. maybe doesn't work to go to bootloader mode.

Very very odd.

sivu, 2017-01-10:
If you see graphics then you are not in the bootloader.

skeezix, 2017-01-10:
I'm wondering if the F1 is wired up in such a way as to .. to show graphic, and then do some software trick into bootloader? Or did they change their firmware to not support F1 -> bootloader now?

Holding F1 down brings up a graphic, but the fan/oven stay off at least. 

I'm using jumper wires..
1) jump nISP to ground
2) jump reset to ground
3) release reset (float)
4) (optional) release nISP .. tried it with float, and leaving it on gnd, shouldn't matter .. from NXP datasheet, sounds liek if reset comes to not-ground, and nISP is low, it shoudl go to bootloader (unless watchdog enabled)

Makes me wonder why they don't put Vcc on the header; "float" is unreliable, makes me crazy. Tempted to solder a Vcc wire onto the LPC chip or somewhere, so I can get at it for a high reference.

Does make me wonder if watchdog is set up somewhere (don't know how thats done .. fused in software maybe), blocking bootloader?

I suspect they've done something with the hardware or software in this most recent release.

My unit is marked:
Eco-Worthy
SN: T-962-F16081570
Date of production: 1608

1608 is pretty clearly August 2016, so its very recent.

sivu, 2017-01-10:
F1 does not work with the original firmware, only if you have already flashed the UnifiedEngineering firmware once.

skeezix, 2017-01-10:
Is there a timing consideration when doing the nISP nReset jumpering? ie: so many milliseconds? Does the autobauding have to be within a second or something?

I'm doing it by hand .. jumper to gnd, jumper to gnd, release, over to laptop and hit mouse and hit Flash Magic, etc. I'm wodnering if its just not fast enough?

The LPC datasheets seem very imprecise about bringing up the hardware bootloader.

I'm not getting anything back from the LPV generally, so I think I'm not getting it; but its hardware, so no way to brick that feature out..

Thanks for your thoughts!

skeezix, 2017-01-10:
And all you should need is gnd + rx + tx on ttl serial right?

skeezix, 2017-01-10:
Currently trying to minimize..

before powering unit - jumper to gnd from nISP and nReset
1) power on machine
2) quickly pull nReset
3) quickly pull nISP
4) click the already position 'read device id' in Flash Magic
... tried all the way down to 2400 bps, no joy. (can't autobaud)

In this setup, the t962 display is half clear, half garbage. Thats better?

skeezix, 2017-01-10:
Using lpc21isp on linux ..

If I specify -termonly to bring up terminal, and try hitting ? .. tool doesn't know about 2400, 4800 bps; 9600 it knows, but when I hit ?  I get back garbage with Synchronized mixed in; if I try with 19200 and hit ? I actually do get "Synchronized" coming back from the LPC chip! If I tell the tool -hex <file> .. etc, it does actually say Synchronized: OK, and then errors out 'no answer on synnchronized'.

Bootloader lacks docs, but I assume now that when you connect over derial, you hit ? and it replies Scynchronized and you presumably must echo back Synchronized as a handshake.

I wonder if my T962 crystal is janky, so that the baud is not quite right or something ...

Inching closer. So I know my initial ? is getting over okay, and I can get back the response string, but I can't get the handshake back .. WEIRD.

Wiring should be okay then..

skeezix, 2017-01-10:
Using -termonly..

?
I get back Synchronized
Type in Synchronized .... while typing, I'm getting a lot of garbage.

Clearly something is not lining up on baudrates ... or perhaps line interference.

(And in lpc21isp I'm using linux in a VM .. perhaps VM is causing timing issues, though it shouldn't be with serial) .. trying some other tricks.

Gods, what a pita this is :)

skeezix, 2017-01-10:
Compiled lpc21isp under cygwin, so at least its native to win32 and not in a VM. (My main linux desktop just isn't going to reach with the small cabling the ISP came with :)

Same problem as in linux VM .. I can put in ? and get back "Synchronized", but when typing back "Synchronized", I get garbage back. At least under cygwin I can try 1200bps, 2400bps, 4800bps, etc, and no joy with any of them.

I wonder if theres anything else on the pcb hooked up to those pins, causing interference somehow..

GitLang, 2017-01-10:
On the oven I have an ordinary SPDT switch on the nISP line. I have a small momentary pushbutton on Reset. I have the FTDI serial end wired to a 5 way screw captivated sort-of-din plug. and matching socket on the oven. On Windows, I check the settings for the USB<->serial lead COM port and make sure it is set to 9N1 57k6 baud, no buffers. At any point during program idle or at or before power on, I turn the switch to ISP position, then I poke the reset switch. NOTE -> the screen flickers and sometimes stays blank, sometimes has a frozen program screen, sometimes garbage. Either way, it is now in programming mode and ready to go. Flash Magic should sync up, find the board and be ready to go. Never tried it on Linux, don't know what the dilio is there. Forget about F1 - it doesn't work until after you have flashed one of the more recent versions in.

skeezix, 2017-01-10:
9N1?

xnk, 2017-01-10:
Really odd issues you’re facing unfortunately.. I have been using 57600bps baudrate without any issues. Does your board revision still use a 11.059MHz XTAL? And your ISP cable is a 3.3V one (or does it require a Target reference voltage to tell it which voltage level to use)? If you see Synchronized in some garbled way at least we know you entered ISP mode successfully (the timing is most definitely not critical as long as the pull reset low, make sure isp is held low while reset is released sequence is followed). The fact that you’re getting so much garbage may indicate a voltage reference issue.

I had to bring out the 3.3V level to make my level-shifted UART interface happy, but people have used several different noname USB-Serial breakouts with no issues.

And yes as others have pointed out, the LCD will be all weird and the fan/heater may turn on in ISP mode.

/wj

skeezix, 2017-01-10:
Whats the easiest to solder 3.3V voltage line? I don't really want to try and get in on there on the LPC itself :)

So yeah, I can do ?, and then I get back Synchronized, but when I put in Synchronized myself itsd all garbage all over; so surely feals like a communication issue, and I can get into bootloader okay.

Am at a total loss, but this ISP does take a Vcc reference that I"m leaving floating for now, so thats a concern.

Thanks for bearing with me guys!

GitLang, 2017-01-10:
Very good point, I was going to emphasise that USB to Serial had to be 3V3 and NOT 5V in big letters but I forgot! The one I got was the TTL-232R-3V3 which was 16ukp direct from FTDI. Other important point for people is that it most definitely must *not* be USB -> RS232 as that normally means +/-12V outputs.

FWIW, the plug and socket I used were

Lumberg SV60  6 Pole Din Connector Plug, DIN EN 60529, 5A, 250 V ac IP40, Screw On 
Lumberg KFV60 6 Pole Din Connector Chassis Socket, DIN EN 60529, 5A, 250 V ac IP40 

Nice quality, not your normal DIN crap, and screw thread retained so it won't fall off during programming. Worthwhile having this stuff bought out and mounted on the outside of the case so you don't have to keep having the lid off.

GitLang, 2017-01-10:
Should have been 8N1, sorry. 
8 data bits
No parity
1 stop bit

GitLang, 2017-01-10:
And no handshaking. If you have hardware or software handshaking turned on it will sit there and do nothing. Flash Magic does software handshaking but the oven does neither.

skeezix, 2017-01-10:
Any good point I can solder onto to get a 3.3V voltage ref?

since the oven is responding to ? okay, I think I'm pretty close; but the garbled responses is worrisome .. thats my only guess as of now. Will have to go probing around with the dmm after work :)

xnk, 2017-01-10:
There is 3v3 available right next to the ISP connector. If you look at the picture on the Wiki home page which has the _six_ pin ISP connector on it the 3v3 pin is the one to the right. The resistors right there _should_ be connected to 3v3. No need to be right at the LPC to get 3v3 for this, less cramped for space closer to the ISP connector :)

And yeah you definitely should connect the reference line - otherwise it’s going to have a really hard time getting the LPC to understand you (I assume the ISP cable will default to a really low level)

https://github.com/UnifiedEngineering/T-962-improvements/wiki/images/tempsensor-mod.jpg <https://github.com/UnifiedEngineering/T-962-improvements/wiki/images/tempsensor-mod.jpg>

/wj

xnk, 2017-01-10:
Yep you _definitely_ don’t want to shove +/-12V EIA RS-232-C levels there.. :/

Ah the good ol’ DIN connectors, that’s a nice touch!

It seems like the LPC might be somewhat 5V tolerant on the ISP I/F even when the supply rail is 3v3 but that should not be relied upon. The nISP and nRESET pins are properly pulled up on the board (otherwise the controller wouldn’t be able to start in normal  mode without a programmer attached) so those should really only be pulled down, but I know that most interface cables will actively drive them high as well.

/wj

skeezix, 2017-01-10:
Can I use nReset for voltage ref? ie: the only time I need to communicate is while not-Reset .. so if its pulling up to 3.3V it should do the trick?

xnk, 2017-01-10:
Should work as long as the ref input is fairly high impedance. 

Hmm just checked the schematics and I think you should try the nISP pin instead as they seem to rely on the built-in pullup in the LPC for the reset pin. The nISP pin has a 4k7 pullup to 3v3.

/wj

skeezix, 2017-01-10:
using reset or isp as v-ref isn't working; either no communication or same garbage back.

Have to see if I can get at that resistor to solder a lead on .. theres hot snot glue on all the cabling, so can't get anything apart too easily.

skeezix, 2017-01-10:
okay, so now I get no communication, when using 3.3 as vref on the isp : Cursed.

USB-ICP-LPC2K

I've placed a small breadboard onto the oven surface (on top of the kapton tape I did.) I jumper over from the oven ISP header 5 pins to the breadboard.

The ISP programmer has: Vcc Reset P0.14 Gnd Rx Tx

For now ignoring P0.14 and Reset -- doing them manually.

Currently have resistor on pcb solder jumpered out to BB as well; ISP programmer reset -> 3.3 ref.
ISP programmer Rx to oven Tx
ISP programmer Tx to oven Rx
ISP programmer Gnd to oven Gnd

no response at all from oven on terminal

If I leave the Vcc to float, I do get responses, but they're garbled.

Close. Can taste it.

xnk, 2017-01-10:
So as this is an ISP cable, could it possibly be that they have actually meant TX to go to TX and RX to RX?

And when you measure ref, is it close to 3v3?

Odd, odd indeed!!

/wj

skeezix, 2017-01-10:
ref is 3.3 exactly.

Given I get somewhat correct responses to ? at 19200bps, I know the Tx/Rx is correct. (I also measured Tx from programmer on o-scope, and decoded characters correctly.)

Very weird that I get no response at all when using the Vcc volt ref.

(It could well be ISP cable is bad, or ISP is bad, but I got it brand new from digikey a few days ago.. unlikely. I doubt I cooked it, either.)

Mind you, I first hooked it up cable backwards since its unmarked (yay!), but even then .. theres no Vcc on oven, so worse case was hooking gnd up to P0.14 I think, say. Should be okay.

Given the ISP is seemingly operating correctly on the scope though, and that I can get semi garbled responses in the right circumstances, it feels 'more or less working'.

But given its _not_ working, it could be anything still :/

xnk, 2017-01-10:
Right, that makes sense w/ regards to the tx/rx. If you unplug the cable from the oven board and manually connect programmer tx and rx together, does it properly echo characters? For that test to be completely conclusive you should scope that looped signal both when you apply a 3v3 reference and without it. Sounds really odd that nothing’s happening when the ref is hooked up!

/wj

skeezix, 2017-01-10:
.. and we're in; retouched solder joint on the vref to make sure its consistent and not screwy.

FLASHED.

THANK YOU EVERYONE!

I hope this experience has helped future folks as well..

-- if the ISP has a voltage ref, try to hook it up
-- theres no easy voltage ref to use .. using nISP and nRESET won't cut it, use the resistor

Process:
- you can keep nISP grounded the whole time (it has to be low on reset)
- ground nRESET (use a jumper, easy)
- turn oven on, yank your nRESET jumper out, run the flash

Timing isn't super accurate, but if you let it flap in bootloader for a few seconds (6+?) it'll run the code, not stay in bootloader forever (??)

The built in firmware will not use F1 to go to bootloader; you have to do it manually.

The bootloader is active when yo uturn power on and have blank or garbage screen.

xnk, 2017-01-10:
Success! :) That was a very very odd failure mode indeed, I’m not sure I would have been THAT persistent when hooking up the oven the first time - I would have grown tired of it long before I got some useful code running.

Did you manage to extract the original firmware from the LPC before you re flashed it, or you didn’t really feel the need to be able to revert? :)

/wj

skeezix, 2017-01-10:
Didn't bother; only forward, never backwards! :) It sounded like the original FW wasn't useful, so I bought the oven with the goal of going to this new FW and the basic mods :)

I should have pulled the FW, just to see if it matched up with anyone elses at the bin level.

At least in the future I can participate in the process with others .. it is damned tempting to mod the case and run a header out to the front panel, so that next reflash is trivial without even removing screws :)

GitLang, 2017-01-10:
It's always the silly little things that steal the most time. Well done!.

Next task is to try to build the source into something that will load and run, then world is your bicuspid mollusc.

skeezix, 2017-01-10:
I'm using the pre-built firmware now; that one good enough for a little while?

I'm planning to do the cold junction temp sensor mod (from the youtube video):
https://www.youtube.com/watch?v=0ITZXY5Mipc&index=3&list=PLm894rlrz1A5Ys9svvNuI_a31BHd8hIXD

I'm assuming the prebuilt FW will either work without the mod (crummily :), or work okay with the mod.

Then I'll see if the thing can actually bake any of my old PCBs (I'm a hobbyist, but have a good stack of them around.)

.... or do I need to build the FW from source to specify the features I need? (ie: cold junction yes, fan control no or yes, etc)?

xnk, 2017-01-10:
No worries, it seems like there isn’t much attention given to the original firmware so I assume most of the ovens comes with exactly the same firmware image. There should be plenty of copies floating around at this time if you need to experience the slow timer and even slower keypad input again ;)

One issue resolved at least, I see that I really need to go sort through all the issues here.

/wj

xnk, 2017-01-10:
The pre-built firmware will be able to deal with all of that. It assumes that the system fan mod is there, but nothing will happen if the mod isn’t there. I know that there are a lot of stuff waiting to go into the tree as well but the pre-built is a good start. That being said it’s always nice to be able to at least reproduce the official build locally.

/wj

skeezix, 2017-01-10:
Maybe add a note to the wiki ..
-) the F1 to enter bootloader, is not supported out of the box FW

And ... Verify you are in bootloader
- the datasheet for the LPC chip isn't very complete in this regard
- the protocol is essentially "?" to help the chip autobaud; once you do the ? it'll response Synchronized
- you then type in Synchronized and it'll return Ok
... So if you can hit ? in a terminal, and get some cvlear text responses, you're okay

And note on wiring..
- you need only serial TTL for this to work; in general this is likely a USB->Serial adapter, or an ISP; could use an arduino or roll your own .. no big heavy ISP needed; I imagine Bus Pirate any many other tools could do it.
- RX TX Gnd, maybe Vcc-ref needed

The actual nISP and nReset can be purely done manually, or rely on an ISP to do it. You can leave them out of wiring if you do it maually.

That ought help with some confusion in the future.

Really, the confusion is in the flashing side (not UE FW side at all), but theres no where else to point people ... any googling for LPC flashing will get you here :)

Thanks!

GitLang, 2017-01-10:
Yes, any usb -> serial adapter will do it, but it has to be 3.3V.  5V TTL outputs can kill the cpu if O/C

xnk, 2017-01-10:
Feel free to add the info in there, always useful with as much info as possible!! I know the F1-bootloader functionality is mentioned as not being present in fw older than 0.2.1 on the flashing wiki page, but maybe not obvious that it’s not there at all in the stock fw..

/wj

skeezix, 2017-01-10:
I've done a number of updates to the wiki; thanks!

closed 2017-01-10 by xnk

Vitalic66, 2020-06-24:
Sorry, need to ask here again,

I tried 3 different usb ftdi devices always with the same result. No communication.

Tried it with win7 win10 and ubuntu 18.04 .

Connected Tx Rx and ground but i only get the garbage output if i remove the ground.

Tx shows 0.9V Rx 3.22V .

I already read about the 3.3 level trick from the resistor next to the isp connector, but which signal do i have to connect? 3.3v to the Tx or/and the Rx pin of the isp connector?

Bootloader mode should be active as the screen stays blank and the heater/fan turn on.

Hope someone still reads in here...