UnifiedEngineering/T-962-improvements #105: Gain and Offsets

baballuci, 2017-01-06:
Hi, i just upgraded my T962 to use 4 MAX30850's, do the gain and offsets in software not work?

I'm currently only using 2 thermocouples ( L & R ). The right had thermocouple seems to be 30 degrees higher than the left. No matter what adjustments i make in software it doesn't make any difference. I've replaced both thermocouples and both lamps for these ( http://uk.rs-online.com/web/p/heat-lamps/7393536/ ).

How can i calibrate the thermocouples?

I'm running firmware 0.5.1 by the way.

Cheers
Charlie

xnk, 2017-01-10:
You’re right, the gain/offset settings are only for the built-in ADC. Those really shouldn’t be visible when the real tc interfaces are used. Normally the TCs shouldn’t need any calibration but there have been reports on one side being significantly hotter than the other in the oven. Some of it is depending on how much gaps are in the oven letting cold air in, some of it is fan speed dependent.

Have you measured the temperature right next to the oven TCs to verify if the temperature is actually correct and the readout wrong, or if the air temperature is in fact hotter in one end? The temperature regulation is based on the average reading of the two TCs.

/wj

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk