UnifiedEngineering/T-962-improvements #106: T-962C : fw 0.5.1 - heat profile issue

vmarguerie, 2017-01-07:
Hi,

I've made the modifications  (masking tape, grounding, cold junction, fan control) to my **T-962C** reflow oven.

Even if i bake PCBs with pretty good result, i wonder why the oven does not succeed to follow the heat profile.
[image]
When i did the hardware mods, i noticed some components were missing on the triac board "t962-2 new2". Is it normal?
[image]
Thanks for your help.

xnk, 2017-01-10:
The missing components on the board is normal for the T-962A so I assume that is the case for T-962C as well. It sure looks really really strange that curve, a very weak heater. Given the button arrangement I’m not sure they actually run the same firmware in it? But as it is the very same board with components looking very similar to the T-962 and T-962A it seems like there’s only one SSR controlling the heaters. Can you see all the heaters glowing? Does the main fan speed regulation work during reflow (limiting the amount of airflow to a very slow, but still moving, rate).

Is this a 120V or 230V setup by the way?

/wj

vmarguerie, 2017-01-11:
Thanks for your feedback.
Yes it is strange ! I see the 12 heaters glowing and the main fan speed works during reflow  (setup to 5).
It's a 230V setup.

Regards. 
Vincent

xnk, 2017-01-12:
Hmm, when the ramp is happening (the one that looks like a straight diagonal line), is the Lamp LED on 100% of the time or is it pulsing at all? It seems so weird that it would heat that slowly if it is indeed running 100%. If you turn min fan speed down all the way down to 0, does it improve the temperature rise rate at all?

/wj

vmarguerie, 2017-01-13:
With the fan off, it doesn't change anything. ..
During the ramp, the lamp Led seems to be on 100% of the time. It doesn't flick but before it becomes visible (red),  there is at least 45s from the 250°C set temp. I did this test with the speed profile. 

Regards 
Vincent

christaylor99, 2017-01-13:
Is it possible that the 230v heaters are in series rather than parallel?  That would put a voltage drop on the heaters limiting how warm they get.  Check for a wiring error or voltage check on the board connector or perhaps a floating ground on the heaters that puts them in series.

vmarguerie, 2017-01-30:
I did the test in manual bake mode, temp set to 300°C. With the heaters disconnected, i have 230Vac on the board connector and only 27Vac when the heaters are connected. I Will have to check inside the oven...
Thanks for the Tip !

vmarguerie, 2017-01-31:
After a tedious disassembly session, I discover that it has 3 groups of heaters mounted in parallel. 
Each group is composed of 2 heaters mounted in series. Does this explain the problem of temperature rise?
I attach a diagram to this comment.
[Oven-T-962C - Heaters wire.pdf](https://github.com/UnifiedEngineering/T-962-improvements/files/743131/Oven-T-962C.-.Heaters.wire.pdf)

Thanks again for your help.

xnk, 2017-01-31:
Looking at the spec it would make sense to have each heater part split in two if they want to support both 115 and 230VAC with relative ease (parallel config for 115, series for 230). They claim 2500W max power, can you measure how much power (or current) your oven draws when "lamp" is constant on? It seems that there's not nearly enough heat output, but I honestly can't see how the original firmware can get more output than our 100%, when there's only one group of heaters all hooked up to the same solid-state relay gated by the same single pin as on the smaller ovens. This is some seriously odd behavior from the oven/firmware.. /wj

scott-t-wilson, 2017-05-23:
vmarguerie,

Did you get the T962-C to work?

JohnWestlake, 2017-09-30:
I can confirm the same results as vmarguerie - the basic issue seem that the oven does not have the heating power - even at 100%.

I run the Ramp Speed test and the oven only got to around 185Deg C at the 4 minute mark (Start of the downward ramp).

I wonder if these are the same heating elements as used in the smaller oven - if so how are the heaters wired in the smaller oven?

I've little experience of IR reflow, but has anyone has success with the T962C (large oven) with leaded solder?

Here's my custom profile (the fasted ramp times I could get the oven to do) only half the board (95mm x 95mm) reflowed successfully with the original software - indicating "hot spots" hence me trying this updated software in the hope that slowly running the fans during reflow would even out the hot spots.

[image]

With the original software (in the above profile) You can see when the cooling fans switch on there's a rapid jump in the Peak temp (the odd double "peaking") - however judging by the lack of complete reflow this is not the true board temp.

mikeanton, 2017-09-30:
Is the graph you posted the actual profile?  Did you try to profile the board by measuring the real board temperature, and then tweaking the profile to get the proper shape as seen by a thermocouple on the PCB?  That is the usual process required to develop a proper reflow curve.

JohnWestlake, 2017-09-30:
Mike,

1.These measurements where made with the thermocouple mounted against the PCB (although maybe 1-2mm above the board so not physically in contact with it.

2. Custom profile run with the ORIGINAL software.

3. Larger oven T962C

The point of the post was to demonstrate that even at 100% duty cycle, the ovens thermal heat output is very weak hence the extended cycle times (Slow ramp speeds).

As I mentioned - performing the Ramp test cycle on the upgraded software and the T962C only managed to reached about 185DegC after 4 minutes before the test started the downward ramp cycle.

I wonder if the incorrect "timebase" of the original software was to compensate for the slowness of the ovens heating cycle due to its lack of thermal output power.

I'll try playing with my custom profile with the new software - basically extending the cycle times to allow the oven to ramp up to the required speed (now that the timebase error has been corrected).

I worry about the reflow time (183DegC to 225DegC (max)) as they say for leaded solder 90secs max (the solder in "molten" state) as the oven just does not have the power to do this!

I'm no expert on reflow ovens / process so am learning as I go, please don't take everything I say here as fact...

mikeanton, 2017-09-30:
John,

The thermocouple should be pressed onto a copper pad when profiling the real board temperature.  Since yours was not touching the board, that would explain the rapid changes in temperature when the fan turned on.  As you say, it is not real, or at least not the temperature that we care about.  I suspect when the fan turns on, it starts to even out the hot spots, and starts to remove some of the heat from the metalwork, which would cause the air to warm up suddenly.

If the larger oven is on the edge of barely having enough power to meet the profile with the fan off, it probably has no hope with the fan on, as you have found.  I know the smaller oven has a bunch of holes in the bottom, so when the fan turns on, it will cool down pretty quickly, but the oven seems to have enough power to make up for this, if the fan is running slowly.  The result of running the fan does seem to help even out the temperature gradient of the oven.  If the large oven has holes as well, then probably there is just too much cool air being stirred in when the fan is on, that can't be overcome by the limited heating power.  Perhaps adding a proper convection fan, that doesn't pull in outside air would work better.  However, you still need a way to pull in outside air to get it to cool down at the end.

Yes, if the oven doesn't have enough power to ramp quick enough, you probably will end up staying in the liquidus zone for too long.  The same thing happens if you can't get the oven to cool down fast enough after reflow.  I run a commercial oven that runs on 240V, so it can certainly ramp up fast enough, but there is no way to get it to cool down quick enough to match the profile exactly.  I've run many thousands of boards through it though, and there haven't really been any problems with any of them.

I think the inaccurate timebase was probably just an oversight, as the smaller ovens have the same problem.  Though they seem to be able to follow a reflow profile properly, so there wouldn't be any reason to have the same problem there as well.  My opinion is, that the original firmware is just plain terrible, so it isn't surprising that they didn't get the timing right...

JohnWestlake, 2017-10-02:
Just as an update - I've coming to the simple conclusion that the T962C oven is just poorly designed and lacks the thermal power to be controlled correctly - the oven needs a cycle time longer then 8 minutes otherwise its cycle is programmed so that the heaters are at 100% most of the time - thus the controller is basically operating "open loop".

Don't buy this oven (the T962C larger version) - its unusable!

As my first few production batches of boards are physically smaller and will fit in the smaller T962 oven - do these work OK once the modifications have been performed (hardware and software)? I'm considering purchasing a smaller oven to tie me over until funds allow me to buy a more professional re-flow "oven".

With the smaller oven, I worry about local hotspots (what is the recommended fan setting) as some PCB's might take up the Max. usable area in the smaller oven...

KLEYNOD, 2017-10-04:
I have this oven and it has 2900W not 2500W. I bought it one week ago , it works good too. no firmware upgrading yet .

KLEYNOD, 2017-10-04:
buy the way check your probe , make it sure it  not twisted , it should be like "V" . I saw people twist them , this is make it slow sensing.

wulfmans, 2017-10-04:
One more thing you should do. Replace the supplied power cord with a 12
awg or better cord.
the one they shipped with mine was 14 awg

mikeanton, 2017-10-04:
Ok, this is odd.  The listing here: https://www.aliexpress.com/store/product/New-Arrival-PUHUI-T-962C-Infrared-IC-Heater-T962C-Reflow-Oven-BGA-SMD-SMT-Rework-Sation/205122_32355471263.html, says that this oven is 2500W.  But, at 110V it is rated at 10A, which makes it 1100W, and at 240V, it is rated at 8A, which makes it 1920W.  I really wonder if it is 2500W at all, or if that is just an exaggeration, as something doesn't add up.

So I guess one questions is what voltage are you running the oven at?  As that would make a big difference in the power that is available.

JohnWestlake, 2017-10-04:
Hi Guys,

Thanks for your replys - Are you UK based with "240Vac" rather then the 230Vac we have in Czech Rep? I measure like 220Vac when the supply is loaded and the lights flicker when the oven controller does its thing - its like a disco in the lab when the ovens in use.

I'll try to measure the power consumption - trouble is my "power analyer" has a UK plug and considering the power requirements of the oven I'm not sure its a good idea try a "travel adaptor" to convert to the Euro style socket here in Czech...

I ordered a smaller T962 (800W) oven yesterday - I'll see how it goes.

JohnWestlake, 2017-10-04:
KLEYNOD,

I'd hold back on the software update - its too cleaver and trys to control the oven correctly, but still in my opinion the oven lacks the thermal output power to be able to be controlled correctly (with the original software its at near 100% across most of the cycle.

Also, the time base has been corrected on the update software - this makes the true cycle time to sure for the thermal power of the oven - I'm guessing that the only reason the original software "works" is the 1/3 longer cycle time to what is indicated by the display (60 seconds is closer to 90 Seconds).

With the corrected timebase of the updated software - the oven cannot follow the thermal profile in the 8 minute period available.

With the new software I had to pre-heat the oven to say 40deg and then run the profile - but its all a botch and the starting temp (and if the "mass" of the oven is already hot) makes a MASSIVE difference to the profile. 

When I spent a couple of days playing with the profile (with the new software) I had to basically set my profile settings to "Bang Bang" type operation and leave the natural time constance of the ovens thermal mass to create the heat profiles - the oven is really a POS.

mikeanton, 2017-10-04:
I also misspoke, as the rating is at 220V, 8A, so 1760W.  In Canada (and the USA), we have nominal 120V, and 240V single phase.

mikeanton, 2017-10-04:
The more commercial oven I use, starts the oven at 80C.  That really does help get everything up to temp faster, so I suspect that is a good way to go, and there is no reason that you couldn't do the same.  The metal in the oven will be hot when you put boards in though, so you will have to be a bit more careful...

JohnWestlake, 2017-10-04:
Hi Mike,

Yes - that's what I finally worked out for myself after a couple of days messing - unfortunately by the time I was clever enough to consider this idea, I'd already wasted a couple of days playing with profiles and didn't have the mental energy to reprogram my "Bang Bang" type profiles - I figured its was cheaper just to buy a new oven then the lost income of 2 days of getting nowhere - even when its pushes my credit card further into the red!

I ordered a T962 for 180 Euro's from Ebay, apparently stocked "locally" here in Czech... lets see!

mikeanton, 2017-10-04:
You will get better at profiling the oven over time, which is pretty important, as you will need to do it fairly often.  It isn't a good idea to always try to use a generic profile for every board.  It is fairly common to need to profile the oven for a specific board design, as things like the color of the board, the number of layers, the component mix, and how big the copper pours are, all affect how the board will reflow.  So, if you want to make sure that you are getting the proper profile for the board, you pretty much need to profile the oven for all new designs.

I find that this process usually takes me about 1.5 hours, which I can do while putting parts on the board or other setup.  Having a multimeter with a temperature input that you can log data from really helps in this process.  That way you can clip a thermocouple to the board, and run a cycle while doing other things.  Then look at the data, and tweak the profile, run again.  Repeat until you get the profile you want to see.

rlimberger, 2017-10-31:
There seem to be different experiences with the T-962C posted here. Is there a consensus as to whether it DOES work with the custom FW or not? I am interested in buying the "C" model due to it's larger size but now I am confused as to it's suitability for the custom FW. Thanks in advance.

JohnWestlake, 2017-10-31:
I now have the smaller oven so can directly compare the two ovens, and I can say 100% the larger C oven simply does not work - it does not have the thermal heating power to follow the "Correct" reflow profile - or anywhere near...

If your based in Europe, your welcome to have my unit for cheap (I'm based in the Czech Rep)... its useless to me - but I really recommend you don't... I guess I can use it as a Pizza oven...

With the smaller oven the heating elements get bright orange during Max reflow heating phase - with the larger C they are a brown / very very dull Red.... 

As far as I can tell they are wired up correctly.

rlimberger, 2017-10-31:
Thanks John, I actually have the smaller one and the heating "ability" is causing some problems with plastic connectors turning brown and melting sometimes :)

So if one would program a curve that the larger oven is capable of following (i.e. longer cycle time), the  new FW would work?

JohnWestlake, 2017-10-31:
I found that the smaller oven can be made to track the profile accurately - BUT there is a 10Deg to 20 DegC temperature difference across the oven with the right-hand side (below the display) atleast +10Deg to 20 DegC higher then the left side.

The mini oven profile can be adjusted to work well (with leaded solder atleast).

With the larger "C" oven, the software's maxim reflow time of 8 minutes does not allow the oven to heatup accurately - only by operating at full power does the oven reach the reflow point but you cannot accurately control the oven when only 100% power has any chance of reflow - this is with leaded solder , I'm not sure you could reflow unleaded solder with this oven!

KLEYNOD, 2017-11-01:
I found also that  different change in temperature  between left and
right side depends on thermocouple wire length , they should have the
same length , if you change the cable , it helps alot , position of one
sensor to controller is a little bit longer , change to new one but let
it be the length same for left and right . DON'T CUT THE WIRE FOR
THERMOCOUPLE.  

another things that I found is the Opamp on the board , they have put
the version with higher offset (TLC27L2C) , changing opamp to TLC27L7C
is better version , or opa2333(http://www.ti.com/product/OPA2333) even
better. This offset in opamp make it problem when you want accurate DC
amplification  of thermocouple  I try to put this information on web
side , but it is better it come from you to after verification on your
system , 

---
Med vänlig hälsning / Best Regards

Vahid Mouknatjou
KLEYNOD AB
Karlavagnsgatan 20
SE-41753 GÖTEBORG
+46 (0)760582002 

2017-11-01 00:42 skrev JohnWestlake:

> I found that the smaller oven can be made to track the profile accurately - BUT there is a 10Deg to 20 DegC temperature difference across the oven with the right-hand side (below the display) [...]
 

Links:
------
[1]
https://github.com/UnifiedEngineering/T-962-improvements/issues/106#issuecomment-340938953
[2]
https://github.com/

KLEYNOD, 2017-11-01:
I have this C version , and it works somehow with original software, it
do the job, I also change the software to "UnifiedEngineerin" version ,
then it shows that time of profile and power to lamps were not enough ,
with original FW it runs in wrong clock and this error actually help a
little bit , you can also run preheat at 50C which could help   , the
thermal mass of cabin is much higher than the little version , I bought
"C" version because my PCB are large . I also found why the "Left "and
"Right" has so much different in temperature ,  wire length of
thermocouple is different , if you change them to new one and let the
length be the same , don't cut or make adjustment of thermocouple sensor
cable, make sure that the tip is in "V" , don't twist it , you get 10C
less with twisted.  

this upgrade can be done on all version of T-962 oven, another problem
in temperature difference was also on board opamp , the use TLC27L2C
version that has 12mV offset , change this opamp to TLC27L7C which has
500uV offset . this offset should be as low as possible when you want
measure DC voltage of thermocouple which is around 40uV/ C ,  if you
want better change it to http://www.ti.com/product/OPA2333 , with zero
drift .  

I did all these change but still was not enough power to lamp for fast
ramp, it need more power, it needs to go from 25C to 150C during 20
second, so I decided to change wiring of lamp to 3 phase system,  you
need almost 200VAC on each lamp, here in Sweden we 240vAC single phase ,
original cable to lamps are connected 2 in serial, 2x3 set . other
country with 120vA parrale connection should be  , 

So I change the cabling Δ (delta) to 420vAc , that five 210vac on each
lamp . 

If you want prsicion oven then make as follow : 

I change also electronic to PID Omron controller with
https://www.ia.omron.com/data_pdf/cat/g32a-ea_ds_e_7_1_csm182.pdf?id=991

MODELL BETECKNING 

ANTAL 

BESKRIVNING 

G32A-EA-US 

2 

Cycle Control Units 

G32A-D40-US 

1 

Short-circuit Units 

G3PA-430B-VD-2 

3 

SSR ,Zero cross function 

E5AC-TCC4A5M-022 

1 

Temperaturer Kontroller 

E58-CIFQ2 

1 

USB-Serial Conversion Cable 

E58-CIFQ2-E 

1 

Communications Conversion Cable 

EST2-2C-MV4  

1 

CX-Thermo Support Software,  version 4.61 or higher 

You will get the best of this "C" version .

---
Med vänlig hälsning / Best Regards

Vahid Mouknatjou
KLEYNOD AB
Karlavagnsgatan 20
SE-41753 GÖTEBORG
+46 (0)760582002 

2017-10-31 22:40 skrev JohnWestlake:

> I now have the smaller oven so can directly compare the two ovens, and I can say 100% the larger C oven simply does not work - it does not have the thermal heating power to follow the "Correct" [...]
 

Links:
------
[1]
https://github.com/UnifiedEngineering/T-962-improvements/issues/106#issuecomment-340915754
[2]
https://github.com/

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk

fah, 2021-01-29:
I ordered a 962C in January 2021. 
I use it in Germany with new 2.5 mm wiring in the house. It draws exactly **3kW@243V**.
At 5°C room temperature (without preheating) it heated to 226° in 3:28 with a constant 3kW.
Cooling was very accurate along the curve. Annoying beeping started at about 70°C.
I used the standard firmware for this test. No mods so far.

Marora6939, 2021-02-19:
Hi 
I am working on T-962C IR oven machine. I did ground, cold compensation, kapton taping, updated firmware but still some of my IR lamps are not glowing.
I did check the temperature at different points in oven with the help of thermocouple device. So, the issue is the first three lamps are glowing and they are doing their work but the other three are not working.
Can anyone help me please?
Thank you

xnk, 2021-02-19:
That is odd indeed, as I’ve only ever been able to spot one IO pin used to control the heating elements. Do you have any pictures of how the heaters are controlled (SSR:s and control wires for them?). Are there two solid-state relays in your unit controlled by two different IO pins? Did all heating elements glow with the original firmware? The T-962C seems to be an odd machine indeed. As fah mentioned above that unit obviously consumes the rated power (with the original firmware), but there has been several reports of some units heating fast enough. It has been brought to my attention that the T-962C has a lower power per area than the T-962A which in turn has a lower power per area than the T-962, so theoretically the T-962C should be the slowest of the bunch (unfortunately).

hodginsa, 2021-11-20:
> I ordered a 962C in January 2021. I use it in Germany with new 2.5 mm wiring in the house. It draws exactly **3kW@243V**. At 5°C room temperature (without preheating) it heated to 226° in 3:28 with [...]

You need to be aware that in the stock firmware 1 second is actually closer to 2 seconds. So the reflow profile is not 7 minutes its 14minutes. Your 226C in 3:28 is more like 7 minutes. On mine it can barely get to 230C in 14 minutes on 220V in North America. Worse in my cold basement.

mikeanton, 2021-11-21:
North America is largely 240V at this point.  I don't know why it is still often referred to as 110/220V, as the real voltages in Canada and the USA are 120/240V.

GitLang, 2021-11-21:
I would think that by now they have fixed that time problem, it's bit of an obvious software bug.

Voltage in the UK was always 240V. When we joined the EU several decades ago we were supposed to harmonise to the EU 230V standard. The lower tolerance on our 240V was within the upper tolerance level on the EU 230V, so the only thing we changed was to call it 230V!. Now we have left the EU I don't know if they will "change" it back to 240V. If you put a meter on it, it usually measures around 237V here, and the oven heats up plenty fast enough.

What is the nominal rating of the higher voltage oven?. if it is 240V, then using 220 volts will drop the power from 3000W to about 2500W. A 16% decrease, but it's not *that* huge.

kbhuinfo, 2021-12-02:
Any ideas on getting rid of the beeping? Or at least decreasing volume? Thanks!

John-Westlake, 2021-12-02:
I just put several layers of Kapton tape over the beepers "output" hole...

pfuriasse, 2022-02-07:
Hello, I wanted to ask you if you can help me. The Puhui T962C oven that we have does not stop at the end of the program and continues heating. Do you know how to solve this problem?

GitLang, 2022-02-08:
Which LEDs flash? The Fan, the Heat, or both. Does it stay on the profile page or go back to the main menu?

pfuriasse, 2022-02-08:
> Which LEDs flash? The Fan, the Heat, or both. Does it stay on the profile page or go back to the main menu?

W
[image]
hen the beep sounds, it cuts off the cooler and the resistors continue to heat up and raise the temperature to 150 degrees again.

xnk, 2022-02-08:
That oven clearly runs the original firmware, not really sure what’s going on there. That one shouldn’t keep heating either of course, and it shows how bad that oven is keeping up with the reflow profile. Keep in mind that that will be some _really_ slow minutes if it’s an unpatched firmware… The comments about the open-source firmware being slow to heat the oven makes perfect sense now, the C-version has to be seriously underpowered.

wolframore, 2023-03-11:
I ordered a 962C, it just arrived the other day.  I liked the size.  It now says its 2900 watts.  I had a little damage to mine.  Packaging looks fine, but the drawer does sit flush on one side.  Looks like 3 spot welds behind the drawer front broke loose.  I applied for damaged refund and they are willing to knock it down $192.  It might be better than waiting for a replacement drawer.  I can probably spot weld them back.  I haven't tried it yet, might have to pull a new circuit for it.  26A is a little dicey with my current circuit. 
[image]

savio-code, 2023-07-01:
I figured out the problem, The IR heaters perform very poorly when pulsed using PWM (the firmware does this especially when starting the reflow process),  you either need to run the IR heaters at 100% duty cycle (pin = 255) or 0% or (pin = 0).

Changed a section of the code to this:

if(Sensor_GetTemp(TC_AVERAGE) < intsetpoint) {
			*pheat = 255;
		}
		else {
			*pheat = 0;
}

My T-962C now follows the curve perfectly. Also i added code to preheat the oven to 60C before beginning the reflow process. This way guaranteed the oven was a warm enough before starting

richardblackman, 2024-01-14:
> Any ideas on getting rid of the beeping? Or at least decreasing volume? Thanks!

Some info here, just desolder one transistor.

https://www.youtube.com/watch?v=EgQAnwqE44g&t=348s