UnifiedEngineering/T-962-improvements #107: Thermocouple scale factors using the ADC

Guitared1953, 2017-01-12:
Looking for some help/direction. I have the smaller T-962 oven.  I added the cold junction compensation (DS18B20) as well as the other recommended modifications and upgraded the firmware to v0.5.1. To check the accuracy of the thermocouples, I attached two externally monitored thermocouples to within 1/16" of the two suspended oven theromcouples. The external thermocouples were monitored using a HP 34970A and Benchlink software. Using the "Bake" mode I found the following:

- Oven displayed thermocouple readings disagreed considerably with the external thermocouple readings.  I was able to improve the disparity with the gain and offset adjustments available at the oven front panel.

- The left and right thermocouples, internal and external, measured considerably different temperatures. To test the idea that maybe the fan was blowing colder air on one thermocouple more than the other I blocked the fan opening with a piece of paper and the left and right measurements came more into alignment with each other.

- I manually increased and decreased the over temperature control to observe the response. I noticed the temperature displayed on the oven became unstable particularly during cool down when the cooling fan was running (unblocked this time). I assume there is noise interfering with the ADC.

I would like to replace the TC amplifiers with some Analog Devices, AD8495,  with built in cold junction compensation. Also, I would like to disable the fan during heating periods only. I'm guessing the fan introduces more problems than it solves.

As a very, very casual C programmer I cannot find the place(s) in the source code to modify the thermocouple scale factor from whatever it is to 5mV/C°. Also, I'd like to turn the fan off at will.

After many hours of trying to accomplish this goal, it has become personal. I'm hoping someone can at least point me in the right direction.

Thanks,
Phil

xnk, 2017-01-12:
The simplest thing first: You can enter the setup menu and turn off the idle fan (so that the fan will only run during active cooling) by adjusting Min Fan Speed to zero.

The AD8495’s hasn’t been used with these boards yet as far as I’m aware but there are two Maxim one-wire TC interfaces that have been used successfully. That will get rid of the built-in ADC all together (https://github.com/UnifiedEngineering/T-962-improvements/wiki/Better-thermocouple-interface <https://github.com/UnifiedEngineering/T-962-improvements/wiki/Better-thermocouple-interface> and https://github.com/UnifiedEngineering/T-962-improvements/wiki/Better-thermocouple-interface-2 <https://github.com/UnifiedEngineering/T-962-improvements/wiki/Better-thermocouple-interface-2>). The built-in ADC does not have a very accurate voltage reference (it’s the 3.3V digital supply voltage if I remember correctly) so I would avoid any TC amplifier with an analog output that has to utilise the LPC ADC.

Some ovens are apparently more uneven than others (I developed this on a small T-962 as well and I don’t have nearly as much left/right difference), but I agree that the first order of business would be to get the built-in ADC out of the picture so that your TC’s inside the oven reads the same value as you externally monitored ones all the time. Some TC’s seems like they’ve had a rough time with kinked wiring that will cause more junctions. If you experience the annoying non-linear response where the readout is good in the lower temperature range but gets complexly off at higher temperatures there may an issue with the TC wiring close to the heat break between the oven and where the electronics sits.

/wj

Guitared1953, 2017-01-14:
I already have the hardware, including the AD8495 PCB and a 2.5V reference for the ADC.  As a hardware engineer I was hoping to use this upgrade as an opportunity to expand my C programming skills.

GitLang, 2017-01-14:
If you want to learn, connect your board to a cheap Arduino. If you want your oven to work, get that quad MAX31855 digital thermocouple board. You don't have to think about gain, offset, or cold juction comp, it's all done for you. The PCB also has a temperature control loop for the small cooling fan, and the bare PCB is available really cheap. The author (him, up there a bit :) put the PCB up on Dirty PCB's.  They make PCBs from peoples artwork, and if the person wishes, they can allow others to order batches of their boards, which is what the designer of this board did. I'm sure you will find all the details somewhere on here. 

I found the email from the PCB makers

18672 _ThermoCouple_board_v2.5.zip 2 
10x10cm (4.9x6.1cm) 
Blue 
1.6mm 
HASL 
None 
0.42kg 
Normal Protopack 10 pieces  $25      
Shipping DHL Hong Kong (3-9 days)  $15
   Total   $40   

The other possibility, if you don't want to buy 10 PCB's, is that some people here are bound to have 9 spares :)

closed 2017-01-19 by Guitared1953