UnifiedEngineering/T-962-improvements #108: MAX31850K better thermocouple interface calibration

hamityanik, 2017-02-06:
Hello,

I did MAX31850K better thermocouple mod, I've calibration issue.

I've installed two MAX31850K and using only two thermocouples on the oven. I'm measuring heat on the PCB with external thermometer, it's actually 280C peak but oven thinks temperature is 245C.

How can I calibrate?
Thank you.

GitLang, 2017-02-06:
There is no calibration, it's in the chip. You either have bad thermocouples in the oven, or you are heating too quickly. Heat flow is very unenven in the oven and it needs time to settle. Try to keep below 2degC per second and make sure you have a decent pre flow soak just below the liquidus. One person here added the fan from a convection oven and it helped I believe. I recall the fan had metal blades in the top of the oven and the motor in the compartment above, the shaft passing through the heat barrier (this is on the T962A, the bigger oven). Yoiu should check your thermocouples at as steady a teemperature as you can achieve.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk