UnifiedEngineering/T-962-improvements #110: MAX31850K thermocouple mod

SouthernAtHeart, 2017-03-25:
Hi, I just got this oven and am fixing to add the different mods.  I will only be doing 1 board at a time, about 3" square, and will set it right in the middle of the tray.  Do you think the thermocouple mod would benefit me much?
Thanks.

GitLang, 2017-03-25:
If you mean the small addon board then 100% definite do it. The cold junction compensation mod does really help, but the small board does away with all calibration and drift problems. The four part article starting here

https://theembeddedworkshop.wordpress.com/my-projects-2/arm-projects/improving-the-t-962-reflow-oven/

is a must read.

jake-b, 2017-05-30:
I just completed the MAX31855K modification based on the design in the wiki and I am happy with the results.

Would appreciate anyone's input on: (a) how to best mount the thermocouple board and (b) how to route the EXTRA1 and EXTRA2 thermocouples into the oven on a semi-perminant basis.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk