UnifiedEngineering/T-962-improvements #111: Calibration

SouthernAtHeart, 2017-04-04:
[image]

Is there a way to calibrate the PID control of it? In this photo you can see it's not quite where it should be when it's warming up. 
It's not bad and is very usable. I just wondered if it could be improved.

frippe75, 2017-04-08:
[image]

Flashed latest 0.5.1 firmware and installed the coldjunction sensor mod using ds18B20.
Not really sure hot to use setup to try to adjust/calibrate from here. It got too hot using the original firmware now it's the opposite.

xnk, 2018-02-05:
Lag is to be expected due to the thermal mass. The oven needs to be warmed up by running a dummy cycle or two before reflow. Not reaching peak temperature may be a result of running the main fan slightly too fast.

closed 2018-02-05 by xnk

SouthernAtHeart, 2018-02-05:
Thanks, I’ll try lowering the fan speed. And running a dummy cycle