UnifiedEngineering/T-962-improvements #113: BROWN VERTICAL STRIPES across the PCB

Houseman74, 2017-05-02:
Hi Folks.
Have made the cold junction compensation plus the upgrade firmware according to the forum and tried to reflow my first PCB.
Apart from some misalignments at the end and wrong quantity of solder-paste, on the PCB it can be seen terrible brown stripes like burnt.
The thermocouples were placed one below the other above the PCB.
Can you please suggest me some hints?
Best regards.
Steve
[image]

mikeanton, 2017-05-02:
I assume the thermocouples are measuring the air temperature?  Have you profiled the oven by putting a reference thermocouple on a PCB pad, and running through an oven cycle?  You don't care what the air profile looks like, but you do want the profile as seen by the PCB to be correct.  This is why different PCB layouts need to be profiled, as you can't usually use the same profile for different board layouts.

The brown spots are the flux in the solder paste, that looks like it may have been overcooked.  You are probably running a bit hot.

Houseman74, 2017-05-10:
Dear Mike. Thanks a lot for reply. Sorry cannot understand what you mean with "reference thermocouple".
Should I buy a new digital thermometer? or calibrate a thermocouple and then run the profile to see how it differs from the set curve? Sorry I am newbie. Thanks for explaining me.

mikeanton, 2017-05-10:
You need to have an additional thermometer to measure the board temperature with.  I use a multimeter that has a thermocouple input that outputs data via bluetooth to a PC, so I can record the temperature curve as seen by the PCB, though you could just record readings manually as an alternative.  The temperature on the PCB is the one you want to match to the desired reflow profile.  The air temperature might be very different, but you don't actually care what it is, as just the PCB temperature is important.  So profiling the oven is the process required to iteratively figure out what the air temperature profile has to look like to get the desired board temperature profile to be correct.  This air temperature required may be different for different board layouts, which is ideally why the oven needs to be profiled for different boards.

Houseman74, 2017-05-10:
Ok thank you very much for your precious info,I appreciate it a lot. I have also a digital multimeter w/ thermometer feature that via RS232 will transmit the data to PC. I will try this weekend with the three TC (2 internal + the multimeter one) layed on a single PCB with Kapton Tape to let them stay in place. Maybe can import also via serial the temp of the two internal cold-junctioned thermocouples inside the oven using the ISP port and excel it all three (also the original profile) to see and check the differences.  Please one last question. Once I will see that the two reading are different should then I adjust the two GAIN & OFFSET of the two L+R TC inside the menu? or the set point? what's the difference? Or should I create a new reflow profile adjusted to the readings in order to have correct temperature? Thank You in advice. Regards and greets from Italy

mikeanton, 2017-05-11:
You don't want the ones that control the oven to be on the PCB.  These should measure the air temperature.  You adjust new or existing profile temperature values, so that you get the correct temperature on the board, even though the oven actually controls the air temperature.  By profiling the oven, you are taking into account what the actual calibration of the thermocouples are, so the calibration doesn't matter much, as long as it stays within the working range of the A/D (i.e., you don't want the values to saturate). 

Since the coupling between the air and the PCB, and the IR absorption of the PCB changes with different board layouts, one generally needs to use different profiles for different boards, so this is not a set it and forget it sort of thing.  You will likely need to profile the oven often.   

I usually use an alligator clip to hold the thermocouple on the PCB, though kapton tape also works well.  The tip of the thermocouple should be in contact with one of the pads on the board, so that you are measuring the temperature of the copper.  You can bend the wire to give it some spring to keep the tip against the board.

It isn't practical to normally attach the thermocouples to the PCBs to be reflowed, which is why the air temperature is controlled instead.  All commercial reflow ovens do it this way, which is why profiling for a given PCB is so important.

Houseman74, 2017-05-15:
Dear Anton, thank You every time so much for helping me understanding more about.
Thanks to Your suggestions I had this weekend time to make my homework, and here we are. 
Here attached is the excel graphs of the built-in "Profile 4300 63Sn/37Pb" referencing and comparing the data collected from the oven serial port (Temperature SET and Internal TC - the mean of the two TC- now as you suggest placed in the Air, just below the two heat elements, and from an external TC attached via alligator clip direct to the PCB going to a multimeter and logging it to a PC.
I have displayed also the cooling phase after in order to see how it goes. ( 6min reflow+4 cool down).
On one part I can see a delay of "heat-response" of the PCB and after the beep of the machine telling the end of the reflow phase the temperature is still hot even if the internal TC (AIR TEMP I assume) gets cooler and sinks drastically.
But I cannot see something else.
What do You think about? Any suggestion/hints, variation are accepted and appreciated as usual.
Thank You for the time you are spending.
Much appreciated.
Regards
Steve
[image]

mikeanton, 2017-05-15:
Yes, it will be delayed, as you have found.  You should bump up the initial temperature, to try to flatten out the soak period.  You may need to drop the final soak temperature, once you get the beginning of the soak to rise, or it will overshoot.  That might help the reflow time as well.  You could probably lower the reflow temp setpoint, as you are overshooting a bit.  Like I said before, you have to tweak the setpoint values to get the proper curve as seen by the board.  It takes a number of iterations to accomplish that.

Houseman74, 2017-05-16:
Dear Mike.
I have found something unusual running a ramp speed test (image attached) so I am asking You following questions:
1. Is it possible that the Internal thermocouples breezing in the air but that floats nearer to the two Heat elements measures less temperature than the one attached to the PCB? it's about 270°C on the PCB at the end of the ramp when the internal TC just measures 40°C less??!!
2. How can I bump up the initial temp? I have seen the 250°C SET temp is reached only after approx. 3 minutes after.
3.I assume I have to steep the SET curve in order to let the PCB respond quicker, right? Have I manually adjust the profile or is there a way to load into the firmware a custom made Profile?
Best regards and thanks a lot for Your patience
Steve 
[image]

mikeanton, 2017-05-17:
1.  Yes, this is possible.  PCBs will absorb more IR than the thermocouple, so it is not unusual for the board to be at a higher temperature than the air temperature.  It's also possible that the oven thermocouples or the board thermocouple are way out of calibration.  I've heard that many of the Chinese thermocouples don't really measure accurately at all.  But again, this doesn't really matter.

2. You don't necessarily need to make the curve steeper, but you do need to set the temperature higher.  You would like the soak to start at around 150C, and climb to about 180C before the reflow spike.  So change the profile temperatures, and times to make the board achieve the proper profile for the solder paste you are using.

3. I think you would have to manually adjust the profile.  This makes the most sense, as you will have many profiles for different boards that you want to run, since a different profile may be required for each board layout.

xnk, 2018-02-05:
Issue cleanup

closed 2018-02-05 by xnk