UnifiedEngineering/T-962-improvements #114: Dealing with counterfeit DS18B20 chips

mcleod-ideafix, 2017-06-15:
It happened to me that I thought my DS18B20 was a fake one. I read about avoiding some cheap units from Amazon and go though mor realiable distributors. Mine was from eBay. A friend of mine had made the cold junction mod with a DS18B20 from the same vendor and it did work for him, so I thought about giving the sensor another chance. By then, I already had placed an order for 10 units from Amazon (yep, this was before reading the warning).

I checked the pullup and noticed that it wasn't making any effect: the resistor seemed to be in short circuit. After pulling it from the board, I noticed that there is an exposed via between the two pads where the resistor should be soldered (between C20 and R2).
[image]

So I placed a thin piece of kapton and resoldered the resistor, checking that it was effectively working. My original DS18B20 sensor actually worked well!

[image]

Then my Amazon order arrived and I decided to check all the probes using a MBED enabled microcontroller, which works with 3.3V signals. I made a small circuit with one DS18B20 with a pull up resistor and some wires so I can check it on both normal and parasitic mode, and with 5V and 3.3V

[image]

With the defaul pull-up resistor (4.7K), my findings were these (all the chips exihibited the same behaviour):
5V supply, normal mode, one sensor: OK
3.3V supply, normal mode, one sensor: OK
5V parasitic mode, one sensor: OK
3.3V parasitic mode, one sensor: FAULT

Then I read about the pull up resistor that needs to be changed from which the datasheet states IF the supply voltage is 3.3V . I tried 1K:
5V supply, normal mode, one sensor: OK
3.3V supply, normal mode, one sensor: OK
5V parasitic mode, one sensor: OK
3.3V parasitic mode, one sensor: OK

So it seems that lowering the pull up resistor, some DS18B20 units that seem to be faulty may work. Wiring the sensor in normal mode instead of parasitic mode also seems to work.

The demo program I was using for my MBED board allowed to address more than one sensor, so I tried putting two of them "in parallel", with this 1K resistor as pull up (each sensor is separately addressed and queried over the same bus):
5V supply, normal mode, two sensors: OK
3.3V supply, normal mode, two sensors: OK
5V parasitic mode, two sensors: OK (FAULT with the original 4.7K resistor)
3.3V parasitic mode, two sensors: FAULT (could they work if I lower the pull up resistor?)

Hope this helps :)

jasmattkrz, 2017-08-18:
**YOU ARE A HERO**

I bought a DS18B20 on ebay three years ago.  I bought a T-962 this week, it worked terribly.  I did all the base upgrades, using my precious DS18B20, but I got nothing.  I've spent at lease two hours on this; I can't believe I've never seen the "1k resistor" fix anywhere else.  You have my extreme gratitude for the time you out into these tests, and your documentation!

xnk, 2018-02-05:
Another great tidbit for the wiki!

closed 2018-02-05 by xnk