UnifiedEngineering/T-962-improvements #115: The display does not work after the firmware update

ZekMan, 2017-06-25:
After the update, the display stopped working. The backlight of the display blinks after the stove is turned on. After that, nothing is displayed on the displays. 

I was also stupid enough not to make a dump of the original firmware. =\

Can you share the original firmware or help with the display?

balcer, 2017-06-25:
I have old MCU with original Chinese firmware. It will be difficult but I can make dump of the hex. 
Did you tried to flash the uC with unpluged display. This was the problem in my case. I think the flashing interface is also connected to display.

GitLang, 2017-06-25:
Is it possible you have left the connection that puts it into programming mode, maybe by leaving a programming lead attached?. That is exactly the effect you would see.

ZekMan, 2017-06-26:
@GitLang 
I tried to put the controller into the firmware mode and reset it again (Pull n_RESET low briefly). If you talk about this.
`Ground n_ISP
Ground and release n_RESET
Release n_ISP
Activate LPC2000 Flash Utility on the correct COM port
Select Read Device ID and hit OK/Enter to dismiss the "please reset your board now" dialog.
Dump flash or Upload new hex file
Change LPC2000 COM-port to a different port (doesn't matter which one), as otherwise the program will continue to hog the COM port and you can't hook up a terminal program to it.
Pull n_RESET low briefly again to start the software again.`

GadgetGroup, 2017-08-08:
Hi, I just bricked my T-962 it seems.
I have uploaded the firmware 0.51 on a LPC2144.
The firmware is loaded without any problems, but the display is all white, with no text.
Heater and fan comes on when power is turn on.

If any one have a original hex it would be lovely :o)

mattmccrink, 2017-09-26:
Any updates on this? I just bricked my T-962 as well. Serial data seems to be streaming correctly, but LCD is blank...

JohnWestlake, 2017-09-30:
My oven (the larger T962C) has a LPC2134 - are you sure your oven uses a LPC2144?

GadgetGroup, 2017-10-12:
Yes, it is a LPC2144,  so if there is no response for a new hex file, I believe I need to begin to see if someone have made an upgrade kit or maybe buy a http://www.zallus.com/product/zallus-learning-controller-kit/ if the good guy is still in business.

GadgetGroup, 2017-10-12:
Or a Controleo3

JohnWestlake, 2017-10-12:
I just ordered a smaller T962 oven (the smallest unit) - here is the HEX dump (I hope I copied all the required memory blocks).

https://www.dropbox.com/s/b6kvk0z91dchwax/T962%20Original%20MiniOven.hex?dl=0

GadgetGroup, 2017-10-12:
Perfect thanks, I will try to flash the oven.
I just purchased the Controleo3 on eBay, just the basic kit.
I need to have the oven up and running as soon as possible.

romansavrulin, 2017-12-14:
@JohnWestlake , Please tell what tool did you use to dump the firmware?

xnk, 2018-02-04:
My old T962 LPC was not read protected in any way, a simple read from Flash magic was all that was needed. But it sounds really strange that the firmware wouldn't even boot properly. Does the board still look the same as in the pictures here? Any output on the serial link (the one used for programming)?

xnk, 2018-02-05:
Looking through the tested configurations, there are reports of successful LPC2144 installations so is really _should_ behave properly.. If there's any output on the serial port when booting that would be very interesting to see.

eschercode, 2020-11-13:
I'm getting this bug too (with v0.52 firmware), but out of 15 boots or so, 2 have successfully displayed the Unified Engineering logo & UI works succcessfully. The rest of the boots the LCD is just backlight, however pressing buttons gets a reaction out of the oven - eg I can start and stop a bake (though I don't know which profile I selected, obviously). Hence, this is probably an issue with driving the display rather than bricking the MCU.