UnifiedEngineering/T-962-improvements #116: Any stencil file available for the TC-IF PCB?

Houseman74, 2017-08-05:
Hi. I have bought by osh park the PCB of 1-wire for external thermocouple expansion.
Anyone knows how to get a stencil for it?
I am afraid doing some mess with the small pads of the IC. Best regards and thanks.
Steve

GitLang, 2017-08-05:
The stencil was an option when you ordered - I think!. I definitely got my stencil from OSH.

jake-b, 2017-08-05:
I ordered mine from oshstencils.com

Houseman74, 2017-08-05:
Nope. It was not an option. Otherwise I would have ordered it.

Houseman74, 2017-08-05:
Thanks Jake. What kind of files do tou have used? Or is it a repository alrrsdy there. Thanks again

jake-b, 2017-08-05:
I believe the OshPark page where you can purchase the PCB also lets you download the design files as a zip file.  Then upload to OshStencils

Houseman74, 2017-08-05:
Thanks alot pal

closed 2018-02-05 by xnk