UnifiedEngineering/T-962-improvements #118: Dallas 18D20 for cold junction mod

revilor, 2017-11-23:
I would like to share my experience with the Dallas 18D20 sensor available from lots of sellers e.g. on Amazon. You might think 18D20 is 18D20, just as I did. But that's not the case: the Dallas 18D20 does not provide the 1-wire mode the Maxim 18D20 does. Therefore the cold-junction mod as described in the Wiki does not work. Fortunately the Dallas type can be used anyway. You just have to run the sensor with external supply voltage. One easy spot to get a 3.3V supply for the senosr is in front of the ISP header. There are two tracks on the PCB where the outer one is GND and the inner one is +3.3V. 

[image]

The pins of the Dallas sensor are GND - Data - VCC when look at the flat side with the label.

[image]

So the cold-junction mod using the Dallas 18D20 goes as follows:

- remove a bit of the green solder mask to free the 3.3V copper track in front of the ISP header and put a bit of solder there
- solder a piece of wire to the VCC pin of the sensor
- solder a piece of wire to the DATA pin, same as in the original mod
- solder the GND pin of the sensor to ground on the PCB, same place as in the original mod
- add the 4k7 resistor - a regular non-SMD type also works, by the way - and connect the DATA wire as described in the original mod
- solder the sensor's VCC wire to the spot prepared in front of the ISP header
- you are done!

In the end everthing should look similar to this:

[image]

revilor, 2017-11-23:
I created a wiki page https://github.com/UnifiedEngineering/T-962-improvements/wiki/Dallas-18D20-for-cold-junction-mod which probably is a better fit for this content. So this issue should be closed or deleted.

xnk, 2018-02-04:
Closing as the wiki was populated with the info

closed 2018-02-04 by xnk