UnifiedEngineering/T-962-improvements #119: T-962 vs T-962A Performance

radensb, 2018-01-15:
Forgive me if this is not the correct place to ask this, but I have been looking at both the T-962 and T-962A for a few weeks trying to understand if there are any performance advantages using these mods and FW between them. I am trying to decide which one to get to perform these upgrades. I am planning on small runs of dev and project boards mainly for my own use and possibly limited runs of boards for community members of some of the forums that I subscribe to.

As I understand it, the T-962A model is larger and has 4 heating elements (at 1500W) controlled by an SSR, but basically the same oven as the T-962 (at 800W) in all other respects. Now that this project has been available for the past few years with many users of both ovens, is there any consensus on which oven performs the best all-around? Things I am thinking about are thermal masses, larger area = larger cold spots, drawer area usability (edges of the drawer become unusable due to uneven heat distribution).

My initial thought was to get the bigger A model as I could support reflowing several boards in one go if I grew into that need and that the larger heated area would stabilize clod spots and help with heat distrabution (with the new FW). Then, I began to think that the smaller unit would be more stable? Now, I have no clue what to think.

I am interested to hear from users of both ovens that have performed these mods and FW upgrades. As in, if you had to choose over again, which oven model (T-962 or T-962A) would you choose and why?

I was really hoping to find some comparisons, videos, data, on this online, but I could not get much more than "the A model is bigger and rated for 1500W", and the famous "these ovens are crap" when refering to un-modded, stock units with factory firmware. Not very helpful...

I appreciate your feedback!

xnk, 2018-02-04:
I have been using the non-A version since day one. It is considered a bit slower than the A-version when it comes to ramping speed - it is very close to being underpowered when running lead-free boards with the main fan running (slowly) to distribute the heat. Running the fan even slightly too fast will result in an oven that won't be able to keep up with LF temperature profiles. The A version has been said to be a bit more forgiving (as it has the same size fan but larger heating elements). There seem to be some differences between the 120V and 230V versions as well as the currents are lower in the non-US versions leading to less voltage drops.

radensb, 2018-02-16:
Thanks for the follow up. I went with the 'A' version of the T-962, and I'm glad that I did. Especially after reading your reply. I decided that the extra room of the 'A' version would also allow more flexibility for upgrades.

I have spent the last month overhauling my oven with the following enhancements:
1. Upgraded Firmware (v0.5.2 - THANKS!)
2. Custom PCB for 4x - MAX31850K thermocouples, System fan control, and Serial to USB interface.
3. Replacement adjustable thermocouple tubes through the lid to vary their location
4. Convection fan from a toaster oven mounted in the lid and insulated.
5. Heater tube diffuses to prevent direct heat exposure.

I plan to test this setup when I get the PCB assembled and add to the Wiki if it is successful.

GitLang, 2018-02-16:
Are you using all 4 thermocouples, averaged somehow, for the temperature control?

radensb, 2018-02-17:
My understanding is that the firmware should be taking care of that. If there is no thermocouple present on the device (ie: it is unplugged), it is represented in the data as a fault indicating that the temperature data is not valid and is not included in an average
.
Werner, correct me if im wrong. My plan was to 3D print a box (actually, printing right now) that has two K-thermocouple panel mount sockets that are wired to the other two MAX31850's on my board. When I want to use one or both of them, I can connect a thermocoupler to the socket on the box and feed it into the heat champer (probably taped to the board(s)).