UnifiedEngineering/T-962-improvements #123: New T962/T962A Controller - PIC32MZ based (Drop-in replacement)

MikePhillyFlyers, 2018-02-08:
Hi All,

**First off, Kudos to WJ for all of your exceptional work on making this software upgrade.  Just truly outstanding work, and we all appreciate you making this free for everyone.**

I decided after getting this oven, and re-flashing with your software, that the crap stock board it comes with does your software no justice.  The stock board is just so bad, such a cheap and sucky design.

So I designed a new board, based around the PIC32MZ micro, and incorporated all of the hack-mods that have been going around, as well as tried to improve as much as I could.

I realize most on here probably have no interest in building a new board, but I still wanted to do it, so at least I could make my own oven work so much better, and have a board that had much more potential for future upgrades, etc.  I posted everything via the project link below (and source is linked to github, as well as the oshpark PCB, and schematic,etc are also on there)

https://hackaday.io/project/47222-t962a-controller-drop-in-using-pic32mz

I know I could have shrunk it down a bit, but being I wanted it to still mount in the existing holes, so I didn't try to squeeze it down anymore...

 **I apologize for posting this note as an 'issue', I didn't know how else to put this out there?**

MikePhillyFlyers, 2018-02-08:
Hi,

Yep, this is 100% working, I've been using it/testing it for a bit now.. currently I have the code setup for 10 custom profiles along with the stock profiles WJ had, plus I added a couple from the original...I didn't see any reason to enable all 31 custom profiles, as it makes browsing through the entire list a real pain!

There is room for 31 total custom profiles (as I setup the first eeprom block for just 'config' data, even though 99% of it is not even used now, but down the road maybe?)

Yeah, maybe you all can find ways to bring the cost down?  as the OshPark board cost is $105.. uggh.. and the BOM comes to around $80 for all the parts, although you can cut out some of the de-coupling parts, etc, (caps, inductors) if need be...

wulfmans, 2018-02-08:
I will look into this and report back.
My Chinese board house can send me 5-6 un populated boards for under 100
bux
If my PNP was operational i could populate them fast.

GitLang, 2018-02-08:
Have you looked at PCBway.com ?. 10off through plated boards, up to 100mm x 100mm for $5. And they are really good quality. They will supply a stencil for top and bottom for another $10

GitLang, 2018-02-08:
Mike - waht inspired you to use a PIC?. Seems such an unusual choice these days>

MikePhillyFlyers, 2018-02-09:
Hey Guys, 

yeah, I had looked at those chinese sites before (like pcbway, etc), I'm seeing mixed reviews on whether their quality is good or not?  that's why for now I was using OshPark, even though they are pricier, their quality is always top notch.

As far as using the 'PIC32MZ', I actually used it for a project a year ago.. I had spent some time researching numerous different microcontrollers, that had a good amount of built in functionality, and also good cpu speed.

I couldn't find anything that was even close to the PIC32, as I wanted something that had SPI, I2C, UARTS, PWM, GPIO ports, TIMERS, USB, etc, etc.... and it can run up to 252Mhz CPU clk....

I know there are much faster processors to use, but most of them don't have much as far as built-in modules, if anything they just have GPIO, and not much else...  I'm not in for bit-banging I2C, etc...

But that's just from all the digging I did at the time, maybe you guys are aware of better ones out there?

wulfmans, 2018-02-09:
I use these guys for commercial boards for clients
 https://www.elecrow.com/pcb-manufacturing.html
they are in Shenzhen and do top notch work and at a very low cost.

they even do board assy for a reasonable cost.
One of my clients has boards being delivered from them this week.
Only failure is they will substitute parts sometimes.

mikeanton, 2018-02-09:
I've been using PCBWay, and Smart-Prototyping for all of my production run boards for a couple of years now.  They are both very good for bare boards, but I've never had assembled ones from them, as I usually do that myself.  You have to figure out some of the oddities of dealing with them (like needing two drill files for plated and unplated holes), but once that is worked out, you can get consistently good quality from them.  I find Smart-Prototyping a little easier to deal with, as at least their quoted price is what you will actually be charged.  PCBWay will often raise the price of the boards, because their quoting system doesn't always account for everything.

I'm pretty sure OSH Park boards are fabbed in China anyhow, so you might as well deal with one of the suppliers directly, so you can get better pricing.  That said, OSH Park is probably still the cheapest, if you want to get really small boards, like under a square inch, so I've used them for that.

GitLang, 2018-02-09:
MikePhilly, read more carefully at who is writing those reviews, and how much older than 14 they are. My experience, and that of colleagues, is that the boards are top notch, as good as any European manufacturer costing more than 20x as much. The "$5 for 10" is almost always a loss leader with any of these people - get the hobyist hooked, or give small manufacturers cheap prototypes, then get their production business later. There's no better incentive to impress clients than with quality. As soon as you go >10 boards, or 2oz Cu, or ENIG finish, the prices goes higher, but still cheap by western standards. Remember, they are using the same chinese production equipments as all the western PCB makers :). Mikeanton - if you panelize your small boards, you are back to paying by the sq.ft. in quantity like any other boards.

wulfmans, 2018-02-09:
Would you happen to have the eagle design files ?
I would like to convert them to orcad and allegro

MikePhillyFlyers, 2018-02-09:
Hey Guys,

yeah, you could be right, the reviews I were reading were lengthy reviews by people over at EEVBLOG, so I was putting a bit of faith in them, but maybe too much... I had always wondered if OshPark was outsourcing to China, they probably are :)

Anyhow, I added an 'Eagle' folder into the github repo for my project, in there are two zips, one for the Eagle .pcb/.sch files, and other for the gerber files....

mikeanton, 2018-02-09:
GitLang: Yes, I am aware that they are cheaper if panelized, but if you only need a couple OshPark is still far cheaper.

Lately I've moved back to Smart-Prototyping from PCBWay, as they have finally reduced their pricing on panelized boards.  They used to charge a small fortune for submitting panels of the same design, unless you contacted them and complained.  That has changed recently, and now their pricing is much better than many of their competitors.  Their quality has always been top notch too, based on my experience of ordering many hundreds of boards from them.  Customer support has been really good too.

MikePhillyFlyers, 2018-02-10:
FYI,

I updated the repo with datasheets for all the important components..  :)

Thanks again guys for the pcb recommendations, I'll push some of my old designs to some of them and see what estimates would be, just to compare...

xnk, 2018-02-13:
That is a very interesting project!! The fan control is the one thing that is really annoying on the stock board, the control that is there was the best I could do. The stock firmware did no attempts at any kind of speed control at all.

MikePhillyFlyers, 2018-02-13:
Hi Werner,

You did an amazing job writing this software!  This is your software ported over to the PIC32 that is running on this board.  No way would I even attempted to make this board if you hadn't written this software, and posted it up for us to use.

But yeah, the stock board this oven came with is total garbage.  Not unexpected though, that's usually how they approach these low-end products, keep the cost as low as possible, so put as little engineering as possible into it.
As soon as I started trying to use this oven, after I flashed it with your software, I just saw how their triac circuit was just horrible at running that cooling fan.  You could hear it making those awful grinding noises, as the fan is being jerked on and off

I know I could have picked a much more powerful processor than the PIC32, but once you go into the next class, you are usually dealing with BGA packaging, which I did not care to deal with for a small project like this.  Adds a new level of difficulty that I wasn't yet ready to tackle.  Of course that is what the oven is made for!  But, being that we are still stuck with the same small LCD, and the rest of the oven hardware, it seemed more sensible to just go to a bit better MCU, but not too much overkill.

Anyhow, I kind of did this as a side project, as I have some fun things at home involving some BGA work, that I wanted to use this oven for, but I couldn't stand that crappy stock board anymore, which is why I went this route.  It's working well now, and I still want to do more with it of course..

I didn't yet do anything with the USB port, it's wired up, but I haven't written any code for it, and I didn't port your UART routines yet to this board, although the UART setup is connected and ready to go..

The only real PITA with the PIC32 so far, is that I can't find any regular ICSP software that will flash a binary, like you could do with the LPC21xx mcu on the stock board.  I know the PIC is basically ICSP, but it seems you need to at least get their PICKIT3 (or a $20 clone version), in order to flash it.  

It's probably just that no one out there has bothered to write a simple freeware flasher for it?  oh well....for me, I already was using the PIC32 for another project, so It was an easy decision to attempt this project with it...

But kudos again to you for all the RE and coding you did to get it working, that was a ton of work. All I had to do was get it all working against the different hardware.

jdeglavina, 2019-03-23:
Curious if you have any extra pcbs from this? I may need to replace the stock controller anyway. 

By the way, has anyone used this with the regular t962? I assume the only other thing you would need is a relay for the heating elements, correct?

jdeglavina, 2019-03-23:
Also, are you near Philly?  I am too.

MikePhillyFlyers, 2019-03-23:
Hi John,

I used to live right outside philly in Drexel Hill, but we live down in Maryland now. I do have 1 extra board left, you can certainly have it, I'd gladly mail it to you.

Also, if this is for a T962 oven (not the 962A), somone on the discussion on my hackaday page noted that the board will not fit in the smaller oven, he was making another spin of the board to fit that oven?

Check the comments on my project over at hackaday..

Also, I think you can send PMs over on hackaday?  we can talk privately there I think if you want me to send you the board?

jdeglavina, 2019-03-24:
Cool! Thanks for offering! I sent you a message on hackaday.