UnifiedEngineering/T-962-improvements #124: is the 18D20 discontinued or is that a typo?

NicholasTracy, 2018-02-12:
I'm wanting to do the cold junction compensation mod but all the wiki says is to use an 18D20 1 wire temp sensor. When searching for one on-line either on Amazon or Mouser or even just a standard Google search it doesn't show up, instead I get hits for the 18B20 which is also a 1 wire temp sensor. in fact I have a few. Is it possible that this is a typo or that the sensor is discontinued? Maybe someone has a link to a seller that has these? If not what would be the possibility of adding support for different i wire sensors or maybe even an analog sensor such as the TMP36 or TMP35 which I also have on hand. Thanks!

xnk, 2018-02-13:
Oh, the code will detect a DS18B20 or more recently also a DS18S20, the D must be a typo! That said, the on-board thermocouple interface (that will utilize this cold-junction compensation) is really bad - if you want better results you really need one of the digital 1-wire or SPI interfaces with built-in cold-junction compensation, as listed in the wiki under improved TC interface(s)

NicholasTracy, 2018-02-13:
Thanks, I'm looking in to that for the future but right now I'm under a time crunch so I'm looking for something I have on hand to help get it along. I have to use lead free paste in the project I bought this for, I have a few temp sensors laying around. Going forward how hard would it be or is it possible to add support for a simple thermistor bead or an analog sensor such as the TMP35/36? I guess that would depend on if there is a spare ADC port available and if it would be difficult to add the look up tables. I'm sure this would be valuable for some of us that can't get the max chips just yet and need better then factory accuracy for the time being. something cheaper then the Dallas 1 wire sensors and something someone might be more inclined to have on hand if say they are in to 3D printing or something similar where the analog sensors are more common place. I would be happy to look in to it if its possible, I would just need help getting pointed in the right file and I'd fork and code it.