UnifiedEngineering/T-962-improvements #131: serial line not usable as human interface, really?

taliesin, 2018-02-19:
Hi,

  I just tried to use the serial interface and found it unusable from the terminal.
The reason (in main.c) being:
    `int len = uart_readline(serial_cmd, 255);`
returns immediately after a single character, which only works for '?'

As far as I can understand this call should be non-blocking as it would block the keyboard interface otherwise. So this needs some kind of re-write, I think.
I'm feeling like I'm missing a point somewhere, how could it ever work the way it's done?!

Helmut

taliesin, 2018-02-19:
Just found Issue #22 which already states the problem.
I'm prepared to take some effort here, any further suggestions?
Probably I'll try to plug in a little shell like from ChibiOS: http://chibios.sourceforge.net/html/group___s_h_e_l_l.html

Helmut

GitLang, 2018-02-20:
A pre made and tested shell is a good idea. Did I get the wrong idea or does that shell only receive data and not transmit?. I could see shellGetLine() but not shellPutLine()

taliesin, 2018-02-20:
It simply uses chprintf() a ChibiOS specific output (resembling printf but working on a chibi stream).
I'm currently dissecting the main working thread to be able to plug it in.

taliesin, 2018-02-26:
Porting the ChibiOS shell did not go well, I used SimpleCLI instead (see my pull request).
This is therefore considered closed.

closed 2018-02-26 by taliesin