UnifiedEngineering/T-962-improvements #133: Some advice on 350+ pin BGA soldering?

MikePhillyFlyers, 2018-03-18:
Hi All,

I'm getting ready to actually start 'using' my oven, instead of just playing with the firmware, LOL.

I made an eval type board for a BGA chip that is 388 pins, it's an STMicro chip from around 2004-2005 timeframe, I'm not completely sure if it's balled with lead or lead-free solder balls?  

anyhow, was looking for some advice from anyone that's actually used this oven already for doing any boards in their oven using BGA chips with > 200 pin counts...

1) **this is a fresh/unpopulated board, with ENIG finish, I have NOT tinned the pads or anything, so do you guys advise tinning the pads first, then applying flux, etc before doing reflow?
(or should I leave the pads as just bare ENIG finish?)**

2) **did you guys make your own custom reflow profile, or using one of the stock ones?**

3) **I've attached 3 of the profiles I was going to pick from, 2 of them are like the Wave3 & Wave8 stock ones, and third is one I made, basically following a profile suggested from a ST reflow doc I came across...**

[image]
[image]
[image]

Thanks everyone for any advice you can provide!

mikeanton, 2018-03-18:
Profile the board and generate a custom profile, or you will have no idea if it reflowed properly.  I would apply paste, rather than flux, as that is what is normally done when assembling with these sorts of parts.  The flux in the paste should be sufficient.

MikePhillyFlyers, 2018-03-19:
Hi Mike,

Thanks for the feedback.... couple of questions...

1) so when you say "profile" the board, are you saying something like paste a couple of test components with paste,etc to see what temp range gets it right?  or you referring to some other way?

2) for the actual pasting of the BGA pads on the board, I'm guessing you mean using a stencil or whatever to just get paste on each pad, then just drop the chip in place, not putting any flux on the BGA chip/balls or anything?

Thanks so much for your help, I've been doing surface mount boards myself for about a year now, but just soldering the stuff myself, with the help of my magnifier/scope if/when needed... but first time I'm attempting a BGA chip, and even using this oven...

mikeanton, 2018-03-19:
Every PCB will actually require a different soldering profile.  This is because it is the air temperature that is controlled by the oven, but what we actually care about is the temperature as seen by the board, and different PCB layouts will absorb the IR energy differently.  So, usually one puts a thermocouple on a SMT pad on the board that is connected to some sort of data logger.  I use a multimeter that has a thermocouple input, and a Bluetooth interface, so that that data can be captured by a tablet.  The idea is to adjust the oven control profile until you get the correct soldering profile as seen by the board.  It usually takes a number of iterations to adjust the profile.

Yes, use a stencil to apply the paste, and put the chip in place.  BGAs are not that easy to solder, especially since you can't inspect them without an X-ray machine.  The paste will supply enough flux, and additional solder to wet the pads and balls properly.  Even if the balls don't melt, the connection should be fine.

GitLang, 2018-03-20:
My opinion is that you are starting on something way too optimistic. A 388 pin BGA is a big deal and not easy for experienced amateurs to get right. Maybe you could find up some old left over bare boards that  take 0805 resistors and caps, then do some practice runs with them. That said, I started without stencils and quickly learned life was very very much easier with them They have come down in price a lot too. Maybe you could start with a simple board for an upcoming project. You can get 10 boards up to 100mm x 100mm (4" x 4") for $5 total from places like PCBway, and they now do a combined stencil for top and bottom for another $10. Downside is shipping and the wait, which is $17 for 15-30 day china post, or around $30 for 3-5 day DHL. 

What mike above says about profiling is the proper thing to do and essential if you are doing bigger batches in industry, but for one or two home boards I haven't found it necessary yet. I've found by experience that 60/40 tin/lead solder paste (I use Kester) works fine with a simple profile and without and board testing, but beware it's toxicity and legality. I always pre-bake the oven at 50 degC for 10+ minutes to get everything warm. I have the center of the tray marked with cross axis sharpie lines and always try to arrange boards centrally. The profile I use starts at 40 degC, climbs at about 0.8 degC per second to 160 degC. It soaks there for 2 minutes then rises at a similar rate to 220 degC where it sits for 30 seconds before cooling at 1 degC per second. By the time it's down to 100 degC or so you can open the tray a bit to cool faster. If the cycle finishes that early you can put the oven on bake at 30 degC and let it fully cool. Cooling too quickly can stress solder joints and leave the solder brittle. I get great results everytime with that regime. The solder paste I use was just out of date when I got it off ebay 2 years ago. I've kept it, triple wrapped, in the refridgerator and bring it into the room a day or two before needed to gradually warm to room temperature. I've never used any extra flux.

When/if you do move on to BGA, have a look at what the phone repair boys do, and the equipment they use, and what is available on ebay, aliexpress, etc. You probably should equip yourself for BGA chip removal and re-balling, for when the soldering fails or has shorts, opens, etc. You can find them the long way (by testing) or the standard way by X-Ray inspection. I've gone down to 0.6mm pitch routinely and 0.5mm pitch once. I still haven't gone near BGA :)

MikePhillyFlyers, 2018-03-20:
Hi Guys,

Thanks so much for all of your feedback, I really appreciate all of the details from your experiences!

I'm always up for a tough challenge, believe me, I'd rather it be too hard versus too easy :)  I have quite a decent RE lab setup, only thing I don't have yet is an actual heating plate, I plan on getting one soon, but for now I will just stick with the oven.  But I already have a decent scope, Hot Air Rework, reballing rig, etc, etc.  

This BGA is not nearly as insane as it sounds, it's quite an older BGA chip, so it's quite large, is more forgiving with the 1.0mm pitch, and has a large area of empty space, with an 8x8 array of balls in the center, that are ALL GND.
(see the attached picture, an old infineon chip is same footprint as this chip)

I have a couple of stencils for it, and plenty of qty of the pre-balled chips, so I"m not worried about wasting some with bad solder joints, etc, it's just a fun hobby, and nothing production at  all.

So I will take your guys advice and take the stencil approach, using solder paste, and use my oven carefully with all 4 TCs equipped, two on the PCB...Also I will do another test with just using the pre-balled BGA and flux, without paste.  As I also know many people do plenty of BGA repair, etc, without using stencils or paste, nothing but flux and the chip.

And thanks for the detailed profile information, I'll create a custom one as well to mimic what you are describing.  I'll be giving it a whirl and whatever happens happens.. worst cast I remove the failed chip attempts, and just clean the board and try again... or until I ruin the pads :)

[image]

MikePhillyFlyers, 2018-05-01:
Hey Guys,

Just wanted to get back to you all, to let you know I followed all of your great advice, and it's working excellent.  I took you advice, had a stencil made, which of course makes things super easy.  Put down some solder paste, which certainly does help the bga balls wick to the pads!   

I made a profile that is very much like GitLang described above, profiled a board or two, with some solder paste to see how it flowed, etc, working great, I'm using my new board I described in the other post.  So I have two extra TCs that I attach onto the PCB for more accurate temp sensing.

I run the oven for about 10-15 minutes, holding at about 50 deg C, I'll run it up and down a bit too just to warm it up, etc.. then after that put a board through.  The BGA attached without issues, so far everything is fine.

Thanks again everyone here for your help, I never used stencils in the past, but they sure do make it fun!

GitLang, 2018-05-01:
Well done that man!. I've also found a 50C oven warmup for 10 or 15 minutes is well worthwhile.