UnifiedEngineering/T-962-improvements #134: System Fan Control Manual/Test Mode?

negativekelvin, 2018-03-19:
Did the system fan control mod but did not see a way to test it by manually turning the system fan on/off.

negativekelvin, 2018-03-20:
I see it spins briefly on powerup so I guess that means it works.  Would prefer a manual control in setup menu.

GitLang, 2018-03-20:
I'd just forget about the little fan. If it's ever needed (rare IMO) then it will progressively spin up. If you want to test aim a hairdryer in the electronics enclosure, through the fan blades maybe. Or you could temporarily lower the threshold temperature where it starts to turn on.

negativekelvin, 2018-03-20:
Probably true but when you go through the trouble of doing the mod you want to see the result.

>Or you could temporarily lower the threshold temperature where it starts to turn on.

How?

GitLang, 2018-03-20:
Edit the source. Much easier to just aim a hair dryer in through the fam though. Or if the case is open and power up (carefull...) then locally heat the cold junction sensor. It is that which senses the temperature of the elctronics enclosure to bring the fan on.

You want the SystemFanSense_Work() function in systemfan.c

`	if (Sensor_IsValid(TC_COLD_JUNCTION)) {
		float systemp = Sensor_GetTemp(TC_COLD_JUNCTION);

		// Sort this out with something better at some point
		if (systemp > 50.0f) {
			sysfanspeed = 0xff;
		} else if (systemp > 45.0f) {
			sysfanspeed = 0xc0;
		} else if (systemp > 42.0f) {
			sysfanspeed = 0x80;
		} else if (systemp > 40.0f) {
			sysfanspeed = 0x50;
		}
	} else {
		// No sensor, run at full speed as a precaution
		sysfanspeed = 0xff;
	}
`
So it is doing a progressive approach. Turns on at 80/256  when  greater than 40 degC, then progressively more as it heats up.  128/256 when > 42 degC, 192/256 when > 45C and full speed over 50 degC. The end 'else' clause just runs it at full speed if it doesn't find the cold junction sensor.

I believe it still uses that sensor even if you are using the (much superior) MAX31850 thermocouple interface which does accurate cold junction compensation itself, so you still need to fit the onewire sensor. Someone will hopefully correct me on that if I'm wrong.

negativekelvin, 2018-03-20:
Thanks, but if I am going to edit/compile then I would just add it to the setup menu.

GitLang, 2018-03-20:
There you are then, you've got two solutions now :)