UnifiedEngineering/T-962-improvements #135: Anyone done a zero crossing mod for cooling fan?

negativekelvin, 2018-03-21:
Has anyone done a zero crossing mod for switching the cooling fan?  Either by replacing the triac driver with an inbuilt zc or by using the 9vac as a zc measurement input?  Any suggestions?

GitLang, 2018-03-21:
The solid state relay block which drives the main fan does do zero crossing. A proper solution to the cooling really needs replacement of the fan and hardware and software control of it, then we would stand a chance of getting the PID sorted out. At the moment is not good - lots of overshoot, ramp lag, etc. If you think how well a 4 pin PC fan has it's velocity controlled we should be able to do better here. But it does need a fan that will go down to very low RPM. That may mean a custom solution, such as using the rotor of a commercial fan and driving from it's own controller which the ovens CPU can command. Tht way it is possible to get good speed control from virtual standstill to full.

negativekelvin, 2018-03-21:
The moc3021 and bt139 does random phase control of the fan doesn't it?

taliesin, 2018-03-22:
I did this modification, see [wiki](https://github.com/taliesin/T-962-improvements/wiki/Basic-electrical-changes), exchanged the moc3021 with moc3083.
You'll have to remove (short) R16 as-well, it delays the turn-on by about 50V. In its original state it turned on whenever it was switched.

The main reason for the erratic fan behaviour is the PWM control (and yeah there are better fans for this purpose). We could use burst fired control instead of it. See https://de.wikipedia.org/wiki/Schwingungspaketsteuerung (sorry there is no wikipedia entry in English for it, but the pictures there should be quite instructive).

At level 14 (for the fan) currently the fan is turned on for 14/256  * 2s = 109ms, this is a bit more than 2 full cycles (assuming 50Hz), so 2 cycles on and 38 cycles off, burst fired control would generate 0.5 cycles on, 9 cycles off, 0.5 cycles on, 9 cycles off ... ( or at least 1 and 19, half cycle control must avoid generating a DC component). This should improve performance.

The PWM also creates another issue if on-time is changed within 2s, it gets jumpy. As the PID controls the fan in a 250ms interval, caveat emptor. With burst fired control the burst generator can be implemented to filter the stepping and create nicely controlled bursts.

taliesin, 2018-03-24:
Just saw the pictures in the wikipedia link actually only show PWM. :-( the point in burst fire is, that the control tries to create as few low frequency components as possible.

GitLang, 2018-03-24:
Still don't see why burst fire cannot be done. Whatever the wiki shows about the fans PWM control, the SSR module used the T962a definitely zero voltage switches the fan. So it ends up as a horrible mess. I'm quite surprised the fan tollerates that sort of treatment.

taliesin, 2018-03-24:
Maybe I did not express myself correctly. I meant the wikipedia animation does not show the burst fire control very well. Of course it can be done in the T962. BTW, both the T962 and the T962a DON'T use a SSR for the fan, but only an opto-triac (MOC3021) and a triac (BT139, the one without a heatsink). There is no zero-crossing circuitry there.
Using the MOC3083 adds zero-crossing, but you will have to remove R16 which increases the switch-on voltage by ~50V, i.e. the triac will not switch at (slightly over) 0V but at 50V, even with the MOC3083.
With the original opto-triac it turned on whenever the PWM said so (and the voltage was above 50V).

The higher the mass of the rotor the more forgiving it is for low frequency components in the driving current. The PID is pretty unstable, it more or less constantly oscillates (have a look at the reflow log), except the setpoint is far from the current value. This jumpiness in PID output value together with PWM control (2s) creates strange output sequences on the fan driver, which is why it turns on much too long somewhere in the soak zone (at approx. 150°C), at least with my oven.

But in the end this all only influences air temperature! The PCB temperature is lagging quite a bit and will not jump anyway (due to its mass). So the PID controller must be adjusted correctly AFTER sampling the relevant temperature for soldering (that of the PCB).

GitLang, 2018-03-24:
This machine definitely has two SSRs. I thought it was only the smaller model that used a discrete triac. Agreed, the PID is very rough, is way out on slope, out on set point and oscillates all over the place. There just seem to be too many non linear things going on to make it much better by altering constants in the standard oven as supplied. Yes, bigger the rotor inertia the more it will smooth things out mechanically, I meant the electrical and bearing hammerings it is getting.

I'm going to make an air flow straightener. I have a length of 15mm plumbing tube spare, so I will cut it into 40 or 50m lengths and solder it together in a frame. Maybe I'll use some old paste and do it in the oven :)

taliesin, 2018-03-24:
You saw my straightener? A bit crude, but works fine, has a 10mm grid. And quite some work to do :-)

GitLang, 2018-03-24:
Yes, excellent idea. I know it's sucking air into the oven but somehow cardboard worries me :).

Have you thought about a diverter?. Maybe two small vanes in a Vee shape, to send some portion ofr the air stream left and right, and leave the rest going more or less straight.

GitLang, 2018-03-24:
> This machine definitely has two SSRs.

Oh no it doesn't!. Huge appologies, I was thinking of something else with two of the same SSR's side by side. This oven has one SSR for the heater, and an MOC3021 driving a BT139 for the cold air fan. Again, appologies.

taliesin, 2018-03-24:
No problem, I thought there might be yet another model. About the diverter: The generated swirl of the fan is very complex and multidimensional (different speed and direction) in every location. A rel. simple diverter can't fix such a stream (at least rarely).

I once saw an animation by ebm-papst about swirls, but I can't find it right now. Many solutions use honeycomb like structures with rather small holes to create high friction for non-horizontal flow components.
Partly this is counter-productive as turbulent air is more effective concerning convection (Reynolds number). 

I would not be too concerned about the cardboard, paper should not auto-ignite below 230°C (https://en.wikipedia.org/wiki/Autoignition_temperature) and I left a gap between the inner housing and the straightener of about 5mm.

radensb, 2018-03-24:
@GitLang Perhaps you are thinking of the ESTechnical upgrade?
[image]

The above is a T962A.

radensb, 2018-03-25:
@taliesin 
Have you thought of reversing the fan (physically) so that it exhausted the heated air out the back? It would draw the cool air in from the bottom, probably with fairly consistent distribution through the drawer. The question is, is the fan suited for that? Its a brush-less metal fan, so i would think it could handle the brief heat.  If so, you wouldn't need to straighten the air being blown into the heat chamber and could keep the fan in the case. This also opens the possibility to exhaust the air through ducting to another location instead out the bottom, which I have never been a **fan** of.

taliesin, 2018-03-25:
I just didn't want to expose the fan to air at 250°C, but you're probably right, it would work. The inner box sheet metal is bent to the top and blows (or sucks) air from the lamps which primarily cools these.
They should be the hottest parts when cooling starts (having the largest mass and temperature), so blowing cool air onto them is probably more effective than the other way round, just my 2c.

The limiting factor is again the flow rate. The higher the delta T, the higher the amount of convection heat.

MikePhillyFlyers, 2018-05-01:
Hi All,

just wanted to chime in, from my experience as I was working on my replacement board for this oven, the main issue with trying to do any type of 'proportional' voltage control with the AC cooling fan ( which is what any type of ZeroCross switching is actually doing) is that the fan 'jerks' quite often because of the irregularity in the voltage waveform that is produced.

ie see this:
http://www.windandsun.co.uk/media/171346/Proportional-Control-Technologies.pdf

Burst, or Proportional (the opto/triac or SSD all end up doing this) works great for the heater elements, as they are much slower to react, but the fan does not run nearly as smoothly when doing this.

Good AC dimmers today use the Phase Angle approach, controlling how much of each individual AC wave is on, the switches are in sync with the 60hz voltage, allowing this to be possible.

I know the fan did get a little better once I replaced the stock Opto with a ZCD version (like you guys mentioned above), but it still jerked like mad at slower speeds, makes that annoying sound.

Please feel free to steal anything from the board I made, the speed control is nice and smooth now.  It wouldn't be too hard to make a little board to just use the speed control part, basically the FL5160 phase cut controller and two mosfets, and the 8-bit DAC to control the 5160.  

Oh well, just wanted to put that out there, I know it's prob. more work than most want to do... just swapping in an SSD or replacing that Opto with a ZCD version certainly will help a bit, if nothing else..

taliesin, 2018-05-02:
Hi,

  the original fan is a http://img.ebmpapst.com/products/datasheets/AC-axial-fan-4656N-ENU.pdf?_ga=2.86512984.1154697111.1525283009-1948972485.1520450846
 
 This is quite astonishing as it's a quality long-life fan from an European manufacturer! Concerning the drive method, even cutting the phase is not what's perfect for these types of motors (might be either shaded pole or a capacitor motor, I'd assume the first, it's not stated in the datasheet). The only thing one could do about it is using a full blown inverter (controlling both voltage and frequency), but this is way out of scope.

After all the investigations I did concerning the temperature control of the PCB, I think the whole fan thing is pretty useless during heating, when you use 'loaded thermocouples', controlling the air temperature is not what gets you good results anyway. I may end up using the fan only during cooling again! Certainly more fine grained control of the fan speed seems unnecessary.

If I had to implement phase cut control, I'd use one of the rectified transformer signals and run a timer delay from there (of course only with the original MOC3021 as ZCD would be counter productive :-) and remove R16!)

rolandking, 2019-02-10:
Reading through all the interesting discussions on this project prior to purchasing my own T962A. 
@MikePhillyFlyers that's a nice circuit to control the fan, somewhat annoying you have to go SPI -> D/A and then control the FL5160. Did you actually get it working reliably, I read some of your comments on your hackaday projects and it seemed you were having odd brownout issues. 
I wondered about the zero crossing mod, fans are inductive which would not usually be the best for ZC however they have inertia which would tend to mitigate the back EMF. 
Has anyone starting with a fresh board design though about changing the fan completely for a 12V DC one, removing the 9V transformer and putting a proper 10W or so 12V power supply on the board to run it? A 12V DC fan of the same size and airflow needs about 500mA and ought to be much easier to control with PWM and save any more mains switching. Most of them of course are plastic, but it's inflow rather than outflow so I don't see an issue. $25 makes the whole janky AC fan issue go away.

GitLang, 2019-02-10:
What exactly do you mean by the janky fan issue?. The fan has never given me much problem. The known issues are how it is driven in the TC loop, and the turbulence it introduces. There have been some good ideas on diffusers, but noone has yet done any serious work on the PID coeffs or control loop, which would reap a lot of benefits, not least would be better temperature control.

I have a to agree, a 12VDC fan would make it easier to control. I just found that Sankyo Denki (San Ace) make a range of high temperature fans, the 9GT series. They are available 40mm.sq to 120mm.sq outline, 25 to 38mm thick, and 12VDC or 24VDC. There is also an option for PWM control.

https://eu.mouser.com/new/sanyo-denki/sanyo-denki-9gt-fans/
https://eu.mouser.com/pdfdocs/Wide_Temperature_Range_Fan-4.pdf

One problem that springs to mind is minimum speed. A PWM fan is not going to be controllable below around 40%, 30 if you are lkucky. The current arrangement can stop the fan.

Somebody on here made a stirring fan, with motor in the top enclosure and metal fan rotor in the top of the oven. Maybe something similar could be done with the cooling fan, but using a BLDC motor controllable down to zero rpm?

aaronw2, 2019-02-10:
I looked into doing a stirring fan, but gave up when I could not find the small blades. Sadly the person did not disclose where they got the parts.

GitLang, 2019-02-11:
I would think he cut them from a piece of 16G or 18G sheet aluminium. It's an easy job if you have done that sort of thing before. Mark a centre point, and cut a disk of Ally about 4in diameter. Mark an inner circle about 1.2in to 1.5in diameter. Mark dividing lines on the circle such that it is divided into 8 or 6 segments, using straight lines right across the diameter and passing through the centre mark. Cut along each of those lines (use tinsnips) from the outside until you just reach the centre circle, and no further. You should have 6 or 8 segments attached to a central disk. Each segment will be a blade. You have to get a pair of pliers, grip the edge of a segment, and give it a small twist through, say, 10 degrees. Do each segment and make sure you twist them by the same amount. You now have a rotor of blades. carefully drill a hole in the centre to suit the shaft you are mounting it on.

MikePhillyFlyers, 2019-02-11:
> Reading through all the interesting discussions on this project prior to purchasing my own T962A.
> @MikePhillyFlyers that's a nice circuit to control the fan, somewhat annoying you have to go SPI -> D/A and then control the FL5160. Did you actually get it working reliably, I read some of your comments on your hackaday projects and it seemed you were having odd brownout issues.
> I wondered about the zero crossing mod, fans are inductive which would not usually be the best for ZC however they have inertia which would tend to mitigate the back EMF.
> Has anyone starting with a fresh board design thought about changing the fan completely for a 12V DC one, removing the 9V transformer and putting a proper 10W or so 12V power supply on the board to run it? A 12V DC fan of the same size and airflow needs about 500mA and ought to be much easier to control with PWM and save any more mains switching. Most of them of course are plastic, but it's inflow rather than outflow so I don't see an issue. $25 makes the whole janky AC fan issue go away.

Yeah, the brownout issues with the FL5160 were because the 9VAC transformer was putting the VS supply right at the brownout margin for the FL5160.. and sometimes when running the elements fully on, and your profile starts going down, the FAN will then need to go full on... right at this point the VS would drop just enough to do a POR to the 5160...

I swapped out my 9VAC for a better 12VAC transformer, 500mA...  the other board I'm making using an STM32, I'm instead providing a small boost supply for a stable 12V to the 5160... and you can keep the 9VAC stock transformer.

I too did consider just plopping in a nice DC fan, with similar CFM, and just elminate the AC all together... that would be nice.... I just couldn't find at the time a DC fan with matching (or better) CFM than the stock fan, that didn't require a 1.5A or higher 12V transformer.  Not that you can't find that!

But I was trying to minimize mods to the case, etc for having to mount  a larger transformer... the one I dropped in was a perfect fit, no drilling required.

But not a bad idea, maybe I'll toy with that for my STM32 version, and just go all DC... I'll check out those fans you referenced and see how things look...

rolandking, 2019-02-11:
211cft/m is about 2x the airflow of the stock fan (if the link above is correct), you should be able to find DC fans of about 100cuft/m which run on about 500mA. And of course just putting in that AC transformer isn't enough, you need DC. When I was musing this I was going to use one of the on-board 10W or so 12v power supplies which are $15 or so. However as gitlang points out above, the problem with those fans, even the ones with PWM control built-in, is minimum speed, you can't get them to spin really slowly and you can't get them to stop either (without just cutting the power); that was one piece I hadn't thought through. Also, as far as I can tell, they tend to start fast and then drop down to the PWM-regulated speed, so every time you start it up it's going to blast air for a few seconds. 
So even though I liked that idea to start with, I'm now not so sure. 
I spent some time looking through the datasheet for the FL51x0; for a device designed to be attached to the mains power supply I'd call that datasheet sketchy. No comments on the power ratings of the caps and resistors required and not much about efficiency. I know you drove yours from regulated 12v and kept just the zero crossing hot which may have been a good plan, I'd really want better documentation before I hooked one of those up to 240v. 
And a complete aside, does anyone know why the ESTechnical mod uses a 25A expensive SSR to switch the fan? It's a 20W fan, seemed a bit of overkill.

MikePhillyFlyers, 2019-02-12:
> 211cft/m is about 2x the airflow of the stock fan (if the link above is correct), you should be able to find DC fans of about 100cuft/m which run on about 500mA. And of course just putting in that [...]

Yeah, I meant to clarify that, if dropping in a 12VAC transformer, you'd of course have to power the DC fan from diode/rectifier output off the stock controller...

Uggh, I didn't know that about those DC fans, that they don't really shut off, not good... and no slow speeds isn't good either... so that kind of turns me off to the entire idea..

So anyhow, the FL51xx, yeah, the datasheet just sucks, it's missing so much info, and the circuit examples were very flaky.  It took me quite some time to figure out how to get the thing to really work, and what portions to eliminate.

So yeah, for the ZC feedback resistor, of course just make sure you buy high voltage resistors, I picked up 400V '0805' package resistors that were fine.

The 'OC Sense 1' & 'OC Sense 2' feed a comparator circuit that is supposed to protect the FETs, but in normal operation they actually impeded the chip from producing the max AC output when fully 'ON' driven to DIM control... 

Just pulling them LO eliminated the diminished switching issues...

Also because of the GND issues between digital gnd and AC neutral/gnd, it becomes tricky if you try to use their circuit design, and power the chip from the AC mains, but control the chip from the rest of the digital circuit.  

Which is why I ended up with the circuit I did, where the AC is just isolated, and FETs are driven via an opto-iso, and 5160 and all logic is driven via same DC supply as the rest of the controller board.  Also made it safer as my board could then be gnd'd to the case with earth gnd.

(in the OEM controller they float the board as you notice the felt spacers they used when mounting the board to the case)

rolandking, 2019-02-13:
@MikePhillyFlyers you got me interested in that dimmer chip now to the point I checked some of their power numbers. I don't understand their neutral driven circuit at all, it seems to rely on a potential between N and GND and in my house at least there's not always one, this probably means I don't understand what "N" is properly. Their other examples I do understand. 

Anyway it got me wondering a different way to do this using one of their mains powered examples. You have the problem you raised of driving the 0-2.5 input signal safely and with no common ground. My thought there was to isolate the original PWM train through an opto-isolator, which deals with the grounding issue, and run a PWM to DAC off the 5V VDD from the FL51x0. There's a few cheap PWM to voltage converters which will output 0-2.5 volts, exactly right to drive the FL51x0. (eg LTC2645)

So it looks a bit like this : PWM -> Opto -- DC to AC crossing --> Opto -> PWM/DAC -> FL51x0

only issue there is I don't think the FL51x0 5V VDD output has the 3-4mA needed to drive the PWM/DAC and the 'receive' side of the opto coupler or it would be a nice self-contained design completely compatible with the current PWM output, totally isolated and self-mains powered. I have an idea for fixing that too, might buy a few bits and try it out.

usernamevalis, 2019-09-15:
Hello Everyone
I came across this looking for info on a different project I'm been working on that also uses the Fl51x0. I've had my own issues and thought I'd share it here:

> @MikePhillyFlyers ...
> So it looks a bit like this : PWM -> Opto -- DC to AC crossing --> Opto -> PWM/DAC -> FL51x0
> 
> only issue there is I don't think the FL51x0 5V VDD output has the 3-4mA needed to drive the PWM/DAC and the 'receive' side of the opto coupler or it would be a nice self-contained design completely compatible with the current PWM output, totally isolated and self-mains powered. I have an idea for fixing that too, might buy a few bits and try it out.

I went this route with but with a digital isolator and digital pot as the Fl51x0 was powered from 240VAC and the pot was controlled from MCU (vias the isolator) that was powered from a separate/isolated DC supply. The FL5150 could NOT supply the +-10mA I needed to power the Isolator IC and pot. My setup looked like this:

MCU SPI -> digital isolator -- DC to AC crossing --> dig pot  -> FL51x0

I found this wide input LDO (https://www.onsemi.com/pub/Collateral/NCP785A-D.PDF) in a desperate attempt to at least evaluate the fl5150 on the PCB I got made. I used the LDO to power the isolator and pot while keeping them referenced to the neutral wire on the 240VAC like the fl5150. So might be of help to you. Word of caution they do get hot.

When powering the Fl51x0 from 240VAC in the 3 wire config make sure to add enough capacitance on the VS pin as I struggled with a lot of brownout issues as well

Powering the FL51x0 from 240VAC was a bit messy for me and not wanting to fab another pcb before I understood the IC a bit more, I left this route and am interested in going the DC powered route but have a question for anyone here regarding the ZC input. 

When powering the fl51x0 from a +-12VDC supply and connecting the ZC input to mains (120VAC / 250VAC) did you not get any nasty surprises with fire/smoke in the chip or connected components?
Im considering following your design but placing an optocoupler between ZC and Mains.

MikePhillyFlyers, 2019-09-22:
Hi,

There will be no issues using the 250VAC for your ZC input, as you can see in their recommendations, for a 240VAC (or so), for the ZC input you are using a 2MEG OHM resistor, which ends up giving you a very low voltage signal at the chip input (a few volts or so)..... the ZC input draws very little current, it just needs a small version of the AC waveform to detect the zero crossings.

In my circuit (the 120VAC), I'm using a 1MEG OHM resistor (as they recommend), and there are no issues.  

* just make sure you USE A 400V rated resistor, as you will get smoke if you try to use a standard resistor that is rated much lower *

Also, you really should make sure you use 3-wire connections to your circuit, ie connect the 240AC GND to your board/case GND, as I did in my design.. otherwise you can get some crazy potentials, and blow stuff up.

I too tried powering the Fl5160 from the AC mains, and not using DC power, but their designs just don't work right, and the chip just did not run.  It seems to be quite buggy, and only works right if you power it from a stable DC supply.

Also, I found it ran much smoother and at full potential if I did NOT use the OC Sense 1/2 inputs, they are supposed to protect MOSFETS from overvoltage, etc, but with them connected in the fets weren't switching at full speed.  I think again an issue in their design.

So I found pulling them down to gnd was much more efficient, and it ran at full range as specified in their output chart.

usernamevalis, 2019-10-18:
> Also, you really should make sure you use 3-wire connections to your circuit, ie connect the 240AC GND to your board/case GND, as I did in my design.. otherwise you can get some crazy potentials, and blow stuff up.

@MikePhillyFlyers thanks for the response. I would like to clarify something you mentioned (quoted above) as I don't often work with HV AC. Are you recommending to connect the Neutral wire to the GND of my circuit containing the MCU etc. driving the FL5150?

The aim of my design is to completely and safely isolate the DC from the AC section, would this not completely defeat that? I'm asking cause I don't know enough, to know what I don't know.

Thanks again

rolandking, 2019-10-19:
I would like to clarify something you mentioned (quoted above) as I don't often work with HV AC. Are you recommending to connect the Neutral wire to the GND of my circuit containing the MCU etc. driving the FL5150?
> 
> The aim of my design is to completely and safely isolate the DC from the AC section, would this not completely defeat that? I'm asking cause I don't know enough, to know what I don't know.
> 
> Thanks again

MikePhillyFlyers, 2019-10-19:
Hi, Yes, rolandking is correct, You want to just make sure the GND of your DC circuit (ie MCU, etc) is connected to real ground in your house, ie the same GND that also your AC has as GND.  ie that's why I said the '3-wire' AC setup, because the 3rd wire, the real gnd, is the actual GND in your fuse box (and also goes to earth ground outside)...

So in the offical setup in the T962A oven, when doing it right you want the GND (ie 3rd wire) of the AC (because it's real gnd in your house) to be connected to the metal of the chassis of the oven... and also the circuit board GND (ie all your DC gnd) to also be connected to this same GND.

(In the OEM setup, that comes with the oven, they used those red felt insulators and kept their entire DC circuit floating, and it was never grounded.  NOT a good idea... they did actually connect the AC GND to the chassis, but they never scraped off the paint before they mounted that gnd wire to the chassis, so normally when you get the oven right out of the box that GND connection is probably not even connected for many... )

This assures that both your DC and AC voltages are referenced from the same true gnd.   Don't worry about neutral, you really don't want to short it to GND.  (even though in reality Neutral is actually connected directly to GND inside your breaker box, the idea of neutral with AC is that the hot/neutral wires are the AC circuit, the 3rd wire GND is for safety, so that your appliance, etc has it's chassis directly connected to GND.  This assures that any type of short could never energize the chassis...if hot ever shorted to the chassis it would be an immediate trip of the breaker, the chassis could never shock you.

So to be safe and not risk blowing up any FETs or FL5160s, just make sure your board GND is connected to AC GND, ie normally the GREEN wire in a 3-wire plug.  It will normally work fine without doing this, but it's always best, especially when testing, to have your gnds all the same.

usernamevalis, 2019-10-21:
@rolandking @MikePhillyFlyers 
Thank you both for your feedback. 
This is what I thought but I wanted to make sure. I really appreciate the help.

fastbike, 2021-01-19:
> @MikePhillyFlyers you got me interested in that dimmer chip now to the point I checked some of their power numbers. I don't understand their neutral driven circuit at all, it seems to rely on a [...]

If you're still looking at this, I'm working on a capacitive power supply circuit design for the FL51x0 to drive an op amp on the hi side of the opto isolator.

GitLang, 2021-01-19:
Potential between ground/earth and neutral will vary a lot depending on type of earthing and local conditions, but is often a few volts and quite noisy in the UK. Either way, making use of it in any way at all is bad news. Besides being against any regulations I know of, if you draw any current between neutral and earth you are going to trip any earth leakage detector / RCB in circuit.

fastbike, 2021-01-19:
The GND symbol in the circuits in the FL51x0 datasheet is NOT  protective earth. 
Make sure your circuit draws power for the chip from either the potential between LINE and LOAD for 2 wire dimmers, or preferably the potential between LINE and neutral.