UnifiedEngineering/T-962-improvements #139: Install C202 IR block from ayoue in oven: Good idea?

dtmr-e, 2018-04-21:
I wonder if this spare part from ayoue would be an advantageous upgrade for the T-962 (if one solders mostly sub-100x100mm PCBs in the center of the oven):

[image]

The C202 is a compact 125x125mm block of heating elements, 100-380°C, 600W, designed for an aoyue IR pre-heater. I assume it would provide more even heat distribution in the center of the oven than the original, thin and long elements. In the original pre-heater, a single thermocouple sticks out in the middle of the tubes, so there must be a hole.

taliesin, 2018-06-15:
Pretty late for an answer, still ...
I think the added thermal mass will not improve the overall performance. Especially for smaller PCBs the radiation pattern will be flat enough in the middle of the drawer (can't really prove that right now!)

vjalle, 2022-01-25:
I built in this thing into my T962:
https://de.aliexpress.com/item/32904855058.html
Works well, but the bottom of the oven has to be vandalized to some extent.

southern-boy, 2022-01-25:
How did you have to modify the bottom of the T962?  Do you have any 
pictures of the mod and the procedure to change it?

GitLang, 2022-01-26:
What made you decide on the bottom of the oven?. Intuitively, I would think the top would be better.