UnifiedEngineering/T-962-improvements #140: Change on board thermocouple amplifier to get more reliable results

STB3, 2018-05-04:
The TLC27L2CD has an iput offset of max. 10mV. For example the voltage at the thermocouple pins at 245°C is approx. 10mV! This could be somehow compensated by the calibration potentiometers but it is not very reliable. The offset drift of approx 1µV/K is fairly o.k. but we can do better.

Desolder the IC and solder a pin compatible LTC2051CS8 and you'll get max. 3µV(!!) offset and a drift over IC temperature of 3nV(!!). This means the LT2051 is a decade better than the old TI OpAmp.

taliesin, 2018-06-15:
I'm using a OPA2333, see https://github.com/taliesin/T-962-improvements/wiki/Fixing-the--internal-Thermocouples

Maurizio66blu, 2022-04-24:
> I'm using a OPA2333, see https://github.com/taliesin/T-962-improvements/wiki/Fixing-the--internal-Thermocouples

I bought a OPA2333 but it is not clear what to do. 
Please, can someone who did this explain it better? Thanks

taliesin, 2022-04-24:
In the text under https://github.com/taliesin/T-962-improvements/wiki/Fixing-the--internal-Thermocouples#replace-the-amplifier-set-correct-gain there should be enough information about what to do. You need to de-solder the original amplifiers, remove the trim resistors and replace them by fixed value resistors with the given values.

Maurizio66blu, 2022-09-13:
Sorry but after making the hardware changes do I need to make changes to the software too or is it enough to change the software settings?

Maurizio66blu, 2022-09-14:
> In the text under https://github.com/taliesin/T-962-improvements/wiki/Fixing-the--internal-Thermocouples#replace-the-amplifier-set-correct-gain there should be enough information about what to do. [...]

Sorry but after making the hardware changes do I need to make changes to the software too or is it enough to change the software settings?

Both the gain and the offset have to be changed, right?
Is that enough or is it also necessary to recompile the software?

taliesin, 2022-09-14:
You need to use my software tree as the new OPamp uses a different gain setting and therefore the temperature calculation in adc.c must be adjusted. Take care to use the correct settings in config.h, they must correspond to the hardware patches you made.

Maurizio66blu, 2022-09-14:
> You need to use my software tree as the new op-amp uses a different gain setting and therefore the temperature calculation in adc.c must be adjusted. Take care to use the correct settings in config.h, they must correspond to the hardware patches you made.
> 
> With this op-amp there is typically no need to adjust gain and offset settings.

Please, I would like to know if you have a way to publish the already modified sources? 
I think they can also be useful to others who want to make the same change.

taliesin, 2022-09-14:
There is not much to do. Simply enable the USE_PRECISION_OPAMP in config.h for the other options read the documentation above and adapt as necessary, I can only guess what your hardware setup is.
The source is in https://github.com/taliesin/T-962-improvements, if you need further insight please read the documentation there, it should be quite complete. If you have detailed questions, I'll do my best to answer them ... it's been a while since I wrote the code.

Maurizio66blu, 2022-09-15:
> There is not much to do. Simply enable the USE_PRECISION_OPAMP in config.h for the other options read the documentation above and adapt as necessary, I can only guess what your hardware setup is. [...]

I have seen that you have made a series of changes to the code which, however, do not interest me.
I would prefer to just take the change to the adc.c file so that everything else remains unchanged as in the latest version (0.52).
Can I do this or do I have to modify some other files after modifying adc.c?

taliesin, 2022-09-15:
I'm sorry, but you will on your own, when pulling in only parts of my code. Taking only adc.c will most probably not work, but give it a try.

Maurizio66blu, 2022-09-16:
> I'm sorry, but you will on your own, when pulling in only parts of my code. Taking only adc.c will most probably not work, but give it a try.

I would like to know if your code is based on version 0.52 or if it is older.

Also how can I then configure the config.h file to implement only the change to the op amp and have everything else working as the standard version 0.52.

However, thank you for your support.

taliesin, 2022-09-16:
> I would like to know if your code is based on version 0.52 or if it is older.

It's based on 0.52 (simple check of the git log would have shown that).

> Also how can I then configure the config.h file to implement only the change to the op amp and have everything else working as the standard version 0.52.

You can't. My fork deviates quite a lot from the original version. So either take it or leave it. However, if you are using the oven stand-alone (i.e. not connected to a computer), the oven will perform almost the same as the original version 0.52, if configured to do so.
You leave away USE_SECONDARY_HEATER, USE_FET_DRIVER and define CONTROL_TEMPERATURE to LR_AVERAGE and should be done. No other HW-patches are necessary.

Maurizio66blu, 2022-09-18:
> It's based on 0.52 (simple check of the git log would have shown that).
>
>
> You can't. My fork deviates quite a lot from the original version. So either take it or leave it. However, if you are using [...]

I forgot to mention that I also have the DS18B20.
So I think I have to put "#define COLD_JUNCTION_IF TC_ONE_WIRE" or am I wrong?

A curiosity: what is the serial port used for?

taliesin, 2022-09-19:
From config.h:

```
// the CJ compensation sensor, internal means the DS18B20 (or alike), no channel!
#define COLD_JUNCTION_IF        TC_INTERNAL
```
... so leave it that way.

> A curiosity: what is the serial port used for?

It's the interface for SimpleCli, a minimalistic shell (commands and logging), however currently there is no tool for the PC to deal with the logging feature.

Maurizio66blu, 2024-08-16:
Sorry if I return to write about this argument.
I have changed the amplifier, I am using now an OPA2333A from TI.
I also replaced the resistors, the 1K (R33, R37) with 1K 0,1% and 47K (R34, R38) with 2 resistor in parallel (158K 0,1%) that are equal to one from 79K.
The trimmers RW3 and RW4 was removed and now is present a short.
In this way the gain of every amplifier is equal to 80 because is 79/1+1.
The software is version 0.5.2 but it show about 850°C for channel!
The junction temperature is about 32,5°C.

I would like to avoid to change the software.
I don't understand why I don't see a normal temperature because the gain is about as the original.
During the pcb board rework, I don't have disconnected any connectors.

Can you help me telling where investigate?
Thanks in advance.

Maurizio66blu, 2024-08-16:
[image]

The OpAmp was reworked with hotair while the resistors with a solder.

taliesin, 2024-08-16:
Use the typical methods to debug electronics:

1. check the voltage at the ADC pin, does it correspond to your reading of 850°C?
2. If not the SW is broken, if yes check the amplifier input, if this corresponds to 850°C, the amplifier is ok, your thermocouple behaves strangely, otherwise your AMP is malfunctioning.
3. if it's the AMP, measure the connections (ohms) from pin to pin and validate the resistors are connected correctly (a solder may be open). Don't directly contact the resistor pins, as you might push them down and this may create a contact.
4. if the resistor connections look good, it might be a dead Opamp, it might have overheated during your hot-air reworking.

Maurizio66blu, 2024-08-17:
The ADC Vref pin 63 has 3,3V as reference.
The ADC pin TC1 and TC2 have 2,645V.
Between pin 2 of U4 and ground I measure 1Kohm.
Between pin 6 of U4 and ground I measure 1Kohm.
Between pin 2 of U4 and pin 1 I measure 79Kohm.
Between pin 6 of U4 and pin 7 I measure 79Kohm.
I'll keep you updated.

Maurizio66blu, 2024-08-17:
At room temperature, which measure do you have on TC1 and TC2?

taliesin, 2024-08-17:
As I have a different gain, my ADC voltages differ (gain=220), the input voltage of K-type thermocouples at room temperature is approx. 0mV, if referenced to GND (junction), which is probably also at room temperature. So your amplifier output should be approximately 0V as-well, which it's not.

2.645V / 80 gives about 32mv, which corresponds to about 800°C+ambient, so the ADC-reading is fine.

Maurizio66blu, 2024-08-17:
I suspect the amplifier because I bought it from Aliexpress in China.
I am going to order one 100% original from Mouser.

Maurizio66blu, 2024-12-21:
Hi!
I have solved the problem changing the OpAmp.

The one bought from Aliexpress is a fake.
To help you then I prefer to add a photo:

[image]

This one instead is original and selled from a serious seller (Mouser):

[image]

AS you can see they are different.
This explain also the differencies of price.

I remain at your disposal for further information.

Maurizio66blu, 2024-12-28:
Hi!
As already explained, I have changed the amplifier, I am using now an OPA2333A from TI.
I also replaced the resistors, the 1K (R33, R37) with 1K 0,1% and 47K (R34, R38) with 2 resistor in parallel (158K 0,1%) that are equal to one from 79K.
The trimmers RW3 and RW4 was removed and now is present a short.
In this way the gain of every amplifier is equal to 80 because is 79/1+1.

The purpose of all this is to improve the temperature reading but I found that I only partially succeeded.

The offset of both channel is -5.00.
By setting a gain of 1.00 on the left channel and 1.10 on the right channel, I get on the right almost the same temperature as the left channel but around 230°C.
Instead at 100°C the difference between the two channels is about 25°C, the right one reads less.

Is there a procedure to follow to align the left channel with the right or vice versa? Assuming that one of the two is the more precise.

GitLang, 2024-12-28:
You shouldn't be trying to align tyh channel readings to each other, you should be trying to align each one with the reading from a known good thermocouple based thermometer. For best results, either clamp the oven's TC to the thermometer's TC or preferably temporarily solder them together. You need to ensure offset and calibration of each individual side is correct. Only then can you believe what the oven is telling you. A far better solution, that many people employed several years ago was to use an addon board based on one of the MAXIM chips that reads the thermocouple and send digital temperature better than 1/10th degree C. Straight to the CPU. It is supported by the modified software here.

On the other hand... I don't know if you are using a later machine. I believe some are no longer software compatible with the software here.

Maurizio66blu, 2024-12-28:
> You shouldn't be trying to align the channel readings to each other, you should be trying to align each one with the reading from a known good thermocouple based thermometer. For best results, either clamp the oven's TC to the thermometer's TC or preferably temporarily solder them together. You need to ensure offset and calibration of each individual side is correct. Only then can you believe what the oven is telling you. A far better solution, that many people employed several years ago was to use an addon board based on one of the MAXIM chips that reads the thermocouple and send digital temperature better than 1/10th degree C, straight to the CPU. It is supported by the modified software here.
> 
> On the other hand... I don't know if you are using a later machine. I believe some are no longer software compatible with the software here.

Hi GitLang!
Please, where are the schematics with the Maxim chips you are talking about?
Thank you.

Maurizio66blu, 2024-12-29:
Can be is this one:
https://theembeddedworkshop.wordpress.com/my-projects-2/arm-projects/improving-the-t-962-reflow-oven/

GitLang, 2024-12-29:
I don't recognise that article, but the MAX31855 is definitely the chip used by several mods. It takes four thermocouples directly, compensates and converts them to a digital data stream. I can't remember if it's I2C or SSI but there's another chip on the board I have which converts it to the one that is read by the oven's CPU. I still have some bare PCB's somewhere. You are welcome to one if you can solder surface mount and you don't live in Vanuatu or similar!. I'll try to find the article.

Actually, it may have been the article you linked. I thought the PCB's I had were smaller and blue, but my memory is old. There's picture [here](https://theembeddedworkshop.wordpress.com/wp-content/uploads/2015/02/p1050329.jpg)
and a single channel Adafruit board [here ](https://www.adafruit.com/product/269)

Maurizio66blu, 2024-12-29:
> I don't recognise that article, but the MAX31855 is definitely the chip used by several mods. It takes four thermocouples directly, compensates and converts them to a digital data stream. I can't remember if it's I2C or SSI but there's another chip on the board I have which converts it to the one that is read by the oven's CPU. I still have some bare PCB's somewhere. You are welcome to one if you can solder surface mount and you don't live in Vanuatu or similar!. I'll try to find the article.
> 
> Actually, it may have been the article you linked. I thought the PCB's I had were smaller and blue, but my memory is old. In fact having looked at [part2 ](https://theembeddedworkshop.wordpress.com/my-projects-2/arm-projects/improving-the-t-962-reflow-oven-part-2/)of the article, it probably is the one. There's picture [here](https://theembeddedworkshop.wordpress.com/wp-content/uploads/2015/02/p1050329.jpg) and a single channel Adafruit board [here ](https://www.adafruit.com/product/269)

Hi GitLang!
I live in Italy and you?
No problem to use and solder smd components.
Please, I would like to know if you have a pcb spare, thank you.

GitLang, 2025-01-11:
Maurizio,

Appologies for taking so long to reply. I hunted everywhere and could not find the PCB's. Then Christmas happened. Then, with pure luck, my wife found them under a bookcase today.
[image]
[image]

I must have a stencil somewhere but I haven't found it. The majority of passives are 0603, and there's a small pitch TSSOP. I will happily send you a board or two gratis. I'm in the UK but I don't really want to give my email address here, so I'm not sure how we organise this.

Maurizio66blu, 2025-01-11:
Hi GitLang!
Please, contact me at the following email address:

[image]

Thank you

Maurizio66blu, 2025-02-03:
Hello everyone!
I am here to publicly thank GitLang for sending me two pcbs for free.
I am immensely grateful to GitLang for his generosity.
Thank you.

GitLang, 2025-02-03:
I still have about 8 more if anybody wants one.