UnifiedEngineering/T-962-improvements #141: Strange resistors on r1 -> 74hc04

Sam99C, 2018-05-05:
Hi, I’m looking at the page describing the ‘Better thermocouple interface’ with the max31850k add on board and there are a couple of resistors added to this IC - I wonder what they do? Any ideas?

Also my oven’s controller board is missing the 470k resistor on q3, (it’s 680k on the schematic but 470k in photos) and the 10k smc resistor goes to ground.  Any ideas what that does?

Thanks

taliesin, 2018-06-15:
Hi,
 which schematic are you referring to? In the schematic of the oven controller board in this repository Q3 is a diode in the input rectifier, there is no R1 connected to the 74hc04 and there is no 10k on the board?
I could probably help, if we were looking at the same schematic.

Sam99C, 2018-06-15:
I’ll have to check my notes and maybe take photos, but I think -
The schematic on the wiki.  It’s the only one I’ve found.

My board is underpopulated which isn’t unusual.  R8 (not r1) 1k resistor is missing between the 74hc04 and the lamp.  It’s actually a solder bridge iirc.  When I put a resistor in there it stopped the ir lamps coming on.

If q3 is the s8550 pnp in the center of the board (in a rectagle called jtag!) mine is very different.   As you know the schematic shows a 680k between base and emitter and an unspecified resistor between 3.3v and collector which I haven’t found nor found c1.  Every photo I can find has 470k resistor, not 680k, besides the transistor, which is missing for me.  There is also a 10k smc resistor on the emitter, only mine doesn’t go to the same place - it’s hand soldered to a grounded pad to the right.  My board may be wired differently as that rectangle of pins isn’t marked jtag on mine.

No one I’ve shown the schematic to knows what q3 does on pin 56 of the mcu.

I can confirm more detail and send photos later if needed

Sam99C, 2018-06-15:
I’m more interested in why the screen almost always doesn’t come up on first turn on, but then it does as long as you don’t leave it off for very long.  The backlight comes on but nothing on the lcd.  The system fan behaves normally (spins up then down iirc).
It feels like a capacitor charges up and then it works, but I haven’t found anything obvious in the  hardware so might be firmware?

taliesin, 2018-06-15:
Concerning the opto-couplers you might want to have a look at https://github.com/taliesin/T-962-improvements/wiki/Basic-electrical-changes. The driving currents are simply too low. Short circuiting R8 may be ok in Eastern Asia, but actually is NOT. Either you go the full length as I did or only change the opto-coupler to a more sensitive one.
The S8550 circuitry pulls the 'S' key low (simulates a key press) for a short while after start-up, I don't know what the original intention was, but I think it is completely useless (at least with the firmware in this repo). My schematics shows a 150k next to the 470nF capacitor which sets a decay time of about 70ms and charges up via about 4k to 3.3V (about 2.5ms).

About pin 56 ... S8850 emitter is connected to pin 55 on my schematic (and board), pin 56 is not connected at all.

The LCD has a RST pin (17) which is connected to a simple RC combination, maybe this is broken. Didn't look into the specs of the display, but maybe you should check the connection and the parts there. If C17 was damaged the RST wouldn't stay low for a proper start-up time.

You seem to have received a 'Chinese Lemon' (https://en.wikipedia.org/wiki/Lemon_(automobile)). If Chinese was not enough in the first place.

tinfever, 2023-06-20:
In case it helps anyone in the future, after adding the circulation fan on my unit I started having an odd issue where when I turned on the mains switch, the unit would turn on and immediate start a reflow cycle, like the S key was being automatically pressed or glitched some how. I fixed this by cutting the lead to the base of the S8550 transistor.

I didn't measure it but when simulating the circuit shown in one of the schematic versions floating around, it seems like that circuit would constantly be pulling down on the S key signal pin. So that pin is pulled to 1.45V instead of the normal 3.3V. I'm guessing the additional noise from the circulation fan motor was just enough that on start up the MCU was registered a button press and starting the cycle.